open3e/open3e #3: Viessmann CAN-BUS Gateway

DpunktS, 2023-07-07:
Hello, I recently became the owner of a Vitocal 250-A. 

Does your gateway have the same functionality as the WAGO gateway from Viessmann, so can you also describe data points?  Viessmann
provides an extensive data point list for the WAGO gateway.[Viessmann can-bus data points](https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://www.viessmann-community.com/viessmann/attachments/viessmann/customers-heatpump-hybrid/74546/1/6196307%2520Kundendatenpunktliste%2520Vitocal%2520(1).pdf&ved=2ahUKEwjG2vGx9_z_AhU9SPEDHVZWC20QFnoECAwQAQ&usg=AOvVaw0nmGID6Ia15oTUsnw9n43f) 

You could also emulate a completely overpriced Viessmann energy meter. 

 I think it's good to go directly from can-bus to mqtt.

abnoname, 2023-07-15:
Hi, 

yes you can use open3e similar to the official wago gateway for reading out known datapoints. Only writing to datapoints is untested and not safe for now. I will soon commit some experimental code.

Emulating the energy meter (e.g. from mqtt data) is also on my todo list as the can dataframes from EC380A  are known. Those (simple) frames with power data are send regularily from the meters to the external can bus.

This project was developed with a Vitocal 200-S (E3 generation) but was also tested with a Vitocal 250-A :-) Enjoy logging data! open3e is able to publish the datapoints to mqtt.

Daniel210117, 2023-08-10:
If we can set/write to control the power of Heatpump, that would be so amazing. If it‘s possible.

Rosi1981, 2023-08-12:
Can you tell us more about the dataframes from E380CA?

DpunktS, 2023-08-15:
I would like to get the iobroker.can-bus adapter to run for my vitocal 250-a, but no matter what I set, I only get nonsensical values.  Can someone help me with what I have to set so that correct values ​​come out.

M4n197, 2023-08-15:
> Can you tell us more about the dataframes from E380CA?

So far I've only managed to interpret messages with ID "511" (0x1FF), which translates into the current date and time. These messages are sent every second (1Hz). 
| byte idx | value |
| --- | --- |
| 0 | day |
| 1 | month |
| 2 | year |
| 3 | hours |
| 4 | minutes |
| 5 | seconds |
| 6 & 7 | always `0x00` |

The content of messages with ID "598" (0x256) seems to be the measured grid voltages (line to neutral, rms).  
| byte idx | value |
| --- | --- |
| 0 | line voltage phase 1 ? |
| 1 | always `0x00` |
| 2 | line voltage phase 2 ? |
| 3 | always `0x00` |
| 4 | line voltage phase 3 ? |
| 5 | always `0x00` |
| 6 | ??? |
| 7 | always `0x13` |

By now I've only logged messages during operation. Therefore I can't quite say, from where these messages originate from. (The E380CA is communicating with the pv-inverter VitoCharge VX3.) 

My testing procedure is slow and has its flaws, but on the other hand i'm just a tinkerer in his freetime and no data-analyst. I hope this might help someone or result in further input on this topic.  If there is any interest, I can supply some captured raw data.

DpunktS, 2023-08-22:
I have connected a Vitoair FS 300E to my Vitocal 250-a via CAN bus. Is there any knowledge about this? I have not yet received any data points from the Vitoair.

philippoo66, 2023-10-09:
@Rosi1981 

the energy meter works in a different way than with open3e currently supported. open3e for now is based on UDS request/response, the E380 posts its data by CANopen PDOs. some important are (sorry, German terms) 

[image]

ip6li, 2023-10-10:
Are there any information available how Vitocal 250 uses data points delivered by E380CA-ZK05665? I would prefer to use Open3E instead of two E380CA-ZK05665. Fronius smartmeter delivers all information which are needed to control Vitocal 250 by its API.

philippoo66, 2023-10-10:
up to now we did not (even try to) figure out how Vitocal 250 uses data delivered by E380. sorry - matter of time ;-) 

if you will find out, please let us know!

djarek, 2023-11-21:
The energy meter is used by the internal EMS (if enabled) to either monitor power flow to/from the grid or to monitor a 3rd party photovoltaic inverter, depending on the configuration. Optimization algorithms (if activated) in the EMS make decisions based on the measurements from the meters.

TheSmartGerman, 2023-11-22:
I do have a SDM620v2 Modbus already installed in my Main to use the data in Homeassistant. Right at the moment I'm considering to buy a E380 to get more flexible Energy usage from my PV with my Vitocal 250. 
Will it be possible to "map" the modbus registers from RS485 to CAN? Is the PDO write already possible?

@philippoo66 
Is your listing shows all E380 Data? 

Offtopic:
I use SG inputs at the moment to use exzessive PV power, but when I set state 3 or 4 the Vitocal seems to operate at 100% (Hotwater draws 2,5kw / Heating draws 7kw). And in this case suddenly draws a lot of Battery / Grid current too.
Can someone confirm the useage of a E380 will help to control the Heatpume more dynamic. 
Is it possible to config how much the heatpump will draw or feed into the grid?

philippoo66, 2023-11-22:
@MyHomeMyData you know the energy meter better...

We/open3e do/does not do any PDO writing. Only the energy meter sends PDOs, Vitocal is communicating via UDS, a different higher-layer protocol than CANopen. Writing by data identifier is working (almost since the beginning). For scalar int / decimals it is implemented for 'plain values'. Anything else (complex data, others than numerical) you need to specify the hex values to get written. This is no 'rocket science', but you need to know what to do.

Do you intend to use open3e for writing to Vitocal? Then let us know what you want to write and we will tell you what to do.

In general I think if you have some 'smart (home) system', it is a better idea to control setpoints and modes etc. of Vitocal than to use the built-in SG functionality. This way it should be possible to control the drawn power of the heat pump. As far as I remember @jwende does already operate that way.

playcom-de, 2023-11-22:
I refused to buy the overpriced electricity meter (E380) from Viessmann. I also have a Vitocal 250 and would like to control it via my Smart Home (Home Assistant). To do this, I bought this (https://www.amazon.de/dp/B0CF9SVTNY) Modbus electricity meter and "only" connected it in front of the heating. My smart home already knows how much electricity my PV system (E3DC) generates and how much of it goes into the grid. What's much more interesting (for me) is how high the Vitocal's consumption is and the E380 doesn't show me that at all, does it? In any case, I now have the Vitocal's consumption in the Smart Home and can use it to define my own rules. At least as soon as I also have the Vitocal (via Open3E) in the Home Assistant. I'm still working on that at the moment.

philippoo66, 2023-11-22:
I already thought about disadvising E380 but since I have an old fashioned Optolink type gas boiler, I prefered to say nothing at all ;-) I think what the energy meter meters depends on where it is installed.

what is your problem in getting open3e working? it should be quite easy today...

MyHomeMyData, 2023-11-22:
@TheSmartGerman Pls. have a look to the Readme of [E3onCAN](https://github.com/MyHomeMyData/E3onCAN). There's listed what I figured out about E380 CAN messages so far.

I'm using a E380, Vitocharge VX3 and Vitocal 250. With this combination it is possible to use advanced enegery management functions of vitocal, e.g. "PV Überschuss in Wärme". However, to my understanding this will not work with other PV suppliers.

Possibly it could help to place a question regarding energy management to Viessmann forum.

jwende, 2023-11-22:
Just an idea: would it be possible to ‚emulate‘ an E380 by using data from an other energy meter ?

achgut, 2023-11-22:
Another thought: Check out EVCC https://github.com/evcc-io 
I use it to charge my car when PV is available. Open3E is used with a MQTT Broker to integrate PV Input and Battery to EVCC. 
 
With EVCC you can get the data from many smart meters and I believe it should be possible to switch on water heating if PV power is available.

djarek, 2023-11-22:
> **Offtopic:** I use SG inputs at the moment to use exzessive PV power, but when I set state 3 or 4 the Vitocal seems to operate at 100% (Hotwater draws 2,5kw / Heating draws 7kw). And in this case suddenly draws a lot of Battery / Grid current too. Can someone confirm the useage of a E380 will help to control the Heatpume more dynamic. Is it possible to config how much the heatpump will draw or feed into the grid? @djarek Is this the configuration you mentioned?

Not exactly, when the internal EMS is activated and the appropriate features are activated, the Heatpump will by controlled by the internal EMS to maximize self-consumption and minimize curtailment due to feed-in limitation (assuming your installation has such a limit imposed by the DSO). There's no way to customize how aggressive the optimization is.

> I'm using a E380, Vitocharge VX3 and Vitocal 250. With this combination it is possible to use advanced enegery management functions of vitocal, e.g. "PV Überschuss in Wärme". However, to my understanding this will not work with other PV suppliers.
Actually, it does work with 3rd party PV inverters, but requires extra equipment and new Vitocal firmware(2023).

> Just an idea: would it be possible to ‚emulate‘ an E380 by using data from an other energy meter ?

In a Vitocal + Vitocharge (or just Vitocharge) system this may result in issues with the Vitocharge, since there are fairly specfic requirements on the control logic (feed-in limitation). Other than that, an emulated meter is obviously not supported ;).

jwende, 2023-11-22:
Interesting - thank you.
Just a little question: what is EMS, how is this activated and what is DSO?
J.

TheSmartGerman, 2023-11-23:
> Some thoughts from my side:
> 
>     * I don‘t use SG for the time being (I don‘t have the contacts connected yet)
> 
>     * I don‘t think that SG will bring any good value for the same reason as already mentioned (no control about the amount of energy used during that time)
> 
>     * Someone from VM told that the 250 could be dynamically controlled if you are use an E380 energy meter, but I don‘t found any reasonable details so far
> 
> 
> Μy strategy so far: I have a smart meter between my PV, Home and energy grid - so I can see how much goes into the grid. So I have following rules planed:
> 
>     * heating is always set to standard - default value 18°
> 
>     * Hot water production is enabled all day - default value 40°
> 
>     * Hot water target is changed to 55° if grid production is over 2 kw … mainly in summer
> 
>     * Heating is increased from 18° in steps of 2 if grid production is over 1 kw … mainly in winter (limit at 25°)
> 
>     * Heating is decreased by 2° if no grid production is detected and energy price is over limit (get tibber feed) (lower limit 16 °)
>       All of this can be achieved by using the standard Web API - but I don‘t like it 🤷🏻‍♂️
>       Cheers
>       Jörg

Thx, I'm just found the project and started reading. I like your solution! Maybe a nice fallback. I still like to figure out what's possible with the E380 + Vitocal 250. I don't have any other VM components.
As I mentioned, I do have a SM630 Modbus installed, if it's possible to emulate the E380 would be great!

@djarek How the EMS is activated?

Anyway will try to get the Open3e up and running, right at the moment I can't access my HP due to server maintenance. This is kinde of annoying.

philippoo66, 2023-11-23:
> possible to emulate the E380

since we can easily monitor _every_ information communicated between the 'control unit' and the E380, it should not be a big problem to emulate E380. During Operation E380 only transmits some CANopen PDOs (see above) which could easily get sent by any CAN node. 

What we did not report up to now is if there are certain informations transmitted/communicated during system-startup. But even if there are some more than just the CANopen NMT messages, is will be no problem to realize this, too.

What might be a bit more challenging is to find out if the E380 must be regarded in initial system setup (usually done by Viessmann technican) and how this could be done supplementary...

maromme, 2024-01-09:
> I have connected a Vitoair FS 300E to my Vitocal 250-a via CAN bus. Is there any knowledge about this? I have not yet received any data points from the Vitoair.

@DpunktS Did you manage to get any data points from the Vitoair? There are really little (bette: no) informations to find on the Vitoair can bus.

philippoo66, 2024-01-09:
@maromme  run Open3E_depictSystem to scan your system. afterwards (if no timeout terror issue) you will have everything available. You have to use the ECU address according to the Vitoair then (different from 680h).

maromme, 2024-01-10:
@philippoo66 thanks for your reply.  Sounds promising. But first I'll try that approach: https://github.com/open3e/open3e/discussions/29#discussioncomment-7672913
I anyway use homeassistant and esphome and the necessary hardware is there. 

Thanks to the work done here, it seems there is a list of the vitoairs DIDs. Just need some more time testing.

philippoo66, 2024-01-10:
@maromme  do I understand right that you will use an ESP32? Then better use it as 'pure CAN adapter' and work with canneloni / the virtaul CAN bus...  https://github.com/open3e/open3e/discussions/29#discussioncomment-7621145

TheSmartGerman, 2024-01-11:
@maromme @philippoo66 
I tried the esp32 verstion first, but I failed to get the cannellonie for esp32 running.  
lib's where updated and I was not able to get the right versions. e.g. CAN Libs from Expressif IDF changed function calls are now TWAI_ not CAN_ anymore. I used VSCode + PlatformIO...
This is my attemped: https://github.com/TheSmartGerman/ESP32_CAN_Interface

maromme, 2024-01-11:
I get your point, still I like the idea of an stand allone device... I just have to give it a try first ...

maromme, 2024-01-12:
Wasn't a big issue to get the frist 4 DIDs (temperatures) working (as I knew the DID from the open3e project and a bit of copy&paste):
For OutdoorAirTemperatureSensor (DID: 327 = 0x**0147**)  I send a send message with my ESP can-id 680:
0x 03 22 ** 01 47 ** 00 00 00 00
Then I receive a msg from can-id 690:
0x 10 0C 62 ** 01 47 ** DB FF CB
The part after the DID reveals as a temperature of -3.7°C.
Could you please give me an hint for understanding the syntax / adressing of the messages? I understand that the "03" of the send message indicates IsoTP single Frame with 3 bytes payload, but no clue what abot the following byte "22". The return msg I understand "10" as the "first frame" indicator, "01 47" as the DID and "DB FF" as the themperature, but then no clue.

MyHomeMyData, 2024-01-12:
A good and compact description of UDSonCAN protocol can be found [here](https://www.csselectronics.com/pages/uds-protocol-tutorial-unified-diagnostic-services).

maromme, 2024-01-12:
Great help thank you! That documentation is really good.
That means "0C" indicates 12 byte of data, I do need also the 2nd frame for the full message:
```
   0  1  2  3  4  5  6  7
0x 10 0C 62 01 47 ED FF CB
0x 21 FF 8F 00 EC FF 00 55
```
the message itselfe should be then `62 01 47 ED FF CB FF 8F 00 EC FF 00 55` where the last 55 is padding for to get 12 byte. The first three byte are SID and DID. the 5th and 6th the temperature ("ED FF" I decode as -1,9°C). But the rest?
I guess I'm going a bit off topic here, is it possible to move that to discussions?

MyHomeMyData, 2024-01-12:
Open3Edatapoints.py:
327 : O3EComplexType(9, "OutdoorAirTemperatureSensor",[O3EInt16(2, "Actual", signed=True), O3EInt16(2, "Minimum", signed=True), O3EInt16(2, "Maximum", signed=True), O3EInt16(2, "Average", signed=True), O3EByteVal(1, "Error")])

The sensor data consists of four 2-byte values: Actual, Minumum, Maximum, Average.

philippoo66, 2024-01-13:
@maromme there is a discussion "ESP32 with CAN" already - continue there?! https://github.com/open3e/open3e/discussions/29#discussioncomment-8117128