open3e/open3e #4: udsoncan-1.17.1 no longer available

Dreman65, 2023-07-31:
There is a new version of 'udsoncan' (1.18.1) available. Has this new version any influence on the function of the overall software? The reason for my question is, that my Raspi CAN-configuration (vitodens 300-W-B3HG)  is running well with the can-utils commands but not with the open3E commands.

Dreman65, 2023-08-03:
I have now tested the open3E software with both 'udsoncan' versions: no difference in the failure behaviour on my Raspi CAN-configuration. So the reason for not running on my system seems to have (unfortunately) other reasons. Sorry for the trouble caused.

abnoname, 2023-08-03:
Hi, thank you for your interest in the project :-) Yes udsoncan will be updated in the requirements soon. 
Here some facts for the current development state of E3 gas boilers: We are currently working on decoding datapoints for gas devices. First scans using doip are done. If you want to test, then try connecting the service access point and connect through doip, e.g. python3 Open3Eclient.py -d 192.168.3.1 -r 256 -v. Datapoint 256 should be readable. Sadly at this time the mechanism for accessing uds over can is not fully understood as it is done for heat pumps. If you want to contribute, please try a can diagnostic scan using https://github.com/CaringCaribou/caringcaribou. There you should try cc.py uds discovery and then cc.py uds services.

closed 2024-08-26 by MyHomeMyData