open3e/open3e #10: Time Out Problem

achgut, 2023-10-07:
Hi 
I am running the pyhton script on a raspberry 4 with 64 bit. I read that there might be some timing issues with udsoncan. 
Is there any solution to this?

One strange behavior with my VX3:
If I run "candump can0" and right after that "python3 Open3Eclient.py -c can0 -dev **vdens** -r 256 -v" I get the busidentifier back.
The same does not work with "python3 Open3Eclient.py -c can0 -dev **vx3** -r 256 -v"

philippoo66, 2023-10-09:
Hi @achgut !

As we mentioned in the [Viessmann Thread](https://www.viessmann-community.com/t5/Konnektivitaet/CAN-Bus-Home-Automation-E3-Generation-lokal-und-kostenlos/m-p/367963) the timeout issue is not really a timeout issue but a problem of some constellation of Raspi kind / OS (Version, sometimes 32/64 bit) / Python version / probably CAN adapter. Unforunately we haven't figured out the core cause and accordingly not the solution up to now.

What we saw is that Raspi 4 with 64bit raspbian bullseye usually works, Raspi 3 usually works without further issues.

It is possible to verify the issue by doing candump. If you see 680 03 22 xy zy frame and a subsequent 690 frame, then the UDS communication is working but there is some (library) trouble getting the data 'up through' to open3e.

It might help for figuring out the issue if more pepole would post system constellations where it is / is not working.

> One strange behavior with my VX3:
If I run...

In the USD system there is always a "Main Control Unit" and further 'slave devices'. The Main Control Unit always gets addressed by the ECU Address (COB-ID) 680h and responds on 690h. For the slave devices different ECU Address (COB-ID) need to be used. 

Probably you picked a develop branch where for the VX3 a different ECU address pair (for VX3 as slave) is defined. What 6xy COB-IDs you see in both cases with the RDBI request frames 03 22 01 00?

achgut, 2023-10-09:
I have a VX3 and the E380 meter connected to the can bus and my raspberry.
I do not see 680/690. 

Here is the candump: 
  can0  254   [8]  02 00 01 00 01 00 00 01
  can0  25A   [8]  EB 00 00 00 28 FB FF FF
  can0  250   [8]  63 01 4D FF 67 FF 17 00
  can0  252   [8]  E9 FF D2 FF C9 FF 84 FF
  can0  761   [1]  05
  can0  256   [8]  EE 00 EE 00 EE 00 89 13
  can0  000   [2]  82 4E
  can0  250   [8]  67 01 4A FF 66 FF 18 00
  can0  252   [8]  EA FF D2 FF C9 FF 85 FF
  can0  25A   [8]  ED 00 00 00 35 FB FF FF
  can0  000   [2]  82 59
  can0  254   [8]  02 00 01 00 01 00 00 03
  can0  25A   [8]  6F 00 00 00 22 FB FF FF
  can0  250   [8]  60 01 48 FF 63 FF 0B 00
  can0  252   [8]  E8 FF D2 FF C9 FF 83 FF
  can0  701   [1]  05
  can0  1FF   [8]  09 0A 17 10 2B 1C 00 00
  can0  271   [1]  00
  can0  1FD   [1]  00
  can0  256   [8]  EE 00 EE 00 EE 00 89 13
  can0  25C   [8]  17 7D 05 00 00 00 00 00
  can0  258   [8]  B7 8B 5B 4A B9 8B 2F 4A
  can0  250   [8]  54 01 46 FF 60 FF FA FF
  can0  252   [8]  E9 FF D1 FF C9 FF 84 FF
  can0  25A   [8]  C5 FF FF FF 26 FB FF FF
  can0  254   [8]  02 00 01 00 01 00 04 01
  can0  250   [8]  5D 01 46 FF 60 FF 03 00
  can0  252   [8]  E9 FF D3 FF C8 FF 84 FF
  can0  25A   [8]  1C 00 00 00 27 FB FF FF

philippoo66, 2023-10-09:
680/690 is used by open3e when reading data via UDS. you need to run Open3Eclient (e.g. -a), then you will see the according frames.

philippoo66, 2023-10-09:
bevor wir auf Englisch aneinander vorbei reden... Siehst du auch wenn du den Open3Eclient laufen lässt nichteinmal 680 Frames im Dump? Wir hatten heute auch so einen Fall, wo isotp nix 'raus bekam'. Sieht das bei dir dann auch so aus?

[image]

achgut, 2023-10-10:
Hallo
Habe jetzt verstanden, dass ich beides parallel laufen lassen muss. Im Candump habe ich nach Befehl "python3 Open3Eclient.py -c can0 -dev vx3 -r 1664 -v" folgendes gefunden: 
      **can0  680   [4]  03 22 06 80**
Aber immer nur einen 680 "Aufruf". Kein 690. Bei Vx3 Datapoints noch kein Feedback gelungen.

Das Feedack zum Befehl war folgendes:
_[ERROR] UdsClient: [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)
Traceback (most recent call last):
  File "/home/ubuntu/open3e_org/Open3Eclient.py", line 111, in <module>
    response = client.read_data_by_identifier([did])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py", line 174, in decorated
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py", line 449, in read_data_by_identifier
    response = self.send_request(req)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py", line 2181, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' %
udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)_

Manchmal bekomme ich auch ein richtiges Feedback (selten, z.B. Busidentifier wenn ich Vdens wähle). 
Befehl python3 Open3Eclient.py -c can0 -dev vdens -r 256 -v . Dann bekomme ich den Wert und im Candump ist 680 und 690. Aber nur manchmal. Hier bekomme ich auch oft dasselbe Timeout:

_[ERROR] UdsClient: [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)
Traceback (most recent call last):
  File "/home/ubuntu/open3e_org/Open3Eclient.py", line 111, in <module>
    response = client.read_data_by_identifier([did])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py", line 174, in decorated
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py", line 449, in read_data_by_identifier
    response = self.send_request(req)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py", line 2181, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' %
udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)_

philippoo66, 2023-10-10:
ok, langsam kommen wir der Sache näher glaube ich. 

du hast also inzwischen 2 Terminal Fenster offen, wo in dem einen der candump läuft und du in dem anderen den Open3EClient aufrufst, richtig?

Die Sache mit dem Filter bei candump kennst du auch, oder? 
`<can_id>:<can_mask> (matches when <received_can_id> & mask == can_id & mask)`
wenn du also 600:700 schreibst, bekommst du nur Frames, bei denen der COB-ID 6yz ist, also alle, die mit 6 anfangen. das ist hilfreicht für unsere Betrachtungen.

Als nächstes muss ich mal fragen, was du überhaupt für eine Gerätekonstellation hast. Wenn du mit einer VX3 reden möchtest, ist es nämlich ein Unterschied, ob die VX3 alleine ist (ggf. mit einem Energiezähler) oder ob sie mit einer Vitocal verbunden ist. Das Stichwort heisst hier "Main Control Unit" und das habe ich [hier beschrieben](https://github.com/abnoname/open3e/issues/11#issuecomment-1752994328). 

Dann gibt es noch zwei weitere 'Hürden':

- Die erste ist, dass es Konstellationen gibt, wo das angesporchene Gerät nur antwortet (also ein 690 Frame zurück kommt), wenn man einen CAN Frame mit 8 Bytes dahin schickt (Data Length Code (DLC) = 8). Der DLC ist das, was du in den eckigen Klammern im Dump siehst.

Wir haben noch nicht abschliessend klären können, womit das genau zusammenhängt, aber die sicherere Sache ist wohl, immer 8 Byte Frames zu schicken. Um das mit Open3Eclient sicherzustellen, modifizierst du die Zeile 67 folgendermassen:

[image]
 
Um das bei einem cansend zu erreichen, schreibst du 
`cansend can0 680#0322068000000000` statt `cansend can0 680#03220680`

- Die zweite mögliche Hürde ist das Timeout Issue, welches auch irgendwie mit irgendwelchen Konstellationen (hier eher das 'System Raspi und Adapter') zusammenhängt. Wenn dieses Problem auftritt, bekommst du zwar von Open3E (bzw udsoncan bzw isotp) eine Timeout Exception angezeigt, auf dem Bus (im Dump) sieht man aber den 680 Frame und den folgenden 690 Frame ganz korrekt.

Also: Wir müssen klären, ob a) die VX3 bei dir die Main Control Unit ist und überhaupt mit 680 angesprochen werden kann, und b) ob der 690 Frame (nur) nicht kommt, weil der 680 Request nicht mit 8 Bytes geschickt wurde, und c) ob bei dir möglicherweise diese 'Library Problematik' zuschlägt. 

Weisst du, wie du dafür jetzt weiter vorgehst?

achgut, 2023-10-10:
Vielen Dank für die ausführliche Beschreibung. Werde ich mir am Wochenende mal anschauen und probieren.

philippoo66, 2023-10-10:
dann passt die 680/690 ja zumindest. dann mach das noch mit den immer 8 Bytes und dann sollte `Open3Eclient.py -c can0 -dev vx3 ...` hoffentlich funktionieren.

achgut, 2023-10-11:
Perfekt. Mit dem Padding auf 8 Bytes funktioniert es bei mir jetzt auch.
Vielen Dank für die Hilfe.

philippoo66, 2023-10-11:
top! danke für die Rückmeldung!

Sebliner, 2023-10-11:
Ich klinke mich nochmal ein - war bisher stiller Mitleser mit gleichem Problem und bei @achgut scheint es ja gelöst zu sein, bei mir leider nicht. Vielleich habt ihr noch Unterstützung.

Bei mir funktioniert es bei einzelnen Abfragen meistens, allerdings gibt immer irgendwann auch mal einen udsclient Timeout error. Mit -a -v sieht es dann so aus:
[image]
Dabei ist der Error zeitlich immer rein zufällig - manchmal kommt der error direkt, manchmal nach ein paar Einträgen, teils auch erst nach einer langen Liste. 

Bei einer einzelnen Abfrage dann gerne mal so:
[image]

Frage ich mehrere Werte gleichzeitig ab kommt auch fast immer irgendwann ein timeout.

Hardware: VX3 + e380 Energiemeter, nichts weiter. Adapter ist der aus Hotziplotzis Leitfaden. Als System nutze ich eine Proxmox VM mit Debian 12

Padding ist schon entsprechend angepasst, Adapter ebenfalls released und neu gestartet.
Leider habe ich (offensichtlich...) auch keine Ahnung von CAN-Bus und mich mit candump noch nicht weiter beschäftigt. Wäre das für die weitere Analyse der nächste notwendige Schritt?

Am Ende möchte ich die Werte gerne über mqtt in meinen iobroker importieren. Aktuell funktioniert das immer nur einen kurzen Zeitraum, bricht dann aber irgendwann ab. Ich müsste nochmal genau schauen, gehe aber davon aus dass irgendwann einmal ein timeout kommt und so der Prozess gekillt wird?

Danke für eure Unterstützung und dieses absolut großartige Projekt!!

philippoo66, 2023-10-11:
moin!

> Wäre das für die weitere Analyse der nächste notwendige Schritt?

ja, auf jeden Fall. das ist ganz einfach. Starte in einem Terminal Fenster 
`candump can0, 600:700`
und dann mach in einem anderen Fenster die open3e Aktionen. 

Wie gesagt, so eine UDS Kommunikation geht immer so:
- ein Request Frame mit COB-ID 680h wird 'von open3e' auf den Bus geschickt
- ein Antwort Frame mit COB-ID 690h wird auf den Bus geschickt.
Bei kurzen Daten war es das dann schon, bei längeren Daten kommt dann
- ein Flow Control Frame (Daten 30 00 00) mit COB-ID 680h
- ein/mehrere Frame/s mit COB-ID 690h mit den restlichen Daten 

Dieses PingPong Spiel muss halt klappen. Wenn das klappt, und du trotzdem den Timeout bekommst, bist du wahrscheinlich ein Opfer der Hürde #2 von oben.

Wenn du dir nicht sicher bist (oder zur Sicherheit sowieso) kopier bitte den Dump hier rein. Dann schauen wir weiter ;-)

philippoo66, 2023-10-11:
ich hab das aber mal grad mit der Fehlerausgabe auf einem andren System verglichen, die genau so aussieht. Wahrscheinlich bekommt isotp im Fehlerfall den initialen 680 Frame garnicht auf den Bus abgesetzt. Wir arbeiten da grad dran...

Aber trotzdem - schau mal und poste bitte.

Sebliner, 2023-10-12:
Vielen Dank für die superschnelle Antwort!
Gerade einmal mit candump folgendes Ergebnis bekommen:
[image]

Ich hatte dabei eine Anfrage mit 4 gleichzeitig abgefragten Datenpunkten gestellt. Die Anfrage ging komplett durch und alle Werte wurden ausgegeben (ich vermute daher auch jeweils 4x 680/690 im candump). Direkt im Anschluss hatte ich dieselbe Abfrage nochmals gestellt, dort kam allerdings direkt ein Timeout, keine Werte-Ausgabe und im candump kamen auch keine weiteren 680/690 Einträge.
Einmal kam bei weiteren Tests der Timeout nach 2 korrekt ausgegebenen Werten, da waren dann im candump auch genau 2x 680/690. Ich gehe also davon aus dass nicht immer der 680er frame auf den Bus gesetzt wird und dadurch der Timeout zustande kommt?
Außerdem kommt ganz regelmäßig der 661er frame. Ich vermute das gehört so?

Bei -a -v sieht es im candump dann (bis zum timeout error, dann kommt gar nichts mehr) so aus:
[image]

Ich hoffe damit könnt ihr etwas anfangen?

philippoo66, 2023-10-12:
das was ich sehe sieht alles korrekt aus. Und wenn du sagst, wenn der Timeout error kommt, kommt nix mehr, dann haben wir es tatsächlich mit der "3. Hürde" zu tun. Um wirklich ganz sicher zu sein, müsstest du Einzelabfragen machen und den Busverkehr dann zuordnen können. Am besten, in dem du dir den timestamp auch noch ausgeben lässt.

candump -tA can0, 680:FFF, 690:FFF

Mit dem Filter bekommst du wirklich nur noch 680 und 690 Frames, die 661 sind was andres, die interessieren uns nicht.

Wie gesagt, dieses Problem versuchen wir gerade zu lösen oder zumindest zu klären. Das liegt ja irgendwo in den 'tieferen Schichten'. Eventuell ist es sogar Hardware-bedingt, möglicherweise ist es ein Potentialproblem beim Bus. 

Hast du eine galvanisch isolierten CAN Adapter? 
Hast du Gnd auch verbunden oder nur CAN_Hi, CAN_Lo?

Sebliner, 2023-10-12:
Habe gerade über regelmäßige Einzelabfragen (-t 5) und timestamp definitiv geklärt dass es tatsächlich so ist: alle 5 Sekunden kamen passend die beiden frames, genau bei der Abfrage mit errors kam kein 680er (und entsprechend natürlich auch kein 690er) frame mehr über den Bus.

Ich habe den CAN Adapter nur mit CAN_Hi und CAN_Lo verbunden. VX3 und E380 sind ja ebenfalls nur mit den beiden verbunden, daher war ich davon ausgegangen dass es beim Adapter ebenfalls so passt?

Falls es helfen sollte könnte ich die Can-Abfrage auch einmal auf meinen pi 4 umziehen um zu schauen ob das Problem gleich bleibt. Da bräuchte ich aber ein bisschen Zeit zum raussuchen und einrichten... Wenn es aber sowieso nicht am System liegen kann vermeide ich das gerne. Nicht zuletzt weil ich den Pi zumindest dauerhaft nicht als zusätzlichen Verbraucher laufen lassen möchte. Der permamente absurd hohe Eigenverbrauch vom VX3 ist schon brutal genug...

philippoo66, 2023-10-12:
einfacher wäre zunächst, wenn du testhalber den Gnd Draht grade verbinden würdest.

Der isolierte Adapter ist ja theoretisch floatend, und vielleicht sind durch ein Schaltnetzteil irgendwelche (ggf hochfrquenten) Potentiale da, und so ein CAN Transceiver kann da nur bis 30..35 Volt mit umgehen meine ich zu erinnern (der geht dabei nicht kaputt, kann aber den Unterschied zwischen hi und lo nicht mehr erkennen, und vielleicht merkt er, dass sein dominanter Pegel nicht als dominant auf dem Bus auftaucht). Wir haben in industrieller Umgebung den Gnd immer mitgeführt und es wird auch empfohlen (kannst ja mal CAN Ground googeln).

philippoo66, 2023-10-12:
ChatGPT sagt übrigens:

Wenn der Fehler in der Zeile `self._socket.send(*args, **kwargs)` auftritt und die Konfiguration als in Ordnung angesehen wird, kann dies auf ein tiefer liegendes Problem im Zusammenhang mit der Netzwerkkommunikation oder dem zugrunde liegenden Betriebssystem hinweisen. Hier sind einige mögliche Ursachen und Schritte zur Fehlersuche:

1. **Ressourcenmangel:** Es ist möglich, dass Ihr System zu diesem Zeitpunkt knappe Ressourcen hat, die die Netzwerkkommunikation beeinträchtigen. Überprüfen Sie den Systemmonitor, um sicherzustellen, dass ausreichend CPU, Speicher und Netzwerkbandbreite verfügbar sind.

2. **Betriebssystemprobleme:** Probleme auf Betriebssystemebene können zu Kommunikationsproblemen führen. Überprüfen Sie, ob Ihr Betriebssystem und die relevanten Netzwerktreiber auf dem neuesten Stand sind.

3. **CAN-Bus-Probleme:** Stellen Sie sicher, dass der CAN-Bus selbst ordnungsgemäß funktioniert und keine Störungen oder Fehler im Bus auftreten. Dies kann durch Überwachung der Busaktivität und Überprüfung der Kabel und Verbindungen erfolgen.

4. **Fehlerbehandlung:** Implementieren Sie eine angemessene Fehlerbehandlung in Ihrer Software, um mit Timeout-Fehlern umzugehen. Sie könnten in Betracht ziehen, den Sendeversuch mehrmals zu wiederholen, bevor Sie den Timeout-Fehler auslösen.

5. **Protokollierung:** Erweitern Sie die Protokollierung, um detaillierte Informationen über den Zeitpunkt des Fehlers und den Status des Netzwerks oder des CAN-Busses bei jedem Fehler zu erhalten. Dies kann Ihnen helfen, Muster oder Auslöser zu identifizieren.

6. **Threading und Parallelität:** Wenn Ihre Anwendung mehrere Threads oder Prozesse verwendet, um auf den CAN-Bus zuzugreifen, könnte es zu Wettbewerb um den Zugriff auf den Bus kommen. Stellen Sie sicher, dass die parallele Ausführung ordnungsgemäß synchronisiert ist.

7. **Drittanbieter-Bibliotheken:** Überprüfen Sie, ob es bekannte Probleme oder Inkompatibilitäten mit den von Ihnen verwendeten Drittanbieter-Bibliotheken (wie `udsoncan` und `isotp`) gibt. Möglicherweise sind Aktualisierungen oder Patches verfügbar, um solche Probleme zu beheben.

Die Lösung dieses Problems erfordert möglicherweise eine gründliche Fehlersuche, um die genaue Ursache zu ermitteln. Wenn Sie alle möglichen Fehlerquellen sorgfältig überprüfen und keine offensichtlichen Probleme finden, könnte es ratsam sein, sich an den Support der verwendeten Bibliotheken oder an Experten für die spezifische Hardware und Software zu wenden, um weitere Unterstützung zu erhalten.

philippoo66, 2023-10-12:
Wir haben allerdings bei dem anderen System einen Raspi frisch aufgesetzt (mit eigentlich alle Software/Treiber aktuell) und den Bus Gnd verbunden und es hat trotzdem nix genutzt :-/

Sebliner, 2023-10-12:
Ressourcen dürften bei mir auch kein Problem sein. Die VM ist auch erst vor 2-3 Wochen aufgesetzt und entsprechend sollte Software/Treiber zumindest aktuell sein.
Ich werde dann mal versuchen den GND zu verbinden. Zur Sicherheit für mich zum Verständnis: Weil der E380 als mittlere Komponente keinen GND hat verbinde ich VX3 GND direkt mit Adapter GND?
Und Schirmung hatte ich bisher einfach nirgendwo aufgelegt - sollte ich das dann noch tun? Dachte CAN Bis wäre relativ unempfindlich 😅

philippoo66, 2023-10-12:
jap, schliess den Adapter ruhig auf der VX3 an und dann 3-Draht. An E380 hi/lo und von VX3 den Gnd holen sieht etwas seltsam aus (sollte aber keinen Unterschied machen). Die Sache mut Abschulsswiderstand an beiden Enden des Busses weisst du ja und hast auch richtig gemacht? Mit dem Schirm hast du wohl recht, eigentlich braucht CAN den nicht, und den nicht auflegen hab ich auch schon gelesen. Was immer gesagt wird, ist, dass man CAN_hi und CAN_lo verdrillen soll. Aber wie gesagt, das ist bei dem andren System mit diesem Fehler alles schon gemacht. Ich denke die Ursahe liegt irgendwo anders...

Sebliner, 2023-10-12:
Also gerade habe ich entsprechend neu verdrahtet und nun auch GND jeweils angeschlossen, das Fehlerbild bleibt aber leider identisch. Immer wenn error ausgegeben wird kommen weder 680er noch 690er frames im candump an.
can hi und lo sind ziemlich verdrillt - hab ein cat 8 Verlegekabel genutzt welches ich noch liegen hatte. Daran sollte es also auch nicht liegen.
Widerstände habe ich jeweils beim Bus-Ende gesetzt, das sollte also auch passen.

Kann ich bei der Fehlersuche noch irgendwie input geben? Ich könnte über Proxmox natürlich auch verschiedene VMs konfigurieren wenn das irgendwie helfen würde.

Danke auch jeden Fall für die tolle Unterstützung!

philippoo66, 2023-10-13:
Wenn der Fehler gekommen ist - kann man dann eine andere Anfrage schicken oder nach eine kurzen Zeit die gleiche noch mal, oder muss man dann erst CAN down/up machen?

Ein Work-arround könnte ja sein, den Fehler abzufangen und zu behandeln und dann noch 1 oder 2 Retrys machen, wie man das gewöhnlich in 'störsicheren' Kommunikationen macht. Wir haben ja noch garkeine Fehlerbehandlung eingebaut.

Schöner wär natürlich die Ursache zu klären und abzustellen, aber aktuell mangelt es mir da an Ideen, auch zur weiteren Vorgehensweise. Wie gesagt, wir arbeiten da auch dran, vielleicht haben ja andre Leute aus der Gruppe bessere/mehr Ideen als ich. Wenn dir noch was einfällt, mach das / sagt Bescheid! Vielleicht auch noch mit den isotp Leuten in Kontakt treten...

Sebliner, 2023-10-13:
Man kann direkt eine neue Anfrage (erfolgreich) schicken ohne CAN down-up. Also zumindest nach der Zeit die ich brauche um up+enter zu drücken. Ich werde am Abend mal möglichst schnell versuchen 😅

Im Moment scheitert die Abfrage mehrerer Datenpunkte daran, dass mit dem error die komplette (auch regelmäßige) Abfrage beendet wird. Und mqtt publishing daran dass auch bei einzelnen Datenpunkten die regelmäßigen Abfragen nach einem error beendet werden. Insofern wäre ein workaround im Sinne "if error redo after x ms" relativ einfach umzusetzen, oder?

Hatte auch schon überlegt mir ein Script zu schreiben welche mehrere Datenpunkten einzeln nacheinander über mqtt published, aber eine Integration in open3e wäre natürlich wesentlich eleganter und für eventuell weitere User mit dem Problem sehr hilfreich.

philippoo66, 2023-10-13:
ich werde das mal ankurben. eigentlich muss man das client.read_data_by_identifier([did]) ja nur in eine Sub packen, die eienn Error Handler hat und im Fehlerfall kurz wartet und das bis zu 2x wiederholt. Leider bin ich totaler Python Anfänger. Muss man nicht, wenn man eine Sub da rein tut, auch eine 'Main' haben und ein if __name__ == "__main__": usw?

Sebliner, 2023-10-15:
Ich bin leider völlig unwissend bzgl. python, deswegen kann leider überhaupt nichts zur Programmierung beisteuern. Leider ist das einzige was ich anbieten kann ausgiebiges testen. Ich wäre gerne hilfreicher...

philippoo66, 2023-10-15:
wenn ich/wir uns an die isotp Leute wenden, wäre gut wenn du deine Systemkonfiguration und so aufschreiben könntest. irgendwas müssen wir denen ja liefern, wenn wir sie bitten... blöd ist halt, dass das nicht zuverlässig reproduzierbar ist, sondern irgendwie zufällig auftritt.

Sebliner, 2023-10-16:
Systemkonfiguration ist folgende:

Zotac ci327 MiniPC (Celeron N3450, 16GB RAM, SSD)
Darauf Proxmox. Open3e läuft dort in einer VM mit 4GB RAM, 4 CPU Cores. System ist Debian 12. Zusätzlich läuft auf dieser (gerade erst aufgesetzten) VM noch mein iobroker, sonst nichts.
USB-Geräte sind der erwähnte CAN-USB converter aus der Anleitung [https://www.amazon.de/Konverter-Raspberry-Computer-Support-Windows/dp/B07Q812QK8/ref=sr_1_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=1OKT4C076KPSL&keywords=USB+zu+CAN+Konverter+Modul+for+Raspberry+Pi4%2FPi3B%2B%2FPi3%2FPi&qid=1697489162&s=ce-de&sprefix=usb+zu+can+konverter+modul+for+raspberry+pi4%2Fpi3b%2B%2Fpi3%2Fpi%2Celectronics%2C92&sr=1-1-catcorr](url) , ein nano-cul und ein zigbee adapter.

Geräte der VX3 und der E380 Energiezähler, sonst nichts.

BUS Kabel wie erwähnt ein CAT8 Verlegekabel. Insgesamt komplett etwa 12m Länge.

Habe ich etwas vergessen was noch relevant wäre?

Danke!

philippoo66, 2023-10-16:
vielleicht noch die Versionen von udsoncan und can-isotp (evtl. noch von doipclient, aber ich meine, der ist bei CAN nicht 'in Gebrauch')?

Sebliner, 2023-10-18:
Hier der entsprechende output:

_root@Broker:/# pip show udsoncan
Name: udsoncan
Version: 1.20.1
Summary: Implementation of the Unified Diagnostic Service (UDS) protocol (ISO-14229) used in the automotive industry.
Home-page: https://github.com/pylessard/python-udsoncan
Author: Pier-Yves Lessard
Author-email: py.lessard@gmail.com
License: MIT
Location: /usr/local/lib/python3.11/dist-packages
Requires:
Required-by:

root@Broker:/# pip show can-isotp
Name: can-isotp
Version: 1.9
Summary: Module enabling the IsoTP protocol defined by ISO-15765
Home-page: https://github.com/pylessard/python-can-isotp
Author: Pier-Yves Lessard
Author-email: py.lessard@gmail.com
License: MIT
Location: /usr/local/lib/python3.11/dist-packages
Requires:
Required-by:

root@Broker:/# pip show doipclient
Name: doipclient
Version: 1.1.1
Summary: A Diagnostic over IP (DoIP) client implementing ISO-13400-2.
Home-page: https://github.com/jacobschaer/python-doipclient
Author: Jacob Schaer
Author-email:
License:
Location: /usr/local/lib/python3.11/dist-packages
Requires:
Required-by:

root@Broker:/# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

root@Broker:/# uname -a
Linux Broker 6.1.0-12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.52-1 (2023-09-07) x86_64 GNU/Linux_

Passt das so?
Viele Grüße

philippoo66, 2023-10-18:
top! das sieht gut aus - danke!!

Sebliner, 2023-10-20:
Nur zur Info - sofern ich mit meinem gruseligen Englisch alles richtig verstanden habe wird der Fehler ja durch einen bug im Linux Kernel hervorgerufen. Habe gerade erst bemerkt dass es seit so 10 Tagen Debian 12.2 inklusive Kernel update (6.1.0-13) gibt. Dieses gerade geupgraded aber der Fix ist dort offensichtlich noch nicht dabei, zumindest tritt der Fehler weiterhin auf.

philippoo66, 2023-10-20:
Aktuell besteht die Lösung wohl eher in einem 'downgrade' ;-) ich habe Buster intalliert, damit läuft es problemlos :-)

philippoo66, 2023-10-20:
wir haben in irgendeinem altuellen develop branch aber auch ein 'ignore-on-error' eingebaut, damit bricht das Skript dann nicht ab, sondern geht zum nächsten DID (wenn es mehrere abarbeiten soll bzw. zyklisch läuft).

Sebliner, 2023-10-21:
Gerade mal fix eine vm mit Buster erstellt, da kommt dann als Antwort "OSError: [Errno 93] Protocol not supported" 🤦
Aber toll dass ihr so super schnell den workaround eingebaut habt, dann werde ich diesen branch nehmen und es auf der bisherigen vm lassen. So spare ich mir mich mit meinem neuen error unter Buster zu beschäftigen...

Vielen vielen Dank für die tolle Unterstützung!!

philippoo66, 2023-10-21:
den Protocol not supported hatte ich mit Buster frisch auch, aber nach Update/Upgrade (dauert...) war er weg

Sebliner, 2023-10-21:
> den Protocol not supported hatte ich mit Buster frisch auch, aber nach Update/Upgrade (dauert...) war er weg

🤦 Das wäre so naheliegend gewesen das vorher zu machen... Danke!

Wahrscheinlich lasse ich open3e Dank dem workaround nun aber sowieso auf der bisherigen bookworm VM - RAM ist knapp bei mir und sofern ich den branch richtig verstehe müssen lediglich 3 Zeilen für Python Try Except eingefügt werden. Ich hoffe morgen komme ich dazu.

Wahnsinn wie schnell und großartig sich dieses Projekt hier weiterentwickelt!!

achgut, 2023-10-25:
Hallo
Der MQTT Client läuft bei mir nach dem Error Handling gut. Es gibt immer wieder Fehlermeldungen, das das Programm läuft weiter.

`python3 Open3Eclient.py -c can0 -dev vx3 -r 1664,1836,1831 -m localhost:1883:open3e -t 5`
2023-10-24 23:18:01 [ERROR] UdsClient: [TimeoutError] : timed out
2023-10-24 23:19:09 [ERROR] UdsClient: [TimeoutError] : timed out
2023-10-24 23:21:56 [ERROR] UdsClient: [TimeoutError] : timed out
2023-10-24 23:30:50 [ERROR] UdsClient: [TimeoutError] : timed out

Aber nach immer ca. 1 Tag kommt doch der Abbruch. 
@jwende Gibt es hier vielleicht auch noch eine Möglichkeit, den Abbruch zu vermeiden?

Hier ist die Fehlermeldung:
2023-10-25 00:17:28 [ERROR] UdsClient: [TimeoutError] : timed out
2023-10-25 00:29:39 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)
Traceback (most recent call last):
  File "/home/ubuntu/open3e/Open3Eclient.py", line 242, in <module>
    readByDid(eval(did), client, client_mqtt, mqttParamas, dataIdentifiers)
  File "/home/ubuntu/open3e/Open3Eclient.py", line 115, in readByDid
    response = client.read_data_by_identifier([did])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py", line 174, in decorated
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py", line 449, in read_data_by_identifier
    response = self.send_request(req)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py", line 2181, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' %
udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)

jwende, 2023-10-25:
@achgut es wird zu nur der timeoutError abgefangen - nicht timeoutException. Ggf. liegt hier das Problem aber tiefer … oder wir treffen hier einen ‚Summenfehler‘ - der sich nur durch einen reconnect des Clients zurücksetzen lässt.

MyHomeMyData, 2023-10-25:
@achgut 
Du kannst probieren, auch diesen Fehler zu ignorieren. Wenn Du `except timeoutError:' erweiterst, also `except timeoutError,timeoutException:`, sollte das Programm weiterlaufen. Vielleich geht auch die Kommunikation weiter oder es kommt nach 20 Sekunden wieder timeoutException.

achgut, 2023-10-26:
Danke.  Probiere ich mal am Wochenende. Die letzten 2 Tage ist es ohne Abbruch durchgelaufen

philippoo66, 2023-10-26:
hast du irgend n automatisches Update? vielleicht migriert der gefixte Kernel ja langsam auf die Geräte...

achgut, 2023-10-26:
Automatisch nicht. Hole mir aber immer wieder die neuste Version

achgut, 2023-10-27:
Derzeit läuft es weiter. Im Udsoncan.log habe ich auch 2 TimeoutException Fehler gefunden, aber Programm hat nicht abgebrochen. Die Änderung von @MyHomeMyData ist aber nicht im File geändert. Werde ich weiter beobachten.

````ruby
def readByDid(did, client, client_mqtt, mqttParamas, dataIdentifiers):
    try:
        response = client.read_data_by_identifier([did])
    except TimeoutError:
        return

GuidoN0207, 2023-10-28:
Auch ich habe das Problem mit den timeouts und möchte meine Erfahrungen hier teilen.

Eigentlich hatte ich vor, den Open3Eclient auf einer VM auf meinem Proxmox Server (V 8.0.4) laufen zu lassen um meine Vitodens 333-F abzufragen.

Ich habe die Debian Releases Buster 10.13.0, Bullseye 11.7.0 und Bookworm 12.0.0 für die VM’s ausprobiert, bei keiner der Versionen läuft die Abfrage stabil, entweder geht es gar nicht, oder bei Abfragen von größer als einem Wert gibt es timeouts.
Der Proxmox Server ist ein Intel NUC8i3BEK2

Dann habe ich meine alten Raspi’s rausgeholt und darauf Raspbian 11 installiert und was soll ich sagen, es gab keine Probleme mit timeouts.
Hier meine getesteten Konfigurationen, Raspberry PI 2 Model B & Raspberry Pi Zero W :
Als USB-
$ cat /sys/firmware/devicetree/base/model
Raspberry Pi 2 Model B Rev 1.1

$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

$ python -V
Python 3.9.2
$ pip3 --version
pip 20.3.4 from /usr/lib/python3/dist-packages/pip (python 3.9)
$ cat /sys/firmware/devicetree/base/model
Raspberry Pi Zero W Rev 1.1

$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
guido@Zero:~/open3e $ uname -a
Linux Zero 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

$ python3 -V
Python 3.9.2
guido@Zero:~/open3e $ pip3 --version
pip 20.3.4 from /usr/lib/python3/dist-packages/pip (python 3.9)

Evtl. hilft es ja den einem oder anderen...

philippoo66, 2023-10-28:
Hallo Guido,

danke für die Info! Ich bin ja mit meinem Buster auf einem Raspi 3 auch glücklich (problemlos). 

Ich weiss nicht, ob du dir die Mühe gemacht hast, es alles durchzulesen, dewegen noch mal kurz zusammengefasst:

Seit etwa April hat es Kernel Updates gegeben, die dazu führen, dass isotp nicht mehr richtig funktioniert. Das Problem ist erkannt und behoben, nun muss die entsprechende Kernel Version aber bei allen Distributionen zu den Benutzern 'durchmigrieren' - das kann etwas dauern... 

Deshalb unternehmen wir 'nix weiter', weil sich das Problem mit der Zeit von selber lösen wird. Ich weiss, das ist aktuell nicht schön, aber Mehrheitsbeschluss.

Wie auch bei dir sind zur Zeit alte Systeme, die nicht diese Kernel Updates vom April abbekommen haben, klar im Vorteil. Wenn du mit einem neueren System arbeiten möchtest, ist [hier die Lösung](https://github.com/pylessard/python-can-isotp/issues/93#issuecomment-1777833070) dafür beschrieben. Sie besteht in einem 'out-of-tree isotp driver', der den Fix beinhaltet. Soll ganz einfach sein - wahrscheinlich wenn man sich mit sowas auskennt. Ich bleib einfach bei Buster. _"Never touch a running System"_ :-)) 

beste Grüsse!
Phil

jwende, 2023-11-13:
folgender Versuch war bei mir erfolgreich:
- RPi4 mit der aktuellen Version (no GUI) vorbereitet
- Installation entsprechend readme
- es treten timeouts auf
- sudo rpi-update next entsprechend https://forums.raspberrypi.com/viewtopic.php?t=346969
- reboot
- keine timeouts mehr

uname -a
Linux rpi4viessmann 6.1.10-v8+ #1628 SMP PREEMPT Mon Feb  6 19:22:54 GMT 2023 aarch64 GNU/Linux

Beste Grüsse
Jörg

Thyler21, 2024-04-02:
Nach vielen Versuchen brachte folgendes den Erfolg:

1. Image: Linux Open3e 6.6.16+ #1728 Tue Feb  6 17:06:16 GMT 2024 armv6l Bullseye
2. Kernelupdate wie durch @jwende schon beschrieben:
 sudo rpi-update next entsprechend https://forums.raspberrypi.com/viewtopic.php?t=346969
4. Installation von open3e wie in der Readme beschrieben
5. statt "devices.json wurde wieder -dev in der args eingesetzt"
 `-c
can0
-r
1664,1690,1810,1831,1834,1836,2184,2214
-t
5
-m
IP or LOCALHOST MQTTBROKER:1883:open3e
-muser
MQTTUSER:MQTTPASSWORD
-mfstr
{didNumber}_{didName}
-dev
vx3`
6. Den CAN-Adapter und das CAN Interface für den Autostart genau nach [WIKI](https://github.com/open3e/open3e/wiki/020-Inbetriebnahme-CAN-Adapter-am-Raspberry) sowie Candump installiert.
Aktuell ist bei mir nur der VX3 mit dem Energiezähler in der CAN Topologie verkabelt - entgegen der VI Experten, welche sagen, dass der externe CAN-Bus nur mit Hi und Lo verkabelt wird, hab ich wie hier empfohlen auch den GND mit angeschloßen. 
7. Autostart des Scriptes wie in der [WIKI ](https://github.com/open3e/open3e/wiki/030-Installation-und-Inbetriebnahme-von-open3E) beschrieben.

Et voilà ! - keine Verbindungsabbrüche mehr

philippoo66, 2024-04-03:
moin @Thyler21 

das freut mich auf jeden Fall schon mal.

Wir haben natürlich inzwischen auch auf "full-python ISO-TP" umgestellt, so dass die Kernel-Problematik von früher eh schon mal 'umgangen' ist.

Zu Gnd anschliessen oder nicht muss man immer das Gesamtsystem betrachten: 
- Wenn es irgendetwas gibt (und sei es PE oder N), was die CAN Transceiver halbwegs an ein gemeinsames Potential anbindet, ist es empfehlenswert, den CAN Gnd nicht anzuschiessen, weil man dadurch ggf. zusätzliche Schleifen bauen würde, in die Störungen einstrahlen können.
- Wenn ein CAN Transceiver optisch isoliert ist (wie der auf den Inno-Maker Adaptern) und entsprechend völlig ohne Potentialbezug dasteht, ist es dringend geboten, den CAN Gnd mit anzuschliessen, um ein Wegfloaten und damit Klippen der Empfangsstufen zu vermeiden.

Die VI Experten haben also nicht pauschal Unrecht... ;-)

Grüsse!
Phil

Thyler21, 2024-04-04:
Gibt es eigentlich eine Lösung, wenn das Timeout-Problem durch das CAN-Netzwerk bzw. CAN Adapter verursacht wird?
Ich hab das ganze jetzt auf meinen Loxberry gelegt (damit nicht 2 RPI laufen müssen) und nun verabschiedet sich der CAN nach einiger zeit:

`
loxberry@loxberry:~ $ sudo systemctl status systemd-networkd
● systemd-networkd.service - Network Service
     Loaded: loaded (/lib/systemd/system/systemd-networkd.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2024-04-03 22:11:58 CEST; 9h ago
TriggeredBy: ● systemd-networkd.socket
       Docs: man:systemd-networkd.service(8)
   Main PID: 452 (systemd-network)
     Status: "Processing requests..."
      Tasks: 1 (limit: 1065)
        CPU: 1.640s
     CGroup: /system.slice/systemd-networkd.service
             └─452 /lib/systemd/systemd-networkd

Apr 03 23:27:12 loxberry systemd-networkd[452]: can0: Link DOWN
Apr 03 23:27:12 loxberry systemd-networkd[452]: can0: Lost carrier
Apr 03 23:50:15 loxberry systemd-networkd[452]: can0: Link UP
Apr 03 23:50:15 loxberry systemd-networkd[452]: can0: Gained carrier
Apr 03 23:50:15 loxberry systemd-networkd[452]: can0: Cannot configure proxy NDP for the interface, ignoring: No such file or directory
Warning: journal has been rotated since unit was started, output may be incomplete.
`

nach einem reboot läuft es wieder wie folgt:

`
loxberry@loxberry:~ sudo systemctl status systemd-networkd
● systemd-networkd.service - Network Service
     Loaded: loaded (/lib/systemd/system/systemd-networkd.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2024-04-04 07:18:36 CEST; 29min ago
TriggeredBy: ● systemd-networkd.socket
       Docs: man:systemd-networkd.service(8)
   Main PID: 470 (systemd-network)
     Status: "Processing requests..."
      Tasks: 1 (limit: 1065)
        CPU: 441ms
     CGroup: /system.slice/systemd-networkd.service
             └─470 /lib/systemd/systemd-networkd

Apr 04 07:18:35 loxberry systemd[1]: Starting Network Service...
Apr 04 07:18:36 loxberry systemd-networkd[470]: Enumeration completed
Apr 04 07:18:36 loxberry systemd-networkd[470]: can0: Link UP
Apr 04 07:18:36 loxberry systemd-networkd[470]: can0: Gained carrier
Apr 04 07:18:36 loxberry systemd-networkd[470]: can0: Cannot configure proxy NDP for the interface, ignoring: No such file or directory
Apr 04 07:18:36 loxberry systemd[1]: Started Network Service.
Apr 04 07:18:37 loxberry systemd-networkd[470]: eth0: Link UP
Apr 04 07:18:37 loxberry systemd-networkd[470]: eth0: Gained carrier
Apr 04 07:18:52 loxberry systemd-networkd[470]: eth0: Gained IPv6LL
`

philippoo66, 2024-04-14:
leider kenne ich mich da nicht genug mit aus - es gibt wahrscheinlich eine Möglichkeit, den Link zu überwachen und bei Bedarf automatisch neu zu starten. Aber wie gesagt, dazu reichen meine Linux Fähigkeiten nicht. Hast du schon mal ChatGPT gefragt?

leitwolf00, 2024-04-19:
Hallo,

die Datenübernahme per open3e-Client läuft bzw. lief bei mir seit 2 Monaten ohne Problem (can0). Seit dem 17.04. erhalte ich jetzt einen Fehler (siehe Fehlerbeschreibung). Der Vitocharge liefert mir ebenfalls einen Fehlercode

Fehlerbeschreibung:

Script zur Datenübernahme:

$ python3 /home/leitwolfpi/open3e/Open3Eclient.py -c can0 -dev vx3 -r 506,1664,1801,1802,1831,1833,1834,1836,1839,2240 -m 192.168.178.100:1883:/ESP8266 -muser mosquitto1:xxxxxxxx -t 15 *)

2024-04-17 15:56:12 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)
Traceback ..

/Fehlerbeschreibung

 
*)  das Script war seit etwa 3 Wochen aktiv
 

Anlage Fehlermeldung:

**Aktive Status Meldungen: Aktiver Fehler: F.1034 - Kommunikationsstörung Externer CAN-BUS**

Der Vitocharge blieb in diesem Fehlerzusatnd und konnte nur durch Neustartwieder zurückgesetzt werden.
Nach nochmaligem Aktivieren des open3e-Client trat der Fehler wieder auf. Ich habe anschließend auch versucht, einzelne Werte ohne MQTT abzurufen. Immer die gleiche Fehlermeldungen wie oben beschrieben.
Nach Zurücksetzen des Vitocharge und Trennung von dem USB_zu_CAN Konverter (Amazon https://www.amazon.de/dp/B07Q812QK8?ref=ppx_yo2ov_dt_b_product_details&th=1) läuft der Vitocharge wieder fehlerfrei. Der Konverter hängt an einem Raspberrypi 3 (Linux leitwolfrpi 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux).

Zeitgleich läuft eine Anwendung zur Datenübernahme von einer Stiebel WP (can1). Auch nach einer Beendigung dieser Datenübernahme über can1 erfolgte keine Kommunikation mit dem Vitocharge über can0 mehr. Da das Datenaufkommen auf ca. 40% des Wertes reduziert wurde, der bei Aktivierung von can0 und can1 anfiel, dürfte die anfallende Datenmenge meiner Meinung nach nicht der ausschlaggebende Grund des Problems sein.

leitwolfpi@leitwolfrpi:~/open3e $ _ip -details -statistic link show can0_
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
link/can promiscuity 0 allmulti 0 minmtu 0 maxmtu 0
can state ERROR-PASSIVE restart-ms 100
bitrate 20000 sample-point 0.875
tq 3125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1 brp 150
gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp_inc 1
clock 48000000
re-started bus-errors arbit-lost error-warn    error-pass            bus-off
      0                0                   0             20229           568602               528248    
numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 parentbus usb parentdev 1-1.2:1.0
RX: bytes    packets errors dropped missed mcast
    9454045 1181851    0        9484         0           0
TX: bytes packets errors dropped carrier collsns
        128       16           0           27             0             0

-------------------------------------

Bin im Moment etwas ratlos.

Update ...

strange things happen.... Hatte ja zwischendurch den Adapter abgeschaltet und vom Netz genommen. Nachdem ich auf github diesen Chat **open3e** gefunden hatte, habe ich den Adapter nochmals neu angeschlossen, jetzt unter can1 konfiguriert, da der zuvor unter can1 laufende Adapter auf Stiebel WP nach Neustart zu can0 mutiert ist. Beim ersten Versuch gab es noch einen Fehler F.1034 auf der Anlage, der aber komischerweise nach einer Minute sich wieder selbst resettet hat. Dabei habe ich auch gesehen, dass im Verlauf des heutigen Morgens mehrere Meldungen **Einstellungen des Systems wurden geändert**  in Vicare aufgetaucht sind. Nach Neustart des Scripts funktionierten zuerst die candump's  und mittlerweile läuft das ganze wieder problemlos, zumindest bis jetzt. Lediglich **bitrate 250000** habe ich, wie hier empfohlen, beim Start von zuvor 20000 auf den neuen Wert geändert. Es erscheinen auch keine Fehler mehr bei der _ip details statistic_ ...

... bitrate 250000 sample-point 0.875
tq 250 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1 brp 12
gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp_inc 1
clock 48000000
re-started bus-errors arbit-lost error-warn error-pass bus-off
       0                0                 0                 0                  0                0

Die Ratlosigkeit ist allerdings geblieben ...
 

Gruss Harald

rsch90, 2024-04-19:
@leitwolf00 kannst du mir mehr zu deiner Stiebel WP sagen? Wir haben auch eine LWZ 5 S Plus... ich hab bisher noch ein wenig Respekt davor da ran zu gehen. Kannst du mir da ein paar Tips geben? Wie hast du das umgesetzt?

philippoo66, 2024-04-20:
@leitwolf00  freut mich dass es wieder geht. behalt das bitte mal im Auge

leitwolf00, 2024-04-20:
@rsch90 Hallo. Hab dir was angehängt 
[Antwort an Roman.pdf](https://github.com/open3e/open3e/files/15048363/Antwort.an.Roman.pdf)

l2jalbe, 2024-08-15:
Hello, can somebody understand my case? Open3e work for about 48h and then stop with this errors. It runs on raspberry 3B+ that only executes open3e. What is the buffer space?

OSError: [Errno 105] No buffer space available

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/isotp/protocol.py", line 1632, in _main_thread_fn
    super().process(rx_timeout)
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/isotp/protocol.py", line 863, in process
    self.txfn(msg)
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/isotp/protocol.py", line 1692, in python_can_tx_canbus_3plus
    owner.bus.send(can.Message(arbitration_id=msg.arbitration_id, data=msg.data,
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/can/interfaces/socketcan/socketcan.py", line 789, in send
    sent = self._send_once(data, channel)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/can/interfaces/socketcan/socketcan.py", line 806, in _send_once
    raise can.CanOperationError(
can.exceptions.CanOperationError: Failed to transmit: No buffer space available [Error Code 105]
2024-08-15 21:38:30 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)
Traceback (most recent call last):
  File "/home/l2jalbe/.local/bin/open3e", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 364, in main
    readbydid(addr=ecudid[0], did=ecudid[1], raw=args.raw, msglvl=mlvl)
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 220, in readbydid
    value,idstr =  dicEcus[addr].readByDid(did, raw)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/open3e/Open3Eclass.py", line 154, in readByDid
    response = self.uds_client.read_data_by_identifier([did])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/udsoncan/client.py", line 167, in decorated
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/udsoncan/client.py", line 466, in read_data_by_identifier
    response = self.send_request(req)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/udsoncan/client.py", line 2216, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' %
udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)
/home/l2jalbe/loop_open3e.sh: line 15: kill: (196009) - No such process
Exception in thread Thread-1 (_main_thread_fn):
Traceback (most recent call last):
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/can/interfaces/socketcan/socketcan.py", line 804, in _send_once
    sent = self.socket.send(data)
           ^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 105] No buffer space available

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/isotp/protocol.py", line 1632, in _main_thread_fn
    super().process(rx_timeout)
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/isotp/protocol.py", line 863, in process
    self.txfn(msg)
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/isotp/protocol.py", line 1692, in python_can_tx_canbus_3plus
    owner.bus.send(can.Message(arbitration_id=msg.arbitration_id, data=msg.data,
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/can/interfaces/socketcan/socketcan.py", line 789, in send
    sent = self._send_once(data, channel)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/can/interfaces/socketcan/socketcan.py", line 806, in _send_once
    raise can.CanOperationError(
can.exceptions.CanOperationError: Failed to transmit: No buffer space available [Error Code 105]
2024-08-15 21:40:30 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)
Traceback (most recent call last):
  File "/home/l2jalbe/.local/bin/open3e", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 364, in main
    readbydid(addr=ecudid[0], did=ecudid[1], raw=args.raw, msglvl=mlvl)
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 220, in readbydid
    value,idstr =  dicEcus[addr].readByDid(did, raw)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/open3e/Open3Eclass.py", line 154, in readByDid
[... cut ...]
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/udsoncan/client.py", line 466, in read_data_by_identifier
    response = self.send_request(req)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/l2jalbe/.local/lib/python3.11/site-packages/udsoncan/client.py", line 2216, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' %
udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)

l2jalbe, 2024-08-15:
Now i've tried to add some space to the can0 buffer from 10 to 50 using:
**sudo ip link set can0 txqueuelen 50** 
and then using **ip -details link show can0** i cheked the qlen value that increse fromm 10 to 50.
then I've made a script (loop_open3e.sh) that every hour (3600 seconds) kill open3e, set down the can0, wait 5sec, re-up can0 and restarts open3e. What do you think?

Loop_open3e.sh
#!/bin/bash

while true; do
    # Terminare il processo open3e
    pkill -f open3e
    
    # Aggiungere un piccolo ritardo per garantire che il processo sia completamente terminato
    sleep 2
    
    # Svuotare il buffer di can0 disattivando e riattivando l'interfaccia
    sudo ip link set can0 down
    sleep 1
    sudo ip link set can0 up
    
    # Riavviare il processo open3e
    /home/l2jalbe/.local/bin/open3e &
    
    # Attendere 2 minuti prima di ripetere il ciclo
    sleep 120
done

MyHomeMyData, 2024-08-16:
@l2jalbe The message

"File "/home/l2jalbe/.local/lib/python3.11/site-packages/udsoncan/client.py", line 2216, in send_request
raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' %
udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=20.000 sec)"

tells us, an expected answer could not be reveived within 20 seconds. In most cases this is caused by hardware problems. So it's a good idea to check wiring.

Please post the result of the command `ip -details -statistics link show can0`. Execute it after some time of normal operation and when the error happened again.

What is your command line to start open3e?
How do you initalize the can bus (beside your measures regarding buffer length)?

l2jalbe, 2024-08-16:
This are the stas after when error happen:

_l2jalbe@raspberrypi:~ $ ip -details -statistics link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0  allmulti 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE restart-ms 100 
	  bitrate 250000 sample-point 0.875
	  tq 250 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1 brp 12
	  gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp_inc 1
	  clock 48000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 parentbus usb parentdev 1-1.1.2:1.0 
    RX:  bytes packets errors dropped  missed   mcast           
       8776646 1195376      0     575       0       0 
    TX:  bytes packets errors dropped carrier collsns           
        290928   36366      0       1       0       0_ 

I start open3e using open3e @args and my args file is this:
--can
can0
--read
1664, 1690, 1801, 1802, 1810, 1834, 1836
--timestep
10
--mqtt
192.168.1.126:1883:open3e
--mqttformatstring
{ecuAddr:03X}_{didNumber}_{didName}
--mqttuser
xxxxxxx:xxxxxxxxx
--verbose
--config
devices.json

How do you initalize the can bus (beside your measures regarding buffer length)? i can't answer to this question, i don't undestand

l2jalbe, 2024-08-16:
Regarding init: I've edit the file "80-can.network" and enabled the "systemd-networkd service" at start up

my 80-can.network file:
Match]
Name=can0

[CAN]
BitRate=250K
RestartSec=100ms

[26 more replies]

closed 2024-09-19 by MyHomeMyData