open3e/open3e #11: vx3 behind vcal results in NegativeResponseException

surt91, 2023-10-07:
I have a VX3 behind a Vitocal as the "leading device". The queries to the `-dev vcal` datapoints work fine, but when asking for the `-dev vx3` specific datapoints, I get the error below.

My setup is a Raspberry Pi 3 connected via an USB-Adapter to the CAN-Bus using the develop branch.

Do you have an idea how to fix this?

```
python Open3Eclient.py -v -c can0 -dev vx3 -r 1664
Traceback (most recent call last):
  File "/home/surt91/open3e-dev/Open3Eclient.py", line 111, in <module>
    response = client.read_data_by_identifier([did])
  File "/home/surt91/.local/lib/python3.9/site-packages/udsoncan/client.py", line 167, in decorated
    return func(self, *args, **kwargs)
  File "/home/surt91/.local/lib/python3.9/site-packages/udsoncan/client.py", line 442, in read_data_by_identifier
    response = self.send_request(req)
  File "/home/surt91/.local/lib/python3.9/site-packages/udsoncan/client.py", line 2205, in send_request
    raise NegativeResponseException(response)
udsoncan.exceptions.NegativeResponseException: ReadDataByIdentifier service execution returned a negative response ConditionsNotCorrect (0x22)
```

philippoo66, 2023-10-09:
hi Hendrik @surt91 !

sorry for the delayed response!  somehow I missed you question...

The problem is: If the VX3 ist connected to the Vcal, then the Vcal is the "Main Control Unit" and the VX3 is some 'slave device'. Only the Main Control Unit can get addressed by ECU Address 680h (responding on 690h). the current open3e always addresses the Main Control Unit (680/690). So the -dev vx3 assumes the VX3 as Main Control Unit. If you try to read ElectricalEnergyStorageStateOfCharge(1664), in your case you're trying to read it from the Vcal which doesn't serve the datapoint.

We are currently trying to figure out which ECU Address pair has to be used for which device/unit in slave mode. As soon as we figured out, we will add this capability to open3e accordingly.

You might figure out the ECU address of your VX3 by doing an 'ECU scan' like

[image]

better request NodeIdOnExternalCan(2944): ...#03220B8000000000

when you see a subsequent positive response frame in the dump with data 04 62 0B 80 xy where COB-ID is request_id+10h, then this is a candidate for the according ECU address pair for VX3 as slave. probably you will find more than one, so you might try or post them here and we will try to help ;-)

philippoo66, 2023-10-09:
ps. if you found the according ECU Addr/COB-IDs please tell us!

surt91, 2023-10-09:
This worked perfectly. During the scan I got a response at 6B1, so I monkey patched the script to use 
`rxid=0x6B1, txid=0x6A1`
And now I get plausible answers. :)

If I can do anything else, which is useful for the project, let me know.

philippoo66, 2023-10-09:
great! thank you for telling!

interesting - with a Vitoair linked to a Vcal we also got 6A1/6B1 for the Vitoair. Seems as if this is the ECU Addr for the 'first slave'...

next step could be to look eg. for the batteries:

[image]

but (I think) they are on the internal CAN bus of the VX3. so NodeIdOnExternalCan(2944) will not work. These devices serve e.g. 
- BusIdent 256, 
- VIN 377, 
- Soft/Hardware Version 580/581, 
- ParameterIdentificationVersion 1131 (klingt interessant), 
- OemProductVersion 1494, ...

these numbers (DID) are decimal, need to be converted to 2 bytes hex (-> kl mn) and requested by
`cob#0322klmn00000000` 
03 - single frame, 3 bytes payload
22 - service identifier (SID) for 'read data by identifier'
klmn - DID_hex msb..lsb

a positive resopse will have 62 (=22+40) in 2nd or third byte.

do you have one battery ore more?

if you like to do some reserach, have some time, and probably can do some Python programming, maybe we might like to have you with us in the group?! ;-)

surt91, 2023-10-14:
I get responses from 0x6D5 (when sending to 0x6c5 = 1733) for all DIDs you listed above except 1131.
I only have one battery module, so I guess it makes sense that the scan only found one responding CAN Id (apart from 0x690 and 0x6B1).

Regarding your offer to join the group: I do know Python and am kind of interested in learning more. I can probably not spend much time, but would surely find a little bit of time.

philippoo66, 2023-10-16:
moin Hendrik!

jap, jetzt mal den Cell Supervisor auf DIDs zu scannen macht schon Sinn, obwohl es nicht viele sind (vergl. Viguide Demo). Aktuell machen wir das noch 'the monkey way'

[image]
(hier für 680)

Weil wir bei langen Daten die Kommunikation nicht vollenden ist die lange Wartezeit (1 Sekunde) nötig. Ich habe mal versucht, mit ChatGPT das 'ordentlich' zu machen, aber weil ich es nicht ausprobieren kann, war das nicht von Erfolg gekrönt. Der Kern sieht so aus:

[image]

Meinst du, du könntest das (bzw sowas) zum Laufen bekommen?

surt91, 2023-10-20:
Ich bin nun dazu gekommen den Scan durchzuführen. 
Das Skript zum Scannen habe ich in diesem Gist: https://gist.github.com/surt91/63fca6e0fa4db120f181d31d884c622e
Falls es sinnvoll wäre das in das Repo aufzunehmen, stelle ich gerne einen Pull-Request.

Hier das Ergebnis des Scans

```
256:39
265:125
377:19
505:6
506:6
507:7
580:11
581:11
593:9
602:4
603:4
604:79
607:23
616:4
617:75
618:4
619:5
620:4
621:184
625:29
900:4
924:4
928:19
954:184
961:4
964:4
1165:4
1233:71
1234:71
1235:71
1236:71
1237:71
1286:184
1287:184
1288:184
1289:184
1494:11
1538:4
1539:4
1540:29
2375:10
2376:135
2377:48
2378:51
2379:25
2380:20
2534:184
2535:184
2536:184
2537:184
2538:184
2554:65
3056:4
3057:4
```

philippoo66, 2023-10-20:
top, danke! ich mach da bald ne Liste von. das war jetzt mit 6C5/6D5, oder?

ich find das Skript sinnvoll, wir müssen mal @abnoname fragen, wie/wo es hin soll...

surt91, 2023-10-20:
Ja, ganz genau. 0x6C5/0x6D5

achgut, 2023-10-20:
Date und Time sind keine ASCII sondern Zahlen. Hier ein Beispiel:

[image]

philippoo66, 2023-10-20:
auch gut. wie war das mit der ViessmannIdentificationNumber?

achgut, 2023-10-20:
Die ist seltsam. Bei mir steckt sie in den 16Byte als die 2. Ziffer in Hex Zahl. Immer eine 3 als die erste Ziffer.

also:
37 -> 7
33 -> 3 
37 -> 7
etc.

-> Identifikationsnummer: 737xxxxxxx

philippoo66, 2023-10-20:
das sind ja Ascii Zeichen...

[image]

philippoo66, 2023-10-20:
noch ne böde Frage zu Date/Time: 10 0A 17 sind doch nur 3 Bytes - was steht in den andren dreien?

achgut, 2023-10-21:
? Sind doch nur je 3 Byte
           505: RawCodec(3, "Date"), -> 100a17 -> 16.10.23
           506: RawCodec(3, "Time"), -> 161027 -> 22:16:39

achgut, 2023-10-21:
Noch ein weiteres Format: 
1132: RawCodec(97, "ViessmannIdentificationNumberListInternal"),
Bei mir folgendes:
Byte 1: Anzahl der internen Seriennummern (wahrscheinlich, bei mir 6, da ich 6 Batteriemodule habe)
dann je 16Byte die Seriennummern der einzelnen Batteriemodule

philippoo66, 2023-10-21:
> ? Sind doch nur je 3 Byte

sorry, k.A. woher ich "RawCodec(6, ..." im Kopf hatte.

das mit 1132 hab ich eingebaut. Könntest du es bitte ausprobieren?! Müsste eigentlich auch wer mit weniger Batterien probieren, aber bei 0x00 kommt ein Kästchen, sollte kein Problem geben.

philippoo66, 2023-10-21:
Date und Time hab ich auch einen "monkey poke" eingebaut....

Nelson666, 2024-05-16:
Hi,

ich habe das gleiche Setup. Leider finde ich keinen Weg die Datenpunkte für meinen Speicher oder PV auszulesen. Die Readall Methode funktioniert leider auch nicht, aber das scheint ja normal zu sein. Wie kann ich z. B. den Ladezustand des Akkus oder die aktuelle PV Leistung abrufen? Muss ich als Device eigentlich vcal oder vx3 angeben? Es funktioniert ja mit beiden (wenn es funktioniert:)) Ich habe übrigens eine Vitocal 250 mit einer VX3 (PV + Batterie).

philippoo66, 2024-05-16:
Moin @Nelson666 ,

wenn du noch mit `-dev` rummachst hast du wohl kein `-cnfg` angegeben (mit der gerade aktualisierten Version sollte das depictSystem wieder funktionieren).

in einem multi-ECU System musst du deine vx3 bzw. jeden (nicht von der main ECU) zu lesenden Datenpunkt explizit mit der ECU Adresse ansprechen: 
`0x6a1.256` also <ECU_Addr>.<Did>

Leider kenne ich ohne die devices.json die Adresse deiner vx3 nicht, deswegen ist die 6a1 beispielhaft zu verstehen.

Nelson666, 2024-05-17:
Danke für deine Hilfe. ja, ich bin noch ganz neu in dem Thema (open3c). Wo bekommt man denn eine richtige did Liste her, oder wie kann man eine erstellen (lassen)? Als nächstes werde ich mir die Geschichte mit der Konfigdatei anschauen.

Vielen Dank, der Tip mit der Adressierung hat geholfen, ich bekomme jetzt die Werte angezeigt. Bei mir ist es übrigens die 0x6a1, also richtig geraten. :)

philippoo66, 2024-05-17:
> Wo bekommt man denn eine richtige did Liste her, oder wie kann man eine erstellen (lassen)?

dazu ist depictSystem da. das scant erst auf antwortende ECU Adressen, dann jede antwortende ECU auf vorhandene Datenpunkte. 

Es ist allerding recht 'empfindlich' und irgendwie reagieren unterschiedliche Geräte-Softwaren immer wieder anders, ggf. muss man da ein wenig eingreifen (im depict code), damit man weiter kommt. 

Versuch's einfach mal und poste, womit es hängen bleibt, wenn es hängen bleibt. beim letzten 'Kandidaten' hat bsw. geholfen, einfach die Zeilen 97, 98
```
                    else:
                        raise Exception(e)
```
auszukommentieren.

Nelson666, 2024-05-17:
Hi, ja ds depictSystem läuft schon durch, aber irgendwie wußte ich dann nicht weiter. Ich habe das depictSystem mal in einem neuen (leeren) Ordner gestartet (ohne die DidEnums.txt aus der Repo). Da steht halt hinter jedem gefundenen Datenpunkt "(unknown)" dahinter. Was mach ich, wenn es durchgelaufen ist? Muss ich die open3e_dids2json starten?

philippoo66, 2024-05-17:
oh, das ist jetzt dumm gelaufen, weil das depictSystem braucht die Open3Edatapoints und die Open3Eenums für die "known" Datapoints... 

Du musst es in dem open3e Directory laufen lassen und am besten auch in der gleichen venv (wenn die eingerichtet ist). 

Dann stehen die datapoints_6yz.py und die devices.json auch im richtigen Verzeichnis. Sonst werden sie nicht gefunden wenn du `-cnfg devices.json` angibst. das greif alles ineinander... Also nicht auseinanderreissen und am besten im ursprünglichen Verzeichnis noch mal ausführen. es schreibt da nix vorhandenes über.

Nelson666, 2024-05-18:
Hallo noch einmal an alle und schon mal vielen Dank für eure Hilfe. ;)

Ich habe jetzt die git Repo noch mal ausgecheckt und im Ordner ./src/open3c den Befehl `open3e_depictSystem` ausgeführt. Er hat so einiges gefunden, aber ich bin mir nicht sicher, ob das alles passt.

Bei dem Befehl
`open3e -c can0 -r 0x6a1.1664,0x6a1.1836,0x6a1.1831,0x6a1.1690 -v`
bekomme ich folgende Fehlermeldung
`0x6a1 1664 ElectricalEnergyStorageStateOfCharge 67.0
0x6a1 1836 ElectricalEnergyStorageCurrentPower -906.0
0x6a1 1831 PhotovoltaicCurrentStringPower {"String1": 0.0, "String2": 0.0, "String3": 1253.0}
0x6a1 1690 ElectricalEnergySystemPhotovoltaicStatus {"ActivePower cumulated": 1253.0, "RectivePower cumulated": 0.0, "ActivePower String C": 0.0, "RectivePower String C": 0.0, "ActivePower String B": 0.0, "RectivePower String B": 0.0, "ActivePower String A": 1253.0, "RectivePower String A": 0.0, "OpMode": 2.0}
closing 0x680 - bye!
closing 0x6a1 - bye!
`

Wenn ich den oberen Befehl mit `-cnfg devices.json` (oderAbwandlungen davon) ausführe, kommt immer eine Fehlermeldung. Vom Prinzip her reicht das doch, oder muss/kann ich noch weitere Schritte durchführen, um das Script besser anhaben zu können? Leider verstehe ich immer noch nicht, wie ich jetzt die passenden Datenpunkte finde. Die oberen habe ich eigentlich nur zufällig im Netz gefunden.

 Viele Grüße
Nelson

philippoo66, 2024-05-18:
moin Nelson!

die neue Version ist ja eigentlich für `pip install ...` gedacht, und falls du das einmal gemacht hast, ist open3e irgendwo in den Tiefen des Dateisystems installiert und wird auch 'dort ausgeführt'. Wenn du einfach ein clone machst und im /src/open3e agierst, gehen beim Aufruf wahrscheinlich wieder irgendwelche Verweise in die falsche 'Richtung'. Wenn du im /src/open3e Verzeichnis das programm _dort_ starten willst, muss du wahrscheinlich 
```
cd /src/open3e
python3 Open3Eclient-py -c can0 -cnfg devices.json -r 0x6a1.1664,0x6a1.1836,0x6a1.1831,0x6a1.1690 -v
```
machen.

Ich habe es folgendermassen getestet und damit hat alles wunderbar geklappt:

Du machst am besten eine neue venv-Umgebung:

```
python3 -m venv ~/open3e
cd ~/open3e
source ~/open3e/bin/activate
pip install git+https://github.com/open3e/open3e.git
```
Jetzt sollte open3e funktionieren:
```
open3e -c can0 -dev vcal -v -r 256
open3e_depictSystem -c can0
open3e -c can0 -cnfg devices.json -v -r 0x6a1.256
```
mit `deactivate` verlässt du die virtuelle Umgebung wieder.

> Leider verstehe ich immer noch nicht, wie ich jetzt die passenden Datenpunkte finde.

Hat denn jetzt depictSystem eine Datei `Open3Edatapoints_6a1.py` erstellt und steht da was drin (wie gesagt `... : None` ist normal)? Hat es die `devices.json` erstellt?

Wenn ja, muss beides (und die anderen Open3Edatapoints_6yz.py) ind dem Verzeichnis stehen, in dem open3e beim Aufruf auch ausgeführt wird, und dann sollte es funktionieren.

Die Open3Edatapoints_6yz.py Dateien entahlten die gefundenen Datennpunkte. Du kannst sie auch 'einzeln' laden per `-dev _6a1` statt -cnfg... , musst aber natürlich noch die (von 680h abweichende) ECU Adresse beim Aufruf spezifizieren.
```
open3e -c can0 -dev _6a1 -tx 0x6a1 -a -v
```
(wie gesagt, wenn die Datei im Installationspfad steht, sonst wieder wie oben mit `python3 Open3Eclient.py ...`)

Sorry, wir dachten, durch die pip-Taglichkeit wird alles einfacher, aber offensichtlich gibt es auch ne Menge mehr Fallstricke... Halt durch, wir kommen schon noch zum Ziel!

Poste doch bitte
- die Fehlermeldung/en
- die devices.jason
- die Open3Edatapoints_6a1.py

dann kann ich viel mehr sehen/sagen.

Grüsse!

Nelson666, 2024-05-22:
Hallo,

danke für die Erläuterungen. Er hat jetzt die Open3Edatapoints_6yz.py und die devices.json erstellt in dem Ordner erstellt. Jetzt kommt auch kein sofortiger Fehler mehr, wenn cih die Readall Option wähle. Es kommt zwar immer noch ein Fehler, aber das ist sicher ein Bug. Die Ausgabe werde ich nachher noch hochladen.

philippoo66, 2024-05-25:
sorry, ich hab grad erst gesehen, dass du den log nachgeschoben hast.

ja, manchmal liefern die Datenpunkte zwar die gleiche Datenlänge, sind aber trotzdem anders kodiert. Das scheint hier der Fall zu sein. Richtig blöd ist, dass udsoncan nicht sagt, bei welchen DID das passiert, und auch in open3e haben wir da (bisher) keine Anzeige drin bei readAll().

ich habe dir eine Open3Eclass.py gemacht und angehängt, in der bei einem Fehler in readByDid() nicht das ganze 'Programm' beendet wird (und damit das readAll() ohne Ausgabe abbricht), sondern statt einem Wert die Fehlermeldung und die DID Nummer zurückgegeben wird.

```
    def readByDid(self, did:int, raw:bool):
        try:
            if(did in self.dataIdentifiers): 
                open3e.Open3Ecodecs.flag_rawmode = raw
                response = self.uds_client.read_data_by_identifier([did])
                # return value and idstr
                return response.service_data.values[did],self.dataIdentifiers[did].id
            else:
                return self.readPure(did)
        except Exception as e:
            return f"{e}", f"DID {did}"
```

entweder ersetzt du das readByDid() in Open3Eclass.py Zeilen 136-143 durch den Code mit Fehlerbehandlung, oder du ersetzt die ganze Open3Eclass.py durch die anhängende.

Eine kleine 'Herausforderung dabei könnte sein, dass das Projekt ja jetzt im /lib/ Verzeichnis steht, und damit möglicherweise nicht mehr 'einfach' editierbar ist. In diesem Falle müsstest du das Ändern der Datei per 

```
sudo nano /home/pi/open3e/lib/python3.12/site-packages/open3e/Open3Eclass.py
```
oder auch einem anderen Editor deiner Wahl, aber auf jeden Fall mit root Rechten machen.

Wenn du die Datei durch die anhängende ersetzen willst, diese herunterladen, dich in das Verzeichnis begeben, wo sie steht, und dort
- sicherheitshalber die originale Datei umbenennen (muss nicht unbedingt sein, du kannst sie ja jederzeit wieder aus dem Repo runterladen)
```
sudo mv /home/pi/open3e/lib/python3.12/site-packages/open3e/Open3Eclass.py /home/pi/open3e/lib/python3.12/site-packages/open3e/Open3Eclass.py.bkp
```
- und dann die neue Datei in das verzeichnis kopieren
```
sudo cp Open3Eclass.py.txt /home/pi/open3e/lib/python3.12/site-packages/open3e/Open3Eclass.py
```

wenn du danach nochmal ein `-a` (readall) machst, sollte es durchlaufen, und du bekommst bei dem Datenpunkt, wo es schief geht, eine Fehlermeldung als Wert und die Nummer des betroffenenen Datenpunktes zurückgeliefert, aber das 'Programm' läuft weiter. 

Du kannst dann in der Open3Edtapoints_6a1.py diesen Datenpunkt auf raw Codec ändern, und dann sollte es auch ohne Fehler laufen.

ich muss die Endung auf txt ändern, sonst kann ich die Datei hier nicht hochladen. du weisst wahrscheinlich, wie du damit umzugehehen hast? oben in dem copy (cp) command hab ich das ja schon berücksichtigt....

[Open3Eclass.py.txt](https://github.com/open3e/open3e/files/15440882/Open3Eclass.py.txt)

Nelson666, 2024-05-25:
Hallo an alle, 

nochmal vielen Dank für deine Hilfe. Danke für die ausführlichen Erklärungen. Das passt soweit. Ich hatte mir den Code auch schon angeschaut und habe versucht in der decode Methode das byte Handling anzupassen. Ich hatte festgestellt, dass es ein DID Nummer entweder falsch erkannt wird, oder anders codiert ist. Sie Beginnt mit x14 (Jahr 2000 bzw. 20) und dann kommt ein xFF, was ja für die 3. und 4. Stelle des Jahres quatsch ist. Leider habe ich in Python mit bytes Datentypen noch nicht gearbeitet, daher auch noch keinen Fehler gefunden bzw. ausgebessert. ;)

Hier übrigens noch die fehlerhafte DID Nummer 
`0x6a1 264 DID 264 "timestamp out of range for platform time_t"`

Jetzt kann es also für mich los gehen. :)

Ich möchte nun gern die für mich interessanten Did identifizieren. Wie mache ich das am Besten? Wenn ich das ReadAll durchlaufen lasse, sollte ich ja alles bekommen und ich muss die passenden DID's für mich herausfinden. 

Oder gibt es schon eine Liste, mit interessanten Datenpunkten für die VX3?

Letzte Frage, die "neue" Installationsmethode ist ja schön und gut, aber ich verstehe es leider immer noch nicht.

- Wie mache ich Updates (ein `git pull` wird ja nicht klappen. ;) )
- Wie starte ich jetzt diese Installation mit der erstellten venv-Umgebung in einem systemd Script?

philippoo66, 2024-05-25:
Moin Nelson!

danke für die Danks - es macht ja auch Spass, und es freut mich, wenn Leute von unserer vielen Arbeit gebrauch machen.

> Wenn ich das ReadAll durchlaufen lasse, sollte ich ja alles bekommen

ja. Wenn das depictSystem durchgelaufen ist, bommst du eigentlich alle lesbaren Datenpunkte (von denen manche dann auch schreibbar sind).

> gibt es schon eine Liste, mit interessanten Datenpunkten für die VX3?

das hängt ja auch ein bischen davon ab, was du machen willst. Ich hab leider(?) so ein Ding nicht, aber z.B. @MyHomeMyData  hat sich ziemlich intensiv beschäftigt glaubich und kann da vlt mehr zu sagen?!

> die "neue" Installationsmethode ist ja schön und gut, aber ich verstehe es leider immer noch nicht.

das geht mir auch so. Ich habe gerade nochmal um 'Aufschlauung' diesbezüglich gebeten. Ich habe ja am Anfang recht intensiv entwickelt, aber das Team hat auch  immer wieder Zuwachs bekommen (einige haben auch ihre Aktivitäten eingestellt), und ein recht frisches Mitglied hat die Umstellung auf pip gemacht.

Ich hab einfach mal ChatGPT gefragt:

```
Die Installation eines Projekts mit pip install im Vergleich zum Klonen eines Repositories bietet mehrere Vorteile:
1. Einfachheit und Bequemlichkeit

    _Pip Install_: Mit einem einzigen Befehl (pip install paketname) wird das Paket heruntergeladen und installiert. Es werden alle Abhängigkeiten automatisch mitinstalliert.
    _Klonen des Repositories_: Nach dem Klonen müssen oft mehrere manuelle Schritte ausgeführt werden, wie das Installieren von Abhängigkeiten (pip install -r requirements.txt) und eventuell das Einrichten der Entwicklungsumgebung.

2. Verwaltung der Abhängigkeiten

    _Pip Install_: Pip verwaltet Abhängigkeiten automatisch und stellt sicher, dass kompatible Versionen installiert werden. Es überprüft, ob es Konflikte zwischen den Paketen gibt und löst diese weitgehend selbstständig.
    _Klonen des Repositories_: Man muss sicherstellen, dass alle Abhängigkeiten korrekt installiert und kompatibel sind. Dies kann zu Problemen führen, wenn nicht alle Abhängigkeiten in der requirements.txt oder setup.py Datei korrekt angegeben sind.

3. Versionierung und Updates

    _Pip Install_: Mit pip kannst du spezifische Versionen eines Pakets installieren und leicht aktualisieren (pip install --upgrade paketname). Es ermöglicht auch das einfache Zurückrollen auf eine frühere Version, falls nötig.
    _Klonen des Repositories_: Das Aktualisieren auf eine neue Version erfordert das manuelle Pullen der neuesten Änderungen und das Neuinstallieren von Abhängigkeiten. Das Verwalten von Versionen ist weniger bequem.

4. Bereitstellung und Verteilung

    _Pip Install_: Pakete, die auf PyPI (Python Package Index) verfügbar sind, können leicht verteilt und installiert werden. Dies erleichtert die Bereitstellung von Projekten an eine breite Benutzerbasis.
    _Klonen des Repositories_: Benutzer müssen das Repository kennen und es manuell klonen, was weniger benutzerfreundlich ist, besonders für Nicht-Entwickler.

5. Umgebungsisolation

    _Pip Install_: In Kombination mit virtuellen Umgebungen (z.B. venv oder virtualenv) ermöglicht pip eine saubere Isolation von Projekten und deren Abhängigkeiten. Dies verhindert Konflikte zwischen verschiedenen Projekten.
    _Klonen des Repositories_: Man muss manuell sicherstellen, dass virtuelle Umgebungen richtig eingerichtet und verwendet werden.

6. Standardisierung

    _Pip Install_: Es ist der de facto Standard für die Installation von Python-Paketen und wird breit unterstützt und dokumentiert.
    _Klonen des Repositories_: Dies ist oft projektspezifisch und kann variieren, was die Standardisierung erschwert.

**Zusammengefasst bietet pip install eine einfachere, bequemere und standardisierte Möglichkeit**, Python-Projekte und ihre Abhängigkeiten zu verwalten, zu installieren und zu aktualisieren.
```

zum **Update** sagt es ja `leicht aktualisieren (pip install --upgrade paketname)` (aber hehalte auch immer die alte 'Weisheit' im Hinterkopf: _"Never change a running System!"_ ;-)

zu **Starten per systemd** hat @fabian727 gerade was geschrieben: [Virtual Environment](https://github.com/open3e/open3e/issues/98)

```
the systemd service must not call the global python, but the one within the virtual environment.
so change 'python /my/home/open3e/.py´ to ´/my/home/open3e/venv/python /my/home/open3e/.py´
that's it
```

klingt alles recht einfach in meinen Ohren...

Grüsse!

Nelson666, 2024-05-25:
Super, danke. Ich werde das der Reihe nach alles testen und einrichten. Auf die Nummer mit ChatGPT wäre ich nicht gekommen, aber du hast recht, klingt einfach. ;)

Wenn etwas unklar ist, melde ich mich noch mal. Ansonsten weiter so, ich finde das total super, wenn man nicht von der Cloud abhängig ist. Viessmann hat ja einen am Zeiger -> 20€ pro Monat für nen API Zugang. 🤔🤨

In diesem Sinne, ein schönes Wochenende. 
Nelson

MyHomeMyData, 2024-05-25:
@Nelson666 Das sind einige interessante Datenpunkte der VX3:

ID | Beschreibung | Bedeutung und Einheit
-- | -- | --
1690 | ElectricalEnergySystemPhotovoltaicStatus.ActivePower   cumulated | Aktuelle   PV-Leistung (W)
1801 | ElectricalEnergyStorageEnergyTransferStatistic | Energiedaten der Batterie (Wh)
1802 | EnergyProductionPhotovoltaic | Energiedaten   PV (Wh)
1810 | ElectricalEnergyInverterPowerAc.ActivePower | Aktuelle Inverterleistung AC (W)
1834 | ElectricalEnergyStorageStateOfEnergy.StateOfEnergy | SoC   (Wh)
1836 | ElectricalEnergyStorageCurrentPower | Leistung Laden (-) / Entladen (+) der Batterie (W)
2214 | BackupBoxConfiguration.DischargeLimit | Entladegrenze   Backupbox (%)
2240 | BatteryTemperatureSensor.Actual | Aktuelle Batterietemperatur (°C)

Btw.: Kennst Du den ioBroker-Adapter [e3oncan](https://github.com/MyHomeMyData/ioBroker.e3oncan)? Ist über das Beta-Repo in ioBroker verfügbar.

philippoo66, 2024-05-25:
persönlich kenn ich den nicht (weil ich kein E3 Gerät habe), aber der stammt von @MyHomeMyData , einem unserer 'Ur-Entwickler'. Er 'basiert' auf der open3e Arbeit. Man hört nur Gutes davon! Wenn du ioBroker benutzt, vergiss open3e und benutz den Adapter! da ist glaubich vieles einfacher dann und es geht alles, was auch mit open3e geht.

philippoo66, 2024-05-25:
> Viessmann hat ja einen am Zeiger -> 20€ pro Monat für nen API Zugang.

es sind ja nicht 'nur' die Kosten. ich persönlich (und nicht nur ich) will meine Daten nicht im Internet und auf irgend einem 'fremden' Server, der oft genug wieder mal nicht funktioniert wie er soll. Die Granularität wird auch oft bemängelt, genauso wie der zeitliche Verzug ode eben Aussetzer... Ich meine mich auch zu erinnern, dassss es sogar ein Gesetz gibt, was einem das (kostenlose) Recht an seinen IoT Daten zusichert. Um sich da drum rum zu drücken, bietet Viessmann ja auch euínen kostenlosen Zugang an, wo man aber wohl an allen Ecken und Enden mit Beschränkungen konfrontiert wird. 

Ich habe das ja alles vor 1.5 Jahren sehr konstruktiv und wohlwollend in meinem [Offenen Brief an Max Viessmann](https://community.viessmann.de/t5/Gas/Internet-der-Dinge-offener-Brief-an-Max-Viessmann/td-p/278925) angesprochen, aber die Antwort da drauf war der Startschuss zu open3e. Diese beiden Threads dominieren die most-liked Liste im Viessmann Forum, aber dieses Interesse der Kunden ignoriert Viessmann trotz gegenteilig lautender Beteuerungen beharrlich... ;-)

philippoo66, 2024-05-25:
> 'persönlich' kenn ich den Adapter nicht (weil...

huch, da war ich mit dem 'Absender durcheinander geraten... ich lösch das mal wieder ;-)

Nelson666, 2024-05-28:
Hallo @MyHomeMyData,

Danke für die Datenpunkte, wenn meine Liste komplett (oder zumindest zu 90% fertig) ist, dann werde ich sie hier auch mal in eine Tabelle packen.
Ich nutze Home Assistant zur Automatisierung. Ich bin gerade in der Umstellung von OpenHAB auf HA. Sieht für mich alles sehr gut aus, daher kann ich zu ioBroker nicht viel sagen. Ich such gerade die Interessanten Datenpunkte für mich. Allerdings bin ich nicht so tief in der Materie wie Wolfgang03 [github Seite](https://github.com/Wolfgang-03/Open3E-HA?tab=readme-ov-file). Ich versuche eigentlich "nur" herauszufinden wie voll der Akku ist und wieviel Leistung gerade an der PV anliegt, um zu entscheiden, ob der Spüler oder WM angeschaltet werden sollte. :)

Viele Grüße
Kendy

Philip-Wiege, 2024-11-04:
Das ursprüngliche Problem wurde erfolgreich gelöst.

closed 2024-11-04 by Philip-Wiege

philippoo66, 2024-11-04:
@Philip-Wiege schön dass du ein wenig aufräumst! danke! :-)