open3e/open3e #12: Falscher Codec für VX3 Adresse 1831 PhotovoltaicCurrentStringPower

achgut, 2023-10-13:
Hallo

Der Wert PhotovoltaicCurrentStringPower stimmt bei mir nicht (zu niedrig). Der verwendete Codec ist nicht korrekt.

Nachdem ich mit die Rohdaten angeschaut habe, folgende Erkenntnis meinerseits:
- Derzeitige Definition "1831: O3EInt16(12, "PhotovoltaicCurrentStringPower", offset=8, scale=1.0)" zeigt nur die Leistung von einem String
- Es sind aber 12 Byte. Jeweils 4 bilden aus meiner Sicht die Leistung aus einem String ab. VX3 kann bis zu 3 Strings.
- Es sollten also 3 Werte aus diesem Aufruf zurückkommen. 
Beispiel Rohdaten:
00 00 00 00 76 05 00 00 d0 02 00 00
String 1: 0 Watt -> hier bei mir kein PV String sondern Batterie angeschlossen
String 2: 0576h -> 1398W
String 3: 02D0h -> 720W

Leider übersteigt es meine Kenntnisse, hier O3EComplexType oder eine neue Funktion/Class zu implementieren. Vielleicht weiß ja jemand, wie man dies programmiert. Wäre auch toll, wenn noch die Summe aller Strings als 4. Wert zurückkommt.

DpunktS, 2023-10-14:
So sollte es gehen.

1831: O3EComplexType(12, "PhotovoltaicCurrentStringPower", [O3EInt32(4, "String1", scale=1.0), O3EInt32(4, "String2", scale=1.0), O3EInt32(4, "String3", scale=1.0)]),

achgut, 2023-10-14:
Super. Passt.
Geändert und funktioniert.
Pull Request gestellt.

closed 2023-10-14 by achgut