open3e/open3e #14: VX3: Einige Datenpunkte bringen Fehler

achgut, 2023-10-16:
Hallo

Ich habe in der VX3 Liste einige Datenpunkte gefunden, die bei mir einen Fehler produzieren. Hat jemand anderes auch das Problem?

Anbei die Liste der nicht funktionierenden Datenpunkte, die open3e Fehlermeldung und ein Candump Beispiel. 

1577: RawCodec(139, "ElectricalEnergyStorageModuleOneOperatingData"),
1578: RawCodec(139,   "ElectricalEnergyStorageModuleTwoOperatingData"),
1579: RawCodec(139,   "ElectricalEnergyStorageModuleThreeOperatingData"),
1580: RawCodec(139,   "ElectricalEnergyStorageModuleFourOperatingData"),
1581: RawCodec(139,   "ElectricalEnergyStorageModuleFiveOperatingData"),
1582: RawCodec(139,   "ElectricalEnergyStorageModuleSixOperatingData"),
2329: RawCodec(14,   "BatteryEnergyUsedAverage"),
2849: RawCodec(27,   "CrankCaseHeaterOnTimer"),
3085: RawCodec(18,   "Object_DID_3085"),
3086: RawCodec(18,   "Object_DID_3086"),
3087: RawCodec(18,   "Object_DID_3087"),
3088: RawCodec(18,   "Object_DID_3088"),
3089: RawCodec(18,   "Object_DID_3089"),
3090: RawCodec(18,   "Object_DID_3090"),
3103: RawCodec(6,   "Object_DID_3103"),
3107: RawCodec(7,   "Object_DID_3107"),
3108: RawCodec(9,   "Object_DID_3108"),

**Beispiel Fehlermeldung  bei 1577:**
 
python3 Open3Eclient.py -c can0 -dev vx3 -r 1577 -v
Traceback   (most recent call last):
File   "/home/ubuntu/open3e/Open3Eclient.py", line 110, in <module>
response =   client.read_data_by_identifier([did])
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File   "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py",   line 174, in decorated
return func(self, *args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File   "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py",   line 449, in read_data_by_identifier
response = self.send_request(req)
^^^^^^^^^^^^^^^^^^^^^^
File   "/home/ubuntu/.local/lib/python3.11/site-packages/udsoncan/client.py",   line 2223, in send_request
raise NegativeResponseException(response)
udsoncan.exceptions.NegativeResponseException:   ReadDataByIdentifier service execution returned a negative response   ConditionsNotCorrect (0x22)
 
Candump 
can0    680   [8]  03 22 06 29 00 00 00 00
can0    690   [8]  03 7F 22 22 55 55 55 55
danach keine weitere Meldung (Müssten 139 Bytes sein)

philippoo66, 2023-10-16:
wir haben ne VX3 mit hochaltueller Firmware gescannt - wie alt ist deine? es kann sein, dass die neue FW DIDs enthält, die bei älteren nicht drin sind (hatten wir glaubich bei ner andren auch schon). Grad die 3000er sind ja superneu (da gibt es auch n datapointsVx3 Update mit 'echtem' Text).

DID 0629h (1577): das 03 7F 22 22 ist ne Fehlermeldung, die ist komplett so. 
7F = NegativeResponse
22 = Requested SID
22 = ConditionsNotCorrect

achgut, 2023-10-17:
OK. Danke. Muss ich mal prüfen, was der FW Stand ist.
Dann ist es ja kein Fehler vom Programm oder der Kommunikation. Die wichtigen Daten kann ich ja abrufen.

closed 2023-10-17 by achgut

philippoo66, 2023-10-17:
leider kann ich dir grad garnicht sagen welche Version die vom Scan hat. Es gibt auch noch (mindestens) eine dritte Versions-Information (die auch im Display angezeigt wird), ich glaube es war 1494: RawCodec(8, "OemProductVersion"), aber auch da kann ich dir nicht sagen, welche es war ;-) aber ich weiss, dass wir da was verglichen hatten (ich glaube in einem Online Meeting) wegen unteschiedlicher 'Verhalten'...