open3e/open3e #24: Fehlerhafte (negative?) Werte beim VX3 (1810+1824)

Sebliner, 2023-10-26:
Ich logge meine Werte über influxdb und beim Darstellen in grafana ist mir aufgefallen, dass bei ElectricalEnergyInverterPowerAc und ThreePhaseInverterCurrentPower/cumulated absurd hohe Werte gespeichert wurden.

Bei beiden Datenpunkten sank die Anzeige immer weiter Richtung 0, in dem Moment wo Sie allerdings unter null springen "müsste" ging sie bei PowerAc auf 65535, bei CurrentPower auf 4294967295 und sank dann dort noch etwas weiter. Nach kurzer Zeit stiegen die Werte wieder - als PowerAc dann wieder über 65535 rutschte sprang der Werte wieder auf knapp über 0. Bei CurrentPower das gleiche Verhalten.
ThreePhaseInverterCurrentApparentPower/cumulated blieb im gleichen Zeitraum (etwas) über 0, zeigte also kein solches Verhalten.

Zum betreffenden Zeitpunkt heute Mittag war Regen, Ertrag fast komplett weg und Speicher leer. Reine Spekulation: Der Wechselrichter hat zu diesem Zeitpunkt nichts mehr eingespeist, allerdings Scheinleistung "gezogen", wodurch PowerAc eventuell tatsächlich negativ wäre und dann passt die negative Anzeige nicht mehr? Ist aber nur ohne jegliches Wissen geraten...

Ebenfalls auffällig: Bei meinem 3. String (PhotovoltaicCurrentStringPower/String3) sprang der Werte heute (kurz nach Sonnenaufgang) auch 2x mal ganz kurz nach oben auf 4294967295. Auch dort war der Wert vorher/nachher um 0 und wurde nur einmal kurz geloggt.

Ein Screenshot als Beispiel:
[image]

philippoo66, 2023-10-26:
Moin!

tausch mal 
`1810: O3EInt16(4, "ElectricalEnergyInverterPowerAc", scale=1.0),`
gegen
`1810: O3EInt32(4, "ElectricalEnergyInverterPowerAc", scale=1.0),`

bei
`            1824: O3EComplexType(16, "ThreePhaseInverterCurrentPower", [O3EInt32(string_len=4, idStr="cumulated", scale=1), O3EInt32(string_len=4, idStr="L1", scale=1), O3EInt32(string_len=4, idStr="L2", scale=1), O3EInt32(string_len=4, idStr="L3", scale=1)]),`
und
`            1825: O3EComplexType(16, "ThreePhaseInverterCurrentApparentPower", [O3EInt32(string_len=4, idStr="cumulated", scale=1), O3EInt32(string_len=4, idStr="L1", scale=1), O3EInt32(string_len=4, idStr="L2", scale=1), O3EInt32(string_len=4, idStr="L3", scale=1)]),
`

schreib jeweils ein `signed=True` dabei (bei jedem der drei Werte)

`scale=1` kann raus, das ist default

bei `PhotovoltaicCurrentStringPower` dann auch.

sag dann auch bitte Bescheid, ob es dann besser ist!

danke & Grüsse!
Phil

philippoo66, 2023-10-26:
du kannst die Sachen auch kürzer 'gestalten':

`           1825: O3EComplexType(16, "ThreePhaseInverterCurrentApparentPower", [O3EInt32(4, "cumulated", signed=True), O3EInt32(4, "L1", signed=True), O3EInt32(4, "L2", signed=True), O3EInt32(4, "L3", signed=True)]),`

Sebliner, 2023-10-26:
Die Zeilen habe ich entsprechend angepasst. Mangels Sonne und Energie im Speicher sind jetzt aber sämtliche Werte erwartungsgemäß genau 0 - ich logge natürlich weiter und schaue morgen Abend drüber wie es am Tag aussah. Ich melde mich wieder - danke für die Hilfe!

philippoo66, 2023-10-26:
supi. vielleicht muss du den ElectricalEnergyInverterPowerAc auch noch signed=True machen...

Sebliner, 2023-10-27:
Kurzer Zwischenstand: der 1810 (InverterPowerAc) zeigt mit der Änderung auf O3EInt32 fast ausschließlich absurd hohe und willkürliche Werte, nur immer mal wieder kurz zwischendurch theoretisch realistische Werte. Ob die dann stimmen kann ich schlecht vergleichen. Auf jeden Fall stimmen die Werte so grundsätzlich nicht mehr.
Ich würde heute Nachmittag Zuhause erst mal wieder zurück auf O3EInt16 gehen, allerdings signed=True lassen - korrekt?

MyHomeMyData, 2023-10-27:
Ich bekomme für 1810 plausible Werte, wenn ich von zwei signed 16-Bit Werten ausgehen. Vermutlich Wirk- und Blindleistung in Watt:
```
1810: O3EComplexType(4, "ElectricalEnergyInverterPowerAc", [O3EInt16(2, "ActivePower", scale=1.0, signed=True),O3EInt16(2, "ReactivePower", scale=1.0, signed=True)]),
```

philippoo66, 2023-10-27:
ok, anscheinend gilt das so für 1810 und 1603 (vergl. Thread im Viessmann Forum). Haben wir noch mehr Datenpunkte, auf die das anzuwenden ist?

Sebliner, 2023-10-31:
Habe jetzt die letzten Tage entsprechend geändert und geloggt - jetzt scheint alles super zu passen 👍
Danke!

philippoo66, 2023-10-31:
top, danke für die Rückmeldung!

Sebliner, 2023-11-07:
Nochmal eine kurze Rückmeldung von mir mit ähnlichem Problem aber anderem Datenpunkt:
beim loggen vom Datenpunkt 1831 (PVStringPower) wird bei mir beim string2 ebenfalls kurzzeitig ein absurd hoher Wert geloggt zu einem Zeitpunkt als eigentlich 0 angezeigt werden sollte:
[image]
Ideen woran das liegen könnte bzw. was ich probieren könnte? Dieser Wert sollte in der Theorie auch nicht unter 0 fallen....

playcom-de, 2023-11-07:
Das schaut nach einem Überlauf aus. Ein "unsigned integer 32" liegt im Wertebereich von 0..4294967295. Wenn der Wert unter Null rutscht, dann findet ein Überlauf statt. Zum Beispiel 4294967291 entspricht -4.

Sebliner, 2023-11-07:
Dass es durchs unter 0 rutschen kommt hatte ich mir fast gedacht - nur dürfte das bei dem Datenpunkt gar nicht passieren dürfen, oder? Vor allem entpräche das bei den Beispielwerten oben gleich mal ordentlichen -37W...
Würde signed=True das richtig korrigieren?