open3e/open3e #26: decode circuit target temperatures

stobies, 2023-11-03:
I am new to this projects, so first of all a great than you for your work! 

I would like to add the datapoints for the target temperature for the circuits (WAGO datapoints 1643 to 1646). It seems like these datapoints have two values each : the setpoint (set by the user), and the calculated target temperature for the circuit (according to the set heating curve).

How would I go about adding such data points? I would be willing to help with the work but would need a little bit of a push in the right direction to get me started.

jwende, 2023-11-03:
I only see one value:
python3 Open3Eclient.py -c can0 -dev vcal -r 1644 -raw
e600
python3 Open3Eclient.py -c can0 -dev vcal -r 1644
23.0
Definitions:
            1643	: O3EInt16(2, "MixerOneCircuitCurrentTemperatureSetpoint", scale = 10, signed=True),
            1644	: O3EInt16(2, "MixerTwoCircuitCurrentTemperatureSetpoint", scale = 10, signed=True),
            1645	: O3EInt16(2, "MixerThreeCircuitCurrentTemperatureSetpoint", scale = 10, signed=True),
            1646	: O3EInt16(2, "MixerFourCircuitCurrentTemperatureSetpoint", scale = 10, signed=True),

I would look for:
            987	: O3EInt16(2, "MixerOneCircuitFlowTemperatureTargetSetpoint"),
            988	: O3EInt16(2, "MixerTwoCircuitFlowTemperatureTargetSetpoint"),
            989	: O3EInt16(2, "MixerThreeCircuitFlowTemperatureTargetSetpoint"),
            990	: O3EInt16(2, "MixerFourCircuitFlowTemperatureTargetSetpoint"),
and 
            424	: O3EInt16(9, "MixerOneCircuitRoomTemperatureSetpoint", signed=True),
            426	: O3EInt16(9, "MixerTwoCircuitRoomTemperatureSetpoint", signed=True),
            428	: O3EInt16(9, "MixerThreeCircuitRoomTemperatureSetpoint", signed=True),
            430	: O3EInt16(9, "MixerFourCircuitRoomTemperatureSetpoint", signed=True),

stobies, 2023-11-03:
Great, works perfectly. Thanks a lot!

closed 2023-11-03 by stobies

jwende, 2023-11-03:
just a comment:
The decoding for 424-430 should look like this:
           426	: O3EComplexType(9, "MixerTwoCircuitRoomTemperatureSetpoint", [O3EInt16(2,"Comfort"),O3EInt16(2,"Normal"),O3EInt16(2,"Reduced"),RawCodec(3,"Filled")]),
 
Result for my system:
python3 Open3Eclient.py -c can0 -dev vcal -r 426
{'Comfort': 23.0, 'Normal': 20.0, 'Reduced': 14.0, 'Filled': '000000'}

will change that with the next commit of the 'develop' stream

philippoo66, 2023-11-03:
@stobies  please give the develop branch a try. we did a lot of rework regarding the codecs and their assignments to the dids. you will get much more (useful) info when running the develop branch. it's stable as far as we know, if you experience issues please let us know! thanks! :-)

stobies, 2023-11-03:
Thanks, will do.