open3e/open3e #73: Min/max/average vertauscht bei Vitodens 300-W OutsideTemperatureSensor

Frank-Steiner, 2024-02-07:
Moin moin,

meine Vitodens 300-W B3HG-19 liefert beim Auslesen

`vdens 274 OutsideTemperatureSensor {"Actual": 8.8, "Average": -11.9, "Minimum": 21.1, "Maximum": 4.0, "Unknown": 0}`

Bei allen anderen Sensoren sind die Textstrings in der Reihenfolge Actual/Min/Max/Avg, was auch hier richtig waere. In Open3Edatapoints.py gibt's die Zeile mit der richtigen Reihenfolge auch, ist aber auskommentiert. Offenbar liefern andere Anlagen die Reihenfolge anders, so dass es mal geaendert wurde...

cu,
Frank

philippoo66, 2024-02-07:
hm, das ist Mist. ich hatte das grade erst geändert, weil es woanders anders war. Jetzt haben wir ein Problem. ich muss mal nachdenken, ob mir da eine Lösung einfällt, aber ich fürchte mit depictSystem haben wir da schlechte Chancen. Mit -dev vdens würde es gehen, aber die datapointsVdens ist ziemlich alt und ausserdem statisch. 

Da hat Viessmann ja mal wieder nen tollen Bock geschossen!

Du kannst natürlich die beiden 274 Zeilen 'umkommentieren', dann funktioniert's zumindest...

@ alle war das bei den Wärmepumpen mit der 'ungewöhnlichen' Reihenfolge? bei allen?

MyHomeMyData, 2024-02-08:
Meine Vitocal 250 liefert grade keine stimmigen Werte, aber das Format scheint der Standard-Anordnung zu entsprechen:
`274 OutsideTemperatureSensor { "Actual": 8.4, "Minimum": 0, "Maximum": -0.1, "Average": 7.7, "Unknown": 0 }`

Wolfgang-03, 2024-02-08:
Bei mir (Vitocal 250) werden aktuell diese Werte geliefert
`274 OutsideTemperatureSensor { "Actual": 3.4, "Average": 0.0, "Minimum": -0.1, "Maximum": 2.7,  "Unknown": 0 }`
Allerdings werte ich immer nur "Actual" aus, weil ich noch nie gesehen habe, dass die anderen Bereiche sinnvolle Werte liefern. Sonst wäre wohl nicht Maximum kleiner als Actual. Auch ist mir nicht klar über welchen Zeitraum oder wie diese Werte (Minimum, Maximum, Average) ermittelt werden. In den letzten 10 Tage war die niedrigste Aussentemperatur bei mir bei +2.7. Mir ist zwar aufgefallen, dass ihr die Reihenfolge geändert habt, das habe ich jedoch ignoriert. 

Das Gleiche gilt im übrigen bei mir auch bei anderen Sensoren, dass ich die Werte von (Minimum, Maximum, Average) nicht nachvollziehen kann, egal in welcher Reihenfolge ich die Werte interpretieren würde. Zum Beispiel 268 FlowTemperaturSensor liegt bei mir immer im Bereich zwischen 28 und 52 Grad, je nach aktuellen Betriebsprogramm und Umgebungstemperatur. Es werden jedoch folgende Werte geliefert: Minimum:19.3, Maximum:60.8, Average:47.8. Im Prinzip sind die Werte zwar plausibel, jedoch finde ich diese in den historischen Daten bei mir nicht. Das Maximum da die Anlage hatte waren 55.8 Grad, und das auch nur, weil ich zu diesem Zeitpunkt mit dem PV-Modus gespielt hatte.

MyHomeMyData, 2024-02-08:
@philippoo66 Moin Phil, ich habe in der Datei `Open3Edatapoints.json` (brauche ich für den Adapter) einen Versions-String ergänzt.

In der `Open3Edatapoints.json.py` habe ich im develop Branch die 274 wieder auf den Standard zurückgestellt. Für 1684 habe ich einen Temperatursensor-Codec ergänzt.

DpunktS, 2024-02-08:
> Meine Vitocal 250 liefert grade keine stimmigen Werte, aber das Format scheint der Standard-Anordnung zu entsprechen: `274 OutsideTemperatureSensor { "Actual": 8.4, "Minimum": 0, "Maximum": -0.1, "Average": 7.7, "Unknown": 0 }`

@MyHomeMyData  Bei meiner Vitocal 250 sehen die Werte auch so komisch aus, aber die Anordnung ist korrekt.

  `"Actual": 6.6, "Minimum": 0, "Maximum": -0.1, "Average": 5.6, "Unknown": 0`

closed 2024-08-26 by MyHomeMyData