open3e/open3e #78: depict by ip fails with curren master 11.2.24 22:30

AnHardt, 2024-02-11:
```
e3@openE3:~/open3e $ python3 Open3E_depictSystem.py -d 10.83.83.1
read DID enums ...
2951 DIDs listed.
scan COB-IDs 0x680 to 0x6ff ...
ECU found: 0x680 : HMUMASTER
Traceback (most recent call last):
  File "/home/e3/open3e/Open3E_depictSystem.py", line 77, in scan_cobs
    response = client.send_request(
  File "/home/e3/.local/lib/python3.9/site-packages/udsoncan/client.py", line 2164, in send_request
    self.conn.send(payload)
  File "/home/e3/.local/lib/python3.9/site-packages/udsoncan/connections.py", line 80, in send
    self.specific_send(payload)
  File "/home/e3/.local/lib/python3.9/site-packages/doipclient/connectors.py", line 43, in specific_send
    self._connection.send_diagnostic(bytearray(payload))
  File "/home/e3/.local/lib/python3.9/site-packages/doipclient/client.py", line 720, in send_diagnostic
    raise IOError(
OSError: Diagnostic request rejected with negative acknowledge code: 3

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/e3/open3e/Open3E_depictSystem.py", line 313, in <module>
    lstEcus = scan_cobs(startcob, lastcob)
  File "/home/e3/open3e/Open3E_depictSystem.py", line 96, in scan_cobs
    raise Exception(e)
Exception: Diagnostic request rejected with negative acknowledge code: 3
```
on a PI3 with standard (updated) install .

philippoo66, 2024-02-11:
ok, thank you. unfortunately by mistake I just updated the depictSystem in the master branch by the depict_all_python, so I'm not sure if you picked that one or the older one (still in develop branch). Can you please tell me if the Open3E_depictSystem looks 

like this (new, master, all_python)
[image]

or like this (old, develop)
[image]

thank you!

AnHardt, 2024-02-11:
It looked like
like line 28 (master)
```
from udsoncan.connections import PythonIsoTpConnection
```
With Line 26 (develop)
```
from udsoncan.connections import IsoTPSocketConnection
```
i get
```
e3@openE3:~/open3e $ python3 Open3E_depictSystem.py -d 10.83.83.1
read DID enums ...
2951 DIDs listed.
scan COB-IDs 0x680 to 0x6ff ...
ECU found: 0x680 : HMUMASTER
Traceback (most recent call last):
  File "/home/e3/open3e/Open3E_depictSystem.py", line 76, in scan_cobs
    response = client.send_request(
  File "/home/e3/.local/lib/python3.9/site-packages/udsoncan/client.py", line 2164, in send_request
    self.conn.send(payload)
  File "/home/e3/.local/lib/python3.9/site-packages/udsoncan/connections.py", line 80, in send
    self.specific_send(payload)
  File "/home/e3/.local/lib/python3.9/site-packages/doipclient/connectors.py", line 43, in specific_send
    self._connection.send_diagnostic(bytearray(payload))
  File "/home/e3/.local/lib/python3.9/site-packages/doipclient/client.py", line 720, in send_diagnostic
    raise IOError(
OSError: Diagnostic request rejected with negative acknowledge code: 3

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/e3/open3e/Open3E_depictSystem.py", line 284, in <module>
    lstEcus = scan_cobs(startcob, lastcob)
  File "/home/e3/open3e/Open3E_depictSystem.py", line 95, in scan_cobs
    raise Exception(e)
Exception: Diagnostic request rejected with negative acknowledge code: 3
```
what is looking pretty similar.

philippoo66, 2024-02-11:
I wonder why it's doing a send_diagnostic(). the read is a common service request. I'm afraid udsoncan got confused again...

AnHardt, 2024-02-12:
Sorry.
Verstehe nur 'Bahnhof'. Das sind meine ersten Schritte mit 'open3e' an meiner 'PT2 BSZ'.
Basic, Pascal, C, C++, git, Arduino, 6502, AVR, ST32, ESP32, HA sind mir ziemlich geläufig (wenn auch z.T. etwas eingerostet) aber mit Python und seinen Libraries stehe ich noch ziemlich auf Kriegsfuß.

AnHardt, 2024-02-12:
Wenigstens die IP scheint korrekt zu sein.
Wenn dass, WLAN down ist erhalte  ich 
```
e3@openE3:~/open3e $ python3 Open3E_depictSystem.py -d 10.83.83.1
read DID enums ...
2951 DIDs listed.
scan COB-IDs 0x680 to 0x6ff ...
Traceback (most recent call last):
  File "/home/e3/open3e/Open3E_depictSystem.py", line 284, in <module>
    lstEcus = scan_cobs(startcob, lastcob)
  File "/home/e3/open3e/Open3E_depictSystem.py", line 63, in scan_cobs
    conn = DoIPClientUDSConnector(DoIPClient(args.doip, tx))
  File "/home/e3/.local/lib/python3.9/site-packages/doipclient/client.py", line 191, in __init__
    self._connect()
  File "/home/e3/.local/lib/python3.9/site-packages/doipclient/client.py", line 766, in _connect
    self._tcp_sock.connect((self._ecu_ip_address, self._tcp_port))
TimeoutError: [Errno 110] Connection timed out

````

Normalerweise benutze ich auf 'github' english' (wegen der Reichweite) aber hier scheint die Mehrheit der4 Entwickler Deutsch zu sprechen. Gibt es es bevorzugte Sprache für Issues oder Code (comments)? 

Normally I do use (a kind of ) Englisch on 'github' (to increase the reach). Is there a preferred language for issues and or code? 
h)

philippoo66, 2024-02-12:
lass mal das depictSystem weg und versuch's direkt mit open3e client. les da mal den DP BusIdentification aus

`python3 Open3Eclient.py -d 192.168.xy.za -r 256 -v `

wenn ich nix vergessen hab....

AnHardt, 2024-02-12:
Das scheint zu klappen. 'maser' und 'develop' produzieren die selbe Ausgabe ohne Fehlermeldung.
```
e3@openE3:~/open3e $ python3 Open3Eclient.py -d 10.83.83.1 -r 256 -v 
None 256 BusIdentification {"BusAddress": 1, "BusType": {"ID": 2, "Text": "CanInternal"}, "DeviceProperty": {"ID": 2, "Text": "HMUMASTER"}, "DeviceFunction": {"ID": 2, "Text": "HMUMASTER"}, "SW-Version": "2.513.1932.372", "HW-Version": "2.606.105.2", "VIN": "012_changed_3456"} 
closing 0x680 - bye!
```

philippoo66, 2024-02-12:
@AnHardt 

hier ist ein uraltes bask Skrit mit dem du Datenpunkte scannen könntest (wenn du viel Zeit hast ;-)

```
# DID Scan 
#
# -> candump can0 > candump.log
# -> bash didscan

set -ex
for did in {256..3500}
do
	cansend can0 680#0322$(printf '%04X' $did)00000000
	sleep 1
done
```

dazu vorher einen candump starten mit Filter auf 680, 690

da, wo eine Antwort kommt, ist ein Datenpunkt.

Grüsse!
Phil

AnHardt, 2024-02-13:
Danke für Deine Mühen.
Habe in der '[Open3Edatapoints.py](https://github.com/philippoo66/vitualE3/blob/main/Open3Edatapoints.py)' ein paar der interessanten Einträge gesucht und abgefragt - die meisten warten auf meiner Anlage vorhanden. Nach ein wenig copy/paste in der 'Open3Edatapoints.py' konnte ich Temperaturen und Gas-Statistiken für die FuelCell auch sinnvoll interpretieren lassen.
Das ist erstmal motivierend genug um Can-Adapter, Stecker91, Hutschienengehäuse, Netzteil, ... zu bestellen. 
Bis dahin.

philippoo66, 2024-02-13:
kannst du dein copy&paste Arbeit bitte posten oder noch besser in die 'offizielle' datapoints.py einbauen!?

philippoo66, 2024-02-13:
im Viessmann Forum hat grad wer den CAN Adapter gekauft, aber wohl eine Optolink Anlage. Leider ist der wohl im Ausland angesiedelt (englische Konversation), sonst hättest du ihm den Adapter bestimmt günstig abkaufen können ;-)

MyHomeMyData, 2024-02-16:
@wotandamm Du musst die Bibliotheken updaten:
```
pip3 install -r requirements.txt
```
Anschließend sollte es funktionieren.

AnHardt, 2024-02-21:
Ich bin ein gutes Stück weiter.
Mit den Tipps von Sonntag aus der Diskussion [59](https://github.com/open3e/open3e/discussions/59) habe ich ein depict hin bekommen.

Mit dem neueste 'master'='devolopment' Branch bekam ich immer noch:
```bash
e3@openE3:~/open3e $ python3 Open3E_depictSystem.py -d 10.83.83.1
read DID enums ...
2951 DIDs listed.
scan COB-IDs 0x680 to 0x688 ...
1664
ECU found: 0x680 : HMUMASTER
1665
Traceback (most recent call last):
  File "/home/e3/open3e/Open3E_depictSystem.py", line 77, in scan_cobs
    response = client.send_request(
  File "/home/e3/.local/lib/python3.9/site-packages/udsoncan/client.py", line 2143, in send_request
    self.conn.send(payload)
  File "/home/e3/.local/lib/python3.9/site-packages/udsoncan/connections.py", line 79, in send
    self.specific_send(payload)
  File "/home/e3/.local/lib/python3.9/site-packages/doipclient/connectors.py", line 43, in specific_send
    self._connection.send_diagnostic(bytearray(payload))
  File "/home/e3/.local/lib/python3.9/site-packages/doipclient/client.py", line 720, in send_diagnostic
    raise IOError(
OSError: Diagnostic request rejected with negative acknowledge code: 3

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/e3/open3e/Open3E_depictSystem.py", line 313, in <module>
    lstEcus = scan_cobs(startcob, lastcob)
  File "/home/e3/open3e/Open3E_depictSystem.py", line 96, in scan_cobs
    raise Exception(e)
Exception: Diagnostic request rejected with negative acknowledge code: 3
```
Dafür habe ich 'Open3E_depictSystem.py' folgendermaßen gepatcht: [ip-patch](https://github.com/AnHardt/open3e/pull/1/files#diff-2dd4e6e57c1121ae896e81fd792cdca57f964d76fd5f9ee26e1a0f5d2ed2dbbaL39-L104)

Danach ist ist es durchgelaufen.

Da das Can-Interface inzwischen da und provisorisch installier ist das gleiche noch mal für can.
can-patch

Mein momentaner Stand der Dinge ist auf meinem [PT2_branch](https://github.com/AnHardt/open3e/pull/3) zu sehen. Leider setzt der auf dem 'master' von letzte Woche auf. Das mach ich nochmal in richtig.

philippoo66, 2024-02-21:
Moin @AnHardt !

Erstmal danke! Ich fürchte aber, das ist nicht so richtig die Lösung. Normalerweise kommt, wenn keine ECU an einer Adresse ist, einfach ein Timeout - es antwortet halt nix wo nix ist. Wenn da jetzt was antwortet, aber mit einem seltsamen Fehlercode, scheint da ja was zu sein. Wir sollten die Adresse / die ECU nicht einfach ignorieren.

Kannst du bitte noch mal sagen, wie viele Geräte (und was für welche) an dem Bus hängen?

Pier Yves hat eine 'saubere' Möglichkeit geschaffen, Datenpunkte ohne Codec zu lesen - vielleicht sollte ich das depict mal da drauf umbauen. Mit der bestehenden Variante 'missbrauche' ich eigentlich eine interne Funktion. Vielleicht schafft uns das Probleme, die bei alten Versionen von udsoncan nicht 'aufpoppten'...

AnHardt, 2024-03-04:
Inzwischen habe ich mal im Menü der Anlage nachgeschaut was die Anlage meint an Komponenten zu kennen und abgeschriben. Die aus dem 'depict' bekannten 'Bussname' und 'ecuAdr' habe ich mal dazu _gedichtet_.
Wie kann ich aus der Bussadresse die ecuAdr bestimmen? 
Beim scannen über CAN fliegt mir immer noch die '1671' um die Ohren.
Über IP scanne ich nicht mehr, dazu müsste ich mich ja jedes mal vom Viessmann Server trennen.
```json
{
"DiscoveredDevices" : {

  "HMU-Master" : {
    "Bussadresse": 1,
    "ecuAdr" : "0x680-1664",
    "Busstyp": 2, 
    "Busname" : "CanInternal",
    "Herstellernummer": "*redacted*",
    "SW-Version": "2.513.1932.372",
    "HW-Version": "2.606.105.2"
  },
  "BCU" : {
    "Bussadresse": 50,
    "ecuAdr" : "0x688-1672",
    "Busstyp": 2, 
    "Busname" : "CanInternal",
    "Herstellernummer": "*redacted*",
    "SW-Version": "5.518.1929.111",
    "HW-Version": "1.600.500.1"
  },
  "Gebläse" : {
    "Bussadresse": 60,
    "ecuAdr" : "?",
    "Busstyp": 2, 
    "Busname" : "CanInternal",
    "Herstellernummer": "*redacted*",
    "SW-Version": "0.2.2.0",
    "HW-Version": "0.0.0.0"
  },
  "Brennstoffzelle" : {
    "Bussadresse": 94,
    "ecuAdr" : "?",
    "Busstyp": 3, 
    "Busname" : "CanExternal",
    "Herstellernummer": "*redacted*",
    "SW-Version": "0.1.1.0",
    "HW-Version": "0.0.12.0"
  },
  "HMI" : {
    "Bussadresse": 59,
    "ecuAdr" : "0x684-1668",
    "Busstyp": 2, 
    "Busname" : "CanInternal",
    "Herstellernummer": "*redacted*",
    "SW-Version": "6.525.2016.0",
    "HW-Version": "3.503.303.2"
  },
  "Backend-Gateway" : {
    "Bussadresse": 58,
    "ecuAdr" : "?",
    "Busstyp": 2, 
    "Busname" : "CanInternal",
    "Herstellernummer": "*redacted*",
    "SW-Version": "10.504.2113.2",
    "HW-Version": "10.1.2.0"
  },
  "27" : {
    "Bussadresse": 97,
    "ecuAdr" : "?",
    "Busstyp": 3, 
    "Busname" : "CanExternal",
    "Herstellernummer": "*redacted*",
    "SW-Version": "0.0.2.257",
    "HW-Version": "86.17735.0.27"
  },
  "M2IO" : {
    "Bussadresse": 32,
    "ecuAdr" : "?",
    "Busstyp": 1, 
    "Busname" : "PlusBus",
    "Herstellernummer": "*redacted*",
    "SW-Version": "6.506.1841.62",
    "HW-Version": "65535.1.1.65535"
  }
}
}
```

philippoo66, 2024-03-05:
moin @AnHardt !

> Wie kann ich aus der Bussadresse die ecuAdr bestimmen?

wenn wir das wüssten... da gibt es ein paar Sachen, die wir noch nicht durchblicken.

hier https://github.com/open3e/open3e/discussions/59#discussioncomment-8506111 hatte jemand ein ähnliches Problem beim Scannen. du kannst mal oben im depict (aktuell Zeile 55) die 'Problemadresse' 1671 ausschliessen
`    lstskips = [1671]  `
aber wenn es dann bei einer weiteren hängen bleibt ist das auch keine Lösung (eine Lösung ist es eh nicht).

Irgendwas scheint Viessmann geändert zu haben, so komische Rückgabewerte hatten wir früher nicht...

Grüsse!
Phil

philippoo66, 2024-03-05:
hast du zufällig ioBroker im Einsatz und könntest Jürgens Adapter ausprobieren? der macht auch einen System Scan. Wäre interessant zu wissen, ob der durchläuft...

AnHardt, 2024-03-05:
Die '1671' habe ich schon lange, außer für Tests, ausgeschlossen
Nein. Leider kein 'ioBroker', ich benutze 'Home Assistant'.
Ich vermute bei der '1671' bandelt es sich um die Brennstoffzelle, die ja auch auf dem internen CAN-Buss sein soll. Die ist von Viessmann bei Pannasonic zugekauft und spricht möglicherweise einen asiatischen Dialekt.

Philip-Wiege, 2024-11-04:
Ich glaube, das ursprüngliche Problem wurde erfolgreich behoben.

closed 2024-11-04 by Philip-Wiege