open3e/open3e #87: Fehler bei Abfrage von Datenounkten

pellenbeck, 2024-03-21:
Hallo zusammen,

ich versuche aktuell bei mir das Open3e-Setup zum Laufen zu bekommen, da ich gerne direkt die Daten vom Smart Meter und dem VX3 auslesen möchte.

Derzeit gehe ich per GridX-API (Webfrontend Gridbox) und tibber-API an die Daten, habe aber einen zu großen zeitlichen Versatz drin, dass das Laden an der Wallbox dann doch immer etwas über der aktuellen PV-Leistung liegt.

Mein Setup ist ein VX3 (alte Generation ohne integrierte Gridbox) und ein Viessmann E3100CB Smart Meter, die per CAN-Bus verbunden sind.

Ich habe mich nun parallel dazu aufgeschaltet und nutze den Innomaker USB-2-CAN Adapter (https://www.amazon.de/dp/B09K3LL93Q) an einem eigens aufgesetzt PI.
Soweit läuft Open3e und kann ich einzelne Daten lesen, scheitere aber an der Mehrzahl der Werte. Versuche da ohne konkrete Kenntnis von Open3e zu debuggen, komme aber nicht weiter.

Daher die Frage an dich ob du mir dabei helfen kannst.

Schon beim Ausführen von Open3E_depictSystem.py habe ich erste Fehler bekommen:

philipp@pi-can:~/open3e $ python3 Open3E_depictSystem.py
read DID enums ...
2951 DIDs listed.
scan COB-IDs 0x680 to 0x6ff ...
ECU found: 0x680 : EMCUMASTER
ECU found: 0x684 : HMI
ECU found: 0x68f : BMCU
ECU found: 0x6cc : CSC
ECU found: 0x6cd : CSC
ECU found: 0x6ce : CSC
ECU found: 0x6cf : CSC
Received invalid CAN frame. Received Single Frame with length of 5 while there is room for 0 bytes of data with this configuration
7 responding COB-IDs found.
write devices.json ...
done.
scan 680 DIDs 256 to 3500 ...
found 256:36:BusIdentification
found 257:122:StatusDtcList
found 258:122:StatusDtcHistory
(...)

Euer Beispiel aus der Readme funktioniert:

philipp@pi-can:~/open3e $ python3 Open3Eclient.py -c can0 -dev vx3 -r 1664 -v
vx3 1664 ElectricalEnergyStorageStateOfCharge 96.0
closing 0x680 - bye!

Sowas wieder nicht:

philipp@pi-can:~/open3e $ python3 Open3Eclient.py -c can0 -dev vx3 -a -v
reading 0x680, 189 datapoints, please be patient...
2024-03-21 12:10:07 [ERROR] UdsClient: [InvalidResponseException] : ReadDataByIdentifier service execution returned an invalid response. Value for data identifier 0x01c1 was incomplete according to definition in configuration

Oder auch:

philipp@pi-can:~/open3e $ python3 Open3Eclient.py -c can0 -dev vx3 -r 1660 -v
Traceback (most recent call last):
  File "/home/philipp/open3e/Open3Eclient.py", line 363, in <module>
    readbydid(addr=ecudid[0], did=ecudid[1], raw=args.raw, msglvl=mlvl)
  File "/home/philipp/open3e/Open3Eclient.py", line 218, in readbydid
    value,idstr =  dicEcus[addr].readByDid(did, raw)
  File "/home/philipp/open3e/Open3Eclass.py", line 139, in readByDid
    response = self.uds_client.read_data_by_identifier([did])
  File "/home/philipp/.local/lib/python3.9/site-packages/udsoncan/client.py", line 167, in decorated
    return func(self, *args, **kwargs)
  File "/home/philipp/.local/lib/python3.9/site-packages/udsoncan/client.py", line 466, in read_data_by_identifier
    response = self.send_request(req)
  File "/home/philipp/.local/lib/python3.9/site-packages/udsoncan/client.py", line 2255, in send_request
    raise NegativeResponseException(response)
udsoncan.exceptions.NegativeResponseException: ReadDataByIdentifier service execution returned a negative response RequestOutOfRange (0x31)
 

Hast du eine Idee, was das Problem sein kann?

Zudem die Frage, ob ihr schon Datenpunkt (aktuelle Werte Bezug/Einspeisung etc.) zum Viessmann E3100CB Smart Meter kennt?

Viele Grüße

Philipp

MyHomeMyData, 2024-03-21:
@pellenbeck Hallo Philipp, die Meldung "negative response" bedeutet, dass es diesen Datenpunkt auf Deinem System nicht gibt oder das Gerät nicht bereit ist, die Info zu senden. Meistens ist ersteres der Fall. An open3e liegt es aber nicht.

Das Energy Meter kann open3e nicht verarbeiten. Für das Energy Meter E380 gibt er [hier](https://github.com/MyHomeMyData/E3onCAN) eine Lösung. Ob das auch mit einem E3100CB funktioniert, kann ich Dir nicht sagen. Würde mich aber sehr interessieren, ob das geht.

Falls Du mit ioBroker arbeitest solltest Du dir auch den Adapter [e3oncan](https://github.com/MyHomeMyData/ioBroker.e3oncan) anschauen. Der unterstützt E380 und E3-Geräte (ersetzt also open3e).

MyHomeMyData, 2024-03-21:
open3e und E3onCAN arbeiten grundsätzlich unterschiedlich. open3e fragt über das Protokoll UDSonCAN aktiv Daten bei den Geräten ab. D.h. es wird eine Anfrage auf den CAN-Bus gesendet und auf die Antwort gewartet. Diese wird ausgewertet und per MQTT oder übers Terminal ausgegeben.

E3onCAN hört grundsätzlich auf dem CAN-Bus nur zu. Geräte wie die Vitocharge VX3 oder eben das Energy Meter schicken ständig ihre Daten auf den Bus, auch ohne dass jemand fragt. E3onCAN kennt die verwendeten Protokolle und dekodiert die ankommenden Daten. Die Ergebnisse werden genauso wie open3e per MQTT oder Kommandozeile ausgegeben.

Für das E380 lauscht E3onCAN auf den IDs 0x250 bis 0x25D (ich habe vor ca. 2 Stunden ein Update hochgeladen für eine weitere CAN-Adresse des E380). Wenn auf diesen IDs was kommt, wird es dekodiert und ausgegeben. Wenn da nichts kommt, bleibt E3onCAN für das E380 stumm.

Für mich wäre es interessant, wenn Du für zwei bis drei Minuten einen CAN-Mitschnitt machen könntest. Das ginge so:
* CAN-Utilities installieren: `sudo apt install can-utils`
* Mitschnitt anfertigen: `candump can0 -tA > candump_20240321`
 
Mitschnitt stoppen mit Strg-C. Datei hier posten oder mir direkt per Mail senden.

Zum ioBroker Adapter: Bzgl. CAN-Bus funktioniert der genauso wie open3e oder E3onCAN: Der Bus muss mit ifconfig sichtbar sein. Ob es ein physischer oder ein virtueller CAN-Bus ist, ist egal. Musst Du nur in den Adapter-Einstellungen richtig konfigurieren.
Es sollte also so funktionieren:
* in Docker einen virtuellen CAN einrichten (vcan0)
* auf dem Raspi mit CAN-Adapter und in Docker cannelloni installieren
* auf beiden Maschninen cannelloni starten, so dass die CAN-Daten über TCP/IP getunnelt werden
* ich habe das schon auf verschiedenen Rechnern und sogar übers Internet (VPN) erfolgreich getestet. Das wird nicht ganz so zuverlässig laufen wie beim Zugriff auf den physischen CAN, aber sehr wahrscheinlich wirst Du nichts davon merken.

Infos zu cannelloni findest Du hier: https://github.com/mguentner/cannelloni

Auch in den [Diskussionen zu open3e](https://github.com/open3e/open3e/discussions) gibt es weiterführende Infos dazu.

pellenbeck, 2024-03-21:
Danke, das hat mir den Abend erheitert!
cannelloni ist ein super Name für einen CAN Tunnel! :-)

Ich habe testweise ioBroker auf dem PI installiert und den E3onCAN-Adapter getestet: funktioniert und er ließt die VX3-Daten aus.
Allerdings werden die Werte nicht aktualisiert. Auch mit gestartetem "Sammler" (Log "Collect worker started on EMCUMASTER_0x680").

Die Werte für den E380 bleiben leer. Ich werde mal, wie von dir beschrieben, einen Dump erstellen und hier posten.

pellenbeck, 2024-03-21:
Dump anbei.
[dump.txt](https://github.com/open3e/open3e/files/14711531/dump.txt)

MyHomeMyData, 2024-03-22:
@pellenbeck In der Log-Datei sehe ich folgendes:
* Es kommen viele Messages auf den IDs 568 und 569. Da sendet vermutlich das E3100CB. Das Datenformat ist mir noch unklar.
* Davon abgesehen kommt nicht viel. Von der VX3 scheint gar nichts zu kommen.

Damit ist klar, dass Du beim Lauschen keine Daten siehst. Für den E380 wird auf IDs 592 bis 605 gelauscht, da kommt aber nichts. Von der VX3 kommt gar nichts.

**Hast Du eine Anleitung für den E3100CB?** Da sollte ja beschrieben sein, welche Daten des Ding überhaupt bereit stellt. Ich könnte auf dieser Basis dann eine Testversion von E3onCAN für die IDs 568 und 569 erstellen.

Zum ioBroker Adapter: Auch da kann mit "Collect" nichts empfangen werden. Du müsstest aber die Daten abfragen können. Dazu musst Du im Konfigurations-Tab „Zuweisungen zum externen CAN-Adapter“ im Abschnitt "... UDSonCAN" einen oder mehrere Zeitpläne für EMCUMASTER_0x680 erstellen und jeweils eine Liste der Datenpunkte eintragen, die Du abfragen möchtest.

pr0vieh, 2024-03-22:
Ich habe das gleiche Phänomen im E3onCAN bekomme ich nichts...
muss ich dafür am CanBus Zwischen VX3 und E380 parallel hängen oder recht da auch der Anschluss am Ext2 des VX3s ?

MyHomeMyData, 2024-03-22:
@pr0vieh: Leider haben wir nur wenige gesicherte Erkenntnisse über das Kommunikationsverhalten bei verschiedenen Systemkonfigurationen. Viessmann ist da ja sehr zurückhaltend mit Informationen - um es mal vorsichtig auszudrücken.

Nach meiner Einschätzung sollte Ext2 und Ext1 gleich funktionieren. Bei mir ist jedenfalls das E380 auf Ext1 und die Vitocal auf Ext2 angeschlossen. Ob das aber auch ohne Vitocal funktioniert weiß ich nicht.

Einigermaßen gesichert ist:
* Sind VX3 und Vitocal nicht über CAN verbunden, senden beide Geräte wenig bis fast nichts auf den CAN-Bus.
* Sind beide Geräte verbunden aber nicht als Master/Slave konfiguriert (das kann nur der HB bzw. Viessmann), werfen beide Geräte Fehlermeldungen. Die kann man zwar ignorieren, aber sinnvoll erscheint mir das nicht. Die VX3 sendet aber dann einiges an Daten. Die Vitocal sendet trotzdem sehr wenig bis fast nichts.
* Sind beide Geräte verbunden und korrekt konfiguriert ergibt sich das gleiche Bild. Natürlich bleiben die Fehlermeldungen aus.
* Die Vitocal ist sehr gesprächig auf dem internen CAN-Bus.
* Über UDSonCAN antworten beide Geräte, sowohl mit als auch ohne Verbindung untereinander. Sind die Geräte nicht verbunden, braucht man dazu 2 Adapter. Die gewünschten Datenpunkte muss man aktiv abfragen.
* Das Energy Meter sendet ständig sehr viele Daten, spricht aber kein UDSonCAN.

Die Programme unterstützen folgendes:
* E3onCAN hört generell nur zu und kann kein UDSonCAN. E380 wird dekodiert.
* open3e kann UDSonCAN, aber keinen Lauschbetrieb., dekodiert also auch nicht das Energy Meter.
* Der ioBroker-Adapter e3oncan kann UDSonCAN und Zuhören. Dekodiert auch das Energy Meter E380.

pr0vieh, 2024-03-22:
@MyHomeMyData danke für die schnelle Antwort
ich habe kein Vitocal...
Daher ist am VX3 der E380 auf Ex1 und der Can2USB auf Ext2
Deswegen kann der auslesen aber nix Abhören da auf Ex2 ja nix weiter hängt was sich unterhält daher die Idee mich Parallel auf den Ex1 zuhängen wo sich ja VX3 und E380 unterhalten sollten...

ich werde das mal um klemmen und schauen ob die Kommunikation dann zusammenbricht oder weiter läuft dann könnte ich ggf. 
mal nen dump ziehen

pr0vieh, 2024-03-22:
@MyHomeMyData macht es vielleicht sinn die beiden "Tools" zusammen zu führen?
Ein "Lausch Deamon" mit UDSonCan Collector um eine Kommunikation zu emulieren ?
Leider bin ich erst vor 2 Tagen in das Thema eingestiegen und muss erstmal noch ein wenig verstehen ;) danke auf jeden fall für deine Erklärung

MyHomeMyData, 2024-03-22:
@pr0vieh Ich würde mal mit Radio Eriwan antworten: Im Prinzip ja, aber ...
* nach meiner Einschätzung nutzen die meisten Anwender nur UDSonCAN
* für E380 und/oder VX3 steht E3onCAN zur Verfügung. Beide Tools kann man problemlos parallel betreiben
* wenn wir "collect" in open3e einbauen, wird die Liste der Optionen noch unübersichtlicher. Das ist jetzt schon eine Herausforderung.
* Für ioBroker gibt es die integrierte Lösung bereits

Ich fände es deshalb besser, eine Integration für HA zu entwickeln. Vielleicht findet sich ja jemand. Ich habe von HA leider zu wenig Ahnung. Der Aufwand müsste aber überschaubar sein. Das wird doch auch in Python entwickelt, oder?

pr0vieh, 2024-03-22:
Python auf jeden Fall allerdings stecke ich in HA auch erst seit 3 Tagen drin 😉
Ich durchschaue gerade die docker Problematik noch nicht so ganz also AddOn = eigener Docker Container...
Es gibt bei HACS pyscript und AppDeamon die eine dauerhafte Ausführung von Python Scripten ermöglichen oder noch was anderes 🤔?

pellenbeck, 2024-03-22:
> * Es kommen viele Messages auf den IDs 568 und 569. Da sendet vermutlich das E3100CB. Das Datenformat ist mir noch unklar.

Ja, das habe ich auch gesehen. Das wird vermutlich die Summe des Bezugs bzw. der Einspeisung sein. Den Mitschnitt habe ich zu einer Zeit gemacht, als der VX3 versucht hat den aktuellen externen Bezug auf 0 zu halten - was ein ständiges +/- von ein paar Watt bedeutet.

> * Davon abgesehen kommt nicht viel. Von der VX3 scheint gar nichts zu kommen.
> Damit ist klar, dass Du beim Lauschen keine Daten siehst. Für den E380 wird auf IDs 592 bis 605 gelauscht, da kommt aber nichts. Von der VX3 kommt gar nichts.

Als ich den ioBroker Adapter eingerichtet habe und der alle Datenpunkte abgescant hat, habe ich ja aber diverse Werte bekommen.
Ich habe jetzt im Adapter unter "Anfordern von Daten auf einem externen CAN-Adapter über UDSonCAN" ein paar der interessanten Datenpunkte mit 10sec-Timer konfiguriert und bekomme dort jetzt auch immer die aktuelle Werte.
 
> **Hast Du eine Anleitung für den E3100CB?** Da sollte ja beschrieben sein, welche Daten des Ding überhaupt bereit stellt. Ich könnte auf dieser Basis dann eine Testversion von E3onCAN für die IDs 568 und 569 erstellen.

Ich habe leider nicht mehr als die Montage- und Serviceanleitung:
https://static.viessmann.com/resources/technical_documents/DE/de/VSA/6154316VSA00001_1.pdf?#pagemode=bookmarks&zoom=page-fit&view=Fit

pellenbeck, 2024-03-22:
> Nach meiner Einschätzung sollte Ext2 und Ext1 gleich funktionieren. Bei mir ist jedenfalls das E380 auf Ext1 und die Vitocal auf Ext2 angeschlossen. Ob das aber auch ohne Vitocal funktioniert weiß ich nicht.

Mein VX3 hat lediglich eine CAN-Bus-Klemme ("CAN1 extern"). Habe mich daher mit dem USB-Adapter parallel auf die Klemme gesetzt (ohne zusätzliche Terminierung).

pellenbeck, 2024-03-23:
@MyHomeMyData ich habe jetzt mal zwei Dumps à 5 Minuten gemacht und von den gleichen 5 Minuten einen Screenshot der tibber-Werte gemacht. Hilft dir das?
[image]
[candump_20240323_Einspeisung.txt](https://github.com/open3e/open3e/files/14731962/candump_20240323_Einspeisung.txt)
[candump_20240323_WolkenWallbox.txt](https://github.com/open3e/open3e/files/14731963/candump_20240323_WolkenWallbox.txt)
[image]

MyHomeMyData, 2024-03-23:
@pellenbeck Ich habe eine Vermutung:

Auszug aus einem Dump:
```
 (2024-03-23 14:12:42.412115)  can0  569   [8]  23 0A 4E 01 DB 85 8F 4B
 (2024-03-23 14:12:42.412239)  can0  568   [8]  60 0A 4E 01 00 00 00 00
 (2024-03-23 14:12:42.419864)  can0  569   [8]  23 0A 4E 02 78 A5 E1 4A
 (2024-03-23 14:12:42.421238)  can0  568   [8]  60 0A 4E 02 00 00 00 00
 (2024-03-23 14:12:42.427617)  can0  569   [8]  2F 0A 4E 03 00 00 00 00
 (2024-03-23 14:12:42.428991)  can0  568   [8]  60 0A 4E 03 00 00 00 00
```
Bei den ersten 3 Datenbytes habe ich noch keine Idee.
Das 4te Byte scheint ein Code zu sein, der die Bedeutung der folgenden 4 Bytes beschreibt. Dieser Code läuft ständig von 0x01 bis 0x11 durch.
Beispiel ID 569, Code 04:
```
(2024-03-23	14:12:42.436614)	can0	569	[8]	2B	0A	4E	04	04	1D	00	00
(2024-03-23	14:12:42.614621)	can0	569	[8]	2B	0A	4E	04	04	1D	00	00
(2024-03-23	14:12:42.762621)	can0	569	[8]	2B	0A	4E	04	AC	1D	00	00
(2024-03-23	14:12:42.905621)	can0	569	[8]	2B	0A	4E	04	AC	1D	00	00
(2024-03-23	14:12:43.045368)	can0	569	[8]	2B	0A	4E	04	01	1E	00	00
(2024-03-23	14:12:43.185868)	can0	569	[8]	2B	0A	4E	04	44	1E	00	00
(2024-03-23	14:12:43.333992)	can0	569	[8]	2B	0A	4E	04	65	1E	00	00
(2024-03-23	14:12:43.473242)	can0	569	[8]	2B	0A	4E	04	84	1E	00	00
(2024-03-23	14:12:43.656240)	can0	569	[8]	2B	0A	4E	04	A2	1E	00	00
```
Die beiden Bytes nach Code 04 könnten ein Leistungswert sein (LSB first).

Ich erweitere in den nächsten Tagen mal E3onCAN um diese Art der Kodierung. Dann sieht man vielleicht recht schnell, wie die Werte zugeordnet werden können.

pellenbeck, 2024-03-23:
@MyHomeMyData wäre es grundsätzlich möglich e3oncan um ein Script zu erweitern, dass die UDSonCAN-Abfrage vom ioBroker-Adapter noch ergänzt? Also nicht nur "collect" sondern auch ein "request & read" auf angegeben Datenpunkte macht?
Dann könnte ich auf den ioBroker verzichten und die benötigten VX3-Daten direkt per MQTT weiterleiten.

Btw. der EMCUMASTER 0x680.1603 (e3oncan.0.EMCUMASTER_0x680.tree.1603_PointOfCommonCouplingPower) hat den benötigten aktuelle Wert vom Smart Meter in Watt.

MyHomeMyData, 2024-03-23:
> wäre es grundsätzlich möglich e3oncan um ein Script zu erweitern, dass die UDSonCAN-Abfrage vom ioBroker-Adapter noch ergänzt?

Klar, könnte man machen. Aber dann baue ich open3e nach. open3e kann das schon und man kann ja beide Tools einfach parallel betreiben.

> Btw. der EMCUMASTER 0x680.1603 (e3oncan.0.EMCUMASTER_0x680.tree.1603_PointOfCommonCouplingPower) hat den benötigten aktuelle Wert vom Smart Meter in Watt.

Ja, den Datenpunkt nutze ich auch.

pellenbeck, 2024-03-24:
@MyHomeMyData 

> > wäre es grundsätzlich möglich e3oncan um ein Script zu erweitern, dass die UDSonCAN-Abfrage vom ioBroker-Adapter noch ergänzt?
> 
> Klar, könnte man machen. Aber dann baue ich open3e nach. open3e kann das schon und man kann ja beide Tools einfach parallel betreiben.

Nur mit dem Unterschied, dass open3e bei mir scheinbar nicht so recht funktionieren will :-)

MyHomeMyData, 2024-03-24:
> Nur mit dem Unterschied, dass open3e bei mir scheinbar nicht so recht funktionieren will :-)

Gibt es Dinge bzw. Datenpunkte, die mit dem Adapter funktionieren und mit open3e nicht? Dann sollten wir das in open3e in Ordnung bringen.

Hatte gedacht, das Problem sei z.B. 1660 gewesen, da kommt halt "negative response". Das dürfte aber beim Adapter genauso sein. Das Gerät kennt den Datenpunkt nicht oder will ihn nicht senden. Gibt es noch andere Lücken?

MyHomeMyData, 2024-03-24:
@pellenbeck Ich habe E3onCAN erweitert, so dass die CAN-IDs 1384 und 1385 ausgewertet werden können. Die Datenpunktbeschreibungen habe ich generisch gehalten, für jeden Datenpunkt werden die letzten 4 Bytes als zwei 16-Bit-Werte und zusätzlich als ein 32-Bit-Wert gewandelt. Immer signed=True und scale=1.

Die Version liegt im Branch [develop](https://github.com/MyHomeMyData/E3onCAN/tree/main).

Mit `python3 E3onCANcollect.py -c can0 -dev e3100cb -v` kannst Du einfach alles ausgeben lassen, das ist aber sehr unübersichtlich. Mit `python3 E3onCANcollect.py -c can0 -dev e3100cb -m localhost:1883:e3oncan` sendest Du alles an den lokalen MQTT-Broker. Mit dem MQTT Explorer kann man sich da ganz gut eine Übersicht verschaffen und Diagramme für die einzelnen Werte anzeigen.

Mit `python3 E3onCANcollect.py -c can0 -dev e3100cb -v -r 1385.32.1` kannst Du einen einzelnen Datenpunkt ausgeben lassen. Hier also ID 1385 mit Byte4==1 als 32-Bit-Wert. Für 16-Bit-Werte 32 durch 16 ersetzen. Die Werte für Byte 4 gehen von 1 bis 17.

Ideal wäre, wenn Du konkrete Werte des E3100CB bestimmten Datenpunkten zuordnen könntest. 1385.32.1 und 1385.32.2 sind z.B. Zähler, aber welche? Für ID 1384 habe ich nur 0-Werte gefunden. Ziemlich seltsam.
Wenn Du Zuordnungen findest, kannst Du die direkt in `E3onCANdatapointsE3100CB.py` anpassen. Die Logik sollte ersichtlich sein. Der Schlüssel des Datenpunktes darf nicht geändert werden. Nach einen Änderung das Programm neu starten, dann sollte die Änderung wirksam werden.

Die Version des Programm ist nur grob getestet. Da ich keinen E3100CB habe, bin ich da in den Möglichkeiten etwas eingeschränkt. Hoffe, es schmiert nicht zu häufig ab.

pellenbeck, 2024-03-25:
Habe mal angefangen erstmal alle Werte nach MQTT zu schieben und geschaut wo sich Werte ändern und habe diese in Graphe laufen lassen.

Der Wert 1385.16.4 entspricht dem vom VX3 (0x680.1603) und von tibber gelieferten aktuellen Verbrauchswerten. Denke bei 2-3 weiteren Werten sind die einzelnen Phasen und teilweise dann auch die Blindleistungen etc..
Ich bleibe dran ...

pellenbeck, 2024-03-25:
@MyHomeMyData ich habe die Daten nun mal den Tag über analysiert und habe folgendes herausgefunden:

1385.1	32Bit	GridEnergy Comulated Import (Scale?)
1385.2	32Bit	GridEnergy Comulated Export (Scale?)
1385.3	konstant 4
1385.4	16Bit	GridActivePower Total
1385.5	stark schwankend, 3-fache Sinuskurven
1385.6	ganze Zahl zwischen 2 & 8
1285.7	32 Bit	GridVoltage L1 (ohne Nachkommastelle!)
1385.8	16 Bit	GridActivePower L1
1385.9	16 Bit	GridReactivePower L1
1385.10	ganze Zahl zwischen 4 & 13
1285.11	32 Bit	GridVoltage L2 (ohne Nachkommastelle!)
1385.12	16 Bit	GridActivePower L2
1385.13	16 Bit	GridReactivePower L2 
1385.14	ganze Zahl zwischen 24 & 30
1285.15	32 Bit	GridVoltage L3 (ohne Nachkommastelle!)
1385.16	16 Bit	GridActivePower L3
1385.17	16 Bit	GridReactivePower L3

Was ich noch nicht ganz verstehe:
1. Bei den kumulierten Werten ist mir die Skala noch nicht ganz klar - finde die Werte sonst im Zähler selbst nicht.
2. Die Voltage-Angaben sind ohne Nachkommastellen, was ich komisch finde.
3. Was könnte der Wert 1385.5 sein? Habe dazu mal einen Screenshot des Graphen im MQTT Explorer angehangen.
4. Die Werte 1385.3, 1385.5, 1385.6, 1385.10, 1385.14 beobachte ich weiter über den Tag/die Tage.

[image]

MyHomeMyData, 2024-03-25:
@pellenbeck Wow, Du kommst ja zügig voran.
Beim E380 sind die Spannungen auch scale=1 in Volt und die Ströme (GridCurrent) ebenfalls scale=1 in Ampere. Das ist echt völlig idiotisch.
16 Bit Leistungen kommen in Watt, 32 Bit Leistungen in 0,1 Watt, also scale=10. 32 Bit Zähler gibt es mehrere, die einen habe ich mit scale=100 identifiziert, die anderen mit scale=2000. Die Absolutwerte scheinen jeweils losgelöst zu sein von den Anzeigewerten. Aber die Differenzen haben bei mir sehr gut zu selbst aufintegrierten Leistungswerten gepasst. Dazu gibt es auch eine längere Diskussion [hier](https://github.com/MyHomeMyData/E3onCAN/discussions/1).

Zu 1385.32.5 habe ich keine Idee. Das sieht aus als wären es 3 Werte, jeweils 120° phasenverschoben. Sehr seltsam.

Ich bin grade ziemlich ausgeknockt. Seit gestern Abend Fieber, seit heute Morgen heftiger Durchfall. Und es will noch nicht besser werden ...

pellenbeck, 2024-03-25:
@MyHomeMyData, gute Besserung :-)

Die Werte
- 1385.6
- 1385.10
- 1385.14
könnten tatsächlich die Ampere pro Phase sein, allerdings als Integer und nicht als Float - was mich sehr irritiert und nicht richtig sein kann. Kann das ein Fehler im Python-Script sein? Kenne mich da nicht wirklich mit aus.

MyHomeMyData, 2024-04-06:
@pellenbeck Wie ist der Stand der Forschung?

pellenbeck, 2024-04-07:
@MyHomeMyData eigentlich hatte ich ja gehofft, dass du eine Idee hat, warum die Stromstärke nur in ganzen Zahlen kommt (bzw. ausgegeben wird).
Habe daher in den letzten Tag noch nicht weiter geprüft. Die übrigen Werte (siehe oben) sind ja ansonsten klar. 👍

MyHomeMyData, 2024-04-07:
@pellenbeck Ok, ich schaue mir das in den nächsten Tagen mal an meinem E380 an, das dürfte ja gleich sein. Im Moment sehe ich da auch nur ganzzahlige Werte. Bitte poste Deine Version von `E3onCANdatapointsE3100CB.py`, dann kann ich die Definitionen für die identifizierten Werte einfacher übernehmen.

MyHomeMyData, 2024-04-16:
@pellenbeck Habe Deine Erkenntnisse jetzt eingepflegt. Verfügbar im Branch develop. Die Datenpunkte sind als "1385.x" verfügbar, also z.B. 1385.1 für "GridEnergy Comulated". 1384 habe ich erstmal komplett ausgeblendet.

Die Daten zu "GridEnergy Comulated" kommen wohl als Float32-Werte. Das hat @M4n197 für den E380 rausgefunden und es scheint hier auch zu funktionieren.

1385.5 habe ich mal als 2 Float16-Werte dekodiert. Da kommen zumindest vernünftige Zahlenwerte raus. Was das sein soll, ist mir nicht klar.

Bitte ausprobieren. Ich kann das hier nur mit den Log-Dateien testen.

pellenbeck, 2024-04-17:
@MyHomeMyData wow, super. Vielen Dank!
Habe mir die neuste Version gezogen und etwas experimentiert. Folgend drei Screenshots aus meinem MQTT Explorer.

Ich habe in rot die aus den Daten berechneten Ampere berechnet, die ja sehr nah (auf- und abgerundet) zu den Werten von 1385.6, 1385.10 und 1385.14 passen - aber eben gerundet auf eine volle Zahl. Habe beim Spielen mit Int, Float, signed/unsigned, 4 oder 8 Bit etc. keine Lösung gefunden. Hast du eine Idee?

Die Werte für 1385.1 und 1385.2 sehen gut aus und steigen je nach Bezug und Einspeisung. Auch wenn die Werte selbst keinen Bezug zu den Werten auf dem Display des Zählers haben - das muss ich nochmal genauer überprüfen.

1385.5 ist für mich weiter ein Rätsel.
[image]
[image]
[image]

pellenbeck, 2024-04-17:
@MyHomeMyData war gut heute nochmal drauf zu schauen und die Graphen mitlaufen zu lassen.
Ich konnte nun feststellen, dass 1385.3 etwas mit dem Bezug/der Einspeisung bzw. der Summe der Phasen zu tun hat:

Ist die Summe (GridActivePower.Total) < 0 (= Einspeisung), ist der Wert für 1385.3 = 4 (bzw. "04000000").
Ist die Summe (GridActivePower.Total) > 0 (= BEZUG), ist der Wert für 1385.3 = 0 (bzw. "00000000").

[image]

pellenbeck, 2024-04-17:
@MyHomeMyData dann sieht es jetzt so aus.

[image]

Die 1385.5 analysiere ich immer noch. Habe sie aktuell wieder als `E3Int32(2, "Hex-2-signed", scale=1.0, signed=False)` konfiguriert. Da tritt wieder dieses Helix-Muster auf. Es schwank immer in Wellen, läuft auf 65535 (2 Byte) und wird dann wieder klein bzw. fängt von vorne an. Könnte es auch ein Zähler sein? Meine ganz am Anfang auch bei den kWh-Zählern Helix-Muster gesehen zu haben ...

MyHomeMyData, 2024-04-18:
@pellenbeck Schau Dir mal [diesen Beitrag](https://github.com/MyHomeMyData/E3onCAN/discussions/5) an. Aus meiner Sicht und mit Deinen beiden Log-Dateien sieht das gut und stimmig aus. Habe es so im Branch develop übernommen.

Würdest Du es bitte ausprobieren?

MyHomeMyData, 2024-04-20:
Implementierung für Energy Meter E3100CB in [E3onCAN](https://github.com/MyHomeMyData/E3onCAN) ist umgesetzt.

closed 2024-04-20 by MyHomeMyData