open3e/open3e #88: HA oder HACS Integration?

pr0vieh, 2024-03-22:
Ich glaube das Projekt würde viel Zulauf bekommen wenn es eine Home Assistant Community Store (HACS) Integration geben würde.
Zusammen mit dem E3onCan Projekt.

Leider bin ich gerade neu auf HA weiß gerade nicht so recht wie ich Script (Autostart usw. ) das sinnvoll in HA integrieren kann.

jwende, 2024-03-22:
Wir haben jetzt die Visualisierung in einer ersten Variante für HA am Laufen - das setzt eine laufende Instanz von open3e mit einer definierten Liste von DID's voraus. 
Die nächsten Schritte könnte eine Schnittstelle zur dynamischen und zeitgesteuerten Abfrage von DID's sein. 
... über eine HACS Integration würde ich der Zeit nur träumen ...
J.

PS: schaust du hier: https://github.com/MyHomeMyData/ha.vis.vitocal250

Philip-Wiege, 2024-03-22:
@pr0vieh Wie stellst du dir denn im besten Falle die HA Integration vor? Würdest du den HA Host z.B. Raspberry Pi direkt per USB CAN Adapter verbinden wollen oder weiter einen Raspberry Pi und MQTT verwenden?

Im Grunde genommen kann man jetzt auch schon recht einfach die YAML Dateien importieren und bekommt so die DIDs der Vitocal als Sensor Entitäten. Eine weitere Verbesserung könnte dann z.B. sein, dass in den YAML Dateien direkt fortgeschrittene Entitäten wie zum Beispiel ein "Thermostat" zum setzen der WW Temperatur definiert wird.

Das nächste Level wäre dann tatsächlich eine Integration, welche die Entitäten und Geräte bereitstellt und einfach per Klick installiert wird. Die Frage ist nur, was das Minimum Gerüst einer Integration leisten muss.

Grüße
Philip

TheSmartGerman, 2024-03-22:
Minimal wäre ein MQTT Discovery: https://www.home-assistant.io/integrations/mqtt/
schon eine große  Erleichterung. Hierbei würde das händischen Anlegen von Sensoren in HA entfallen.

Beim Discovery wird über MQTT das "Format" des Sensors über MQTT in einer initalien Nachricht mitgeteilt.

jwende, 2024-03-22:
Würde aber imho bedeuten, dass am MQTT Teil von open3e etwas geändert werden muss - oder wer soll die MetaDaten senden?

pr0vieh, 2024-03-22:
Also bei mir hängt der Can2USB direkt am Raspi wo auch HA drauf läuft...

jwende, 2024-03-22:
Dann könnte man vielleicht den ganzen open3e clonen und als custom integration in den HA hängen... allerdings sind die custom extensions alle im docker container - oder? 
J.

rsch90, 2024-03-22:
@TheSmartGerman @jwende 
Ich hatte hier mal was zum Thema MQTT Discovery selber gebaut.

https://github.com/open3e/open3e/wiki/090-Homeassistant#mqtt-und-homeassistant-konfiguration-f%C3%BCr-vx3

Ggf. hilft das schon?:)

jwende, 2024-03-22:
Nice ... hatte ich gar nicht gesehen. Sehr schick. 
J.

Wolfgang-03, 2024-03-23:
@pr0vieh 
> Ich glaube das Projekt würde viel Zulauf bekommen wenn es eine Home Assistant Community 
> Store (HACS) Integration geben würde.

Da stimme ich absolut zu!

Ich habe meine Vitocal 250 HA-Integration am laufen [Siehe hier](https://github.com/open3e/open3e/discussions/54#discussioncomment-8112269). Inzwischen gibt es auch Controls für die Heizkreispumpen, mit welchen man die Leistung der Pumpen anpassen kann. Der aktuelle Stand meiner Integration ist hier zu finden:
 
[Open3E-HA](https://github.com/Wolfgang-03/Open3E-HA/tree/main)
[Open3E-HA - Visualisierung](https://github.com/Wolfgang-03/Open3E-HA/blob/main/installation.md#visualisierung-in-home-assistant)

Im Moment habe ich noch eine separate VM mit Debian für Open3e laufen. Mein Ziel wäre es Open3e direkt in HA (HACS) laufen zu lassen. Soweit ich mich bis heute eingelesen haben sollte das im Prinzip möglich sein. 

Der USB2CAN-Adapter kommt direkt an den HA-Host. Ja, die "custom integration (HACS)" würde innerhalb von HA dann als "docker container" laufen. Ist aber soweit ich gesehen habe unproblematisch, da das innerhalb von HA "out of the box" geht. Wir müssten die Open3e-Python-Scripte etwas anpassen, dann sollten auch "dynamischen und zeitgesteuerten Abfrage von DID's" möglich sein. Mit einem eigenen Decipt-Script für HA könnten auch Sensoren und Controls in HA, in Abhängigkeit der vorhandenen Geräte und damit DIDs angelegt werden. 

Ich würde auf jeden Fall meine Hilfe/Mitwirkung bei so einem Projekt anbieten. 

Grüße
Wolfgang

fabian727, 2024-05-03:
Ist eine MQTT Erkennung von Hass von Interesse?

published man in den topic bereich von /homeassistent etwas, wird das dort automatisch angelegt und erkannt.
Ich hatte mit einer Implementierung schon angefangen, nachdem es aber schon fertige yamls gibt wieder abgebrochen.

Durch den YAML Import weiß ich jedoch nicht welche Daten mir "fehlen".

Es würde das Thema vereinfachen, wenn die option "mfstr" wegfallen würde und ein fixes schema vorgegeben werden würde.

Mit einem weiteren Argument "-hass" / "--homeassistant" würde ich dann eine json auslesen und sämtliche Themen an home assistant als sensoren (read only) bekannt geben. So war meine erste Idee.

Der Code:
`    elif(args.homeassistant != None):
        mqtt_client.on_message = publish2homeassistant
        file=open("HomeAssistantDatapoints.json")
        text=file.read()
        allDevices=json.loads(text)

        #TODO: currently hard coded, could be set via json?
        deviceName="Viessmann Heatpump"
        canBaseId="680"

        #TODO: V is just for Viessman to make it more unique within HomeAssistant; could also be O3E but i don't like the O or is it 0?
        deviceIdentifier="V"+canBaseId
        for attribute, entity in allDevices.items():
            #is it a real entry or "just" versioning (last entry within json)
            if attribute != "Version":
                entityConfig={
                "name":entity["id"],
                #TODO: this state topic is currently hard coded to fit an "mfstr" of "{ecuAddr:03X}/{didName}"
                "state_topic":"open3e/"+canBaseId+"/"+entity["id"],
                "unique_id":deviceIdentifier+"_"+entity["id"],
                "device":{
                    "identifiers":[
                        deviceIdentifier
                    ],
                    "name":deviceName
                }
                }
                if "icon" in entity:
                    icon = {"icon":entity["icon"]}
                    entityConfig.update(icon)
                if "device_class" in entity:
                    device_class = {"device_class":entity["device_class"]}
                    entityConfig.update(device_class)
                if "unit_of_measurement" in entity:
                    unit = {"unit_of_measurement":entity["unit_of_measurement"]}
                    entityConfig.update(unit)

                print(entityConfig)
                #TODO only sensor is currently supported!, but there is also:  button, camera, cover, number, select and I guess more (add to json?)
                deviceType = "sensor"
                topic="homeassistant/"+deviceType+"/"+deviceIdentifier+"/"+deviceIdentifier+"_"+deviceType+"_"+entity["id"]+"/config"
                mqtt_client.publish(topic, json.dumps(entityConfig), retain=True)
        time.sleep(3)
`

eine JSON:
device class, icon and unit_of_measurement are optional
`{
  "2352": {
    "codec": "RawCodec",
    "len": 2,
    "id": "AdditionalElectricHeater",
    "args": {},
    "device_class": "energy",
    "icon": "mdi:lightning-bolt",
    "unit_of_measurement": "kW"
  },
  "Version": "20240303"
}`

Interesse? Dann würde ich es fertig schreiben

TheSmartGerman, 2024-05-06:
@fabian727 gefällt mir, ja es gibt die fertigen yamls. Allerdings hat je doch jeder andere interessen / jede Anlagen unterscheidet sich ein wenig.
Ich plane akutell auch ein wenig im HA aufzuräumen, sobald jetzt dann die Heizung nicht mehr benötigt wird. Bisher war ich im Modus "never change a running system". Auf die nächste Heizperiode wäre dann genug Zeit ein wenig zu testen... ;)

Wolfgang-03, 2024-05-06:
@fabian727 das ist eine Interessante Alternative. Allerdings glaube ich auch, wie @TheSmartGerman, dass das schwierig wird, da jede Anlage eine unterschiedliche Konfiguration hat. Meine Überlegungen gehen in die Richtung, dass ich eine richtige HACS-Integration schreiben möchte, welche dann selbst eine Discovery-Modus alla "Open3E_depictSystem.py" hat, um zu erkennen welche Geräte (z.B. Pumpen, Mischventile, etc.) und welche Sensoren (Temperatur, Volumenstrom, etc.) bei der Anlage aktiv sind und diese dann als Entities zu veröffentlichen. 

Ein Vorteil, den ich mir davon verspreche, wäre, dass dann auch die Abfrage der DID's unterschiedlich häufig erfolgen könnte. Über den "Umweg" Open3E -> MQTT -> HASS werden alle DID's (bei mir) alle 30 Sekunden abgefragt, was bei einigen DID's zu wenig und bei anderen zu viel ist (Meine Meinung). 

Zum Beispiel bei meiner PV-Anlage, welche an HASS über Modbus angebunden ist, kann ich in der YAML angeben, wie oft welcher Wert abgefragt werden soll. 

```
  modbus:
      - name: E3DC Firmware
        unique_id: e3dc_firmware
        address: 40051
        data_type: string
        count: 8
        scan_interval: 86400

      - name: E3DC Solar Power
        unique_id: e3dc_solar_power
        unit_of_measurement: W
        data_type: int16
        address: 40067
        device_class: power
        scan_interval: 5
```

Wenn man das so ähnlich für Open3E umgesetzt bekäme, dann könnte eventuell der "User" selbst in einer YAML-Konfiguration angeben, welche Entitäten für ihn relevant sind und wie oft er die Werte im HASS aktualisiert haben möchte. 

Aktuell habe ich meine Vitocal 250-A, zu 90% im Griff, damit sie macht, was ich möchte. Zwar noch mit ein paar Einschränkungen, die mir nicht gefallen, aber daran kann man noch arbeiten. Zum Beispiel, kann ich (bei mir) die Zirkulationspumpe für den Heizkreis 1 (HK1) nur dadurch regulieren, dass ich den HK1 abschalte. Ich kann zwar den Wert für 1102_MixerOneCircuitPumpMinimumMaximumLimit in der Anlage ändern, aber leider hat das keine Auswirkung auf die Pumpe, die läuft einfach mit 100% oder gar nicht. 

Ich habe mir jetzt ein Programm (Script) geschrieben, mit welchem ich tagsüber einen PV-Modus simuliere, d.h. wenn viel Strom vom Dach kommt, dann schalte ich die Temperatur von HK1 schrittweise hoch bis zum Maximum (37 Grad), was zur Folge hat, dass die Temperatur im Pufferspeicher auf 50-60 Grad (regulierbar) ansteigt. Dann lasse ich vom Script den HK1 ausschalten und damit auch die Pumpe vom HK1, womit die Temperatur im Pufferspeicher konserviert wird. Abends wenn wieder Wärme im Haus benötigt wird, schalte ich den HK1 jede Stunden für 15 Minuten an, damit das warme Wasser aus dem Pufferspeicher im Haus verteilt wird. Das sieht dann in etwa so aus: 

[image]

Damit schaffe ich es, mit dem Batteriespeicher der PV-Anlage über die Nacht zu kommen, weil die Wärmepumpe (der Kompressor) damit fast die ganze Nacht ausbleiben kann und sich erst wieder um 04:00 Uhr morgens einschaltet, wenn die Vorlauftemperatur aus dem Pufferspeicher nicht mehr hoch genug ist. 

Wie @TheSmartGerman, plane ich mich damit intensiver zu beschäftigen, sobald die aktuelle Heizperiode um ist, weil ich dann den CanBus-Adapter vom Linux (auf welchen bei mir aktuell Open3E läuft) abkoppeln und direkt an die Maschine auf welcher bei mir HASS läuft anstecken kann. Dann kann ich damit rumspielen und versuchen Open3E als HACS-Integration umzusetzen.

fabian727, 2024-05-06:
@Wolfgang-03 eine HACs Integration ist natürlich ebenfalls interessant. Leider jedoch nur für die Leute derren HASS im gleichen Zimmer, wie die Heizung steht.
Ich bin einer dieser Glücklichen.
Meine Schwester leider nicht.

Sollte es eine HACS Integration werden, können andere SmartHomes-Applikationen entweder HomeAssistant integrieren oder tuhen sich mit dem Extrahieren schwer (openhab, iobroker ... ).
Deswegen bin ich ein FAN von MQTT als AbstraktionsLayer. HomeAssistant bleibt optional, bzw. es können sogar andere Platformen zusätzlich unterstützt werden.

Die 30sekündigen polls sehe ich als ein separates Thema. Mir hats die Ansicht der Topics im MQTT Explorer dadurch zerschossen. Es sah so aus als ob Tasmota-Themen zerissen wurden.
Ich hab auf 2 Minuten-Takte reduziert.
Hier bin ich im allgemeinen ein FAN von CommunicateOnChange. Der letzte Wert ist sowieso auf MQTT bekannt. Wozu reposten.
In das CAN hab ich nicht tiefer geschaut, vermute aber, dass Viessmann einige Werte sowieso regelmäßig schickt/ andere OnChange. Das könnte man ausnutzen.

MyHomeMyData, 2024-05-06:
@fabian727 CommunicateOnChange ist genau das, was [E3onCAN](https://github.com/MyHomeMyData/E3onCAN) macht. Das funktioniert allerdings nur, wenn Vitocal als Master agiert, also z.B. mit einer Vitocharge über CAN vernetzt ist. Für die Daten der Vitocal funktioniert es auch auf dem internen CAN-Bus im Standalone-Betrieb.

CommunicateOnChange wäre auch eine interessante Erweiterung für open3e. Werde mal drüber nachdenken.

Eine weitere Möglichkeit wäre, die E3-Geräte über den ioBroker-Adapter [e3oncan](https://github.com/MyHomeMyData/ioBroker.e3oncan) ins System zu integrieren. Da kann CommunicateOnChange und Polling kombiniert werden. Über die Konfiguration der Datenpunkte (= Objekte in ioBroker) kann problemlos ein Versand per MQTT konfiguriert werden.

fabian727, 2024-05-06:
fände es schöne wenn es eine standalone "App" bleibt.
Ob man es dann per ioBroker einstellt oder HomeAssistant sollte dann eigtl. egal sein.

MyHomeMyData, 2024-05-06:
Zustimmung. Wollte nichts anderes vorschlagen. Auch E3onCAN ist eine standalone App.

Bei der von mir erwähnten Vernetzung geht es nur um die Viessmann-Geräte untereinander. Auf dem externen Bus werden sehr viele Daten ausgetauscht, wenn die Vitocal als Master konfiguriert ist. Ist sie das nicht, passiert da nicht viel. Man muss dann die Datenpunkte aktiv abfragen, so wie es open3e macht. E3onCAN lauscht nur und versendet per MQTT was die E3-Geräte von sich aus ausplaudern.

fabian727, 2024-05-06:
Evtl. wären hier noch ein paar einleitende Worte cool. 

Target of open3e is to broadcast all communicated datapoints of any viessmann device to mqtt.
It also shall act as a master to interact with devices and poll their data.
It shall be capable or replacing a WAGO Gateway of Viessmann.
Its precessor is [e3oncan](https://github.com/MyHomeMyData/ioBroker.e3oncan) which only is capable of listening and polling. 

Ich hoffe ich hab es halbwegs erwischt?

Wolfgang-03, 2024-05-06:
CommunicateOnChange wäre natürlich am besten. Da sollte ich mir vielleicht mal E3onCAN anschauen, damit habe ich mich bisher noch nicht beschäftigt. Da stellen sich mir aber gleich mehrere andere Fragen:

  - Wie sicher kann ich dann sein, dass ich alle Änderungen mitbekomme?
  - Was passiert mit den Datenpunkten die sich ständig oder in sehr kurzer Folge ändern? Liefern diese dann unter Umständen mehrmals pro Sekunde neue Daten?
  - Kann ich Daten ausblenden die mich nicht interessieren?
  - Kann ich Datenpunkte trotzdem proaktiv auslesen (Zum Beispiel bei einem Neustart von HA, damit die Entitäten initialisiert werden)?

Bislang habe ich bei meiner Automatisierung das Problem, wenn ich am Code rumfummle und dann einen Quick-Reload der Configuration oder einen Restart von HA mache, dass dann alle Entitäten kurzzeitig auf "unbekannt" wechseln. Da ich in meiner Automatisierung Abfragen auf Entitäten habe, wie "Wenn Entität X länger als Y Minuten > Wert Z, dann mache ...", läuft die Zeit ab dem Reload wieder von Null. Ich habe schon am Quellcode von Open3E rumgefummelt und bei MQTT.publish retain auf True gestellt aber wirklich geholfen hat das auch nicht...

MyHomeMyData, 2024-05-06:
@fabian727 Ich fürchte, das trifft es nicht wirklich. Ich versuche eine Sortierung:
* open3e fragt aktiv Datenpunkte bei den Viessmann-Geräten ab, die in einer Liste spezifiziert sind. Einmal oder regelmäßig oder per MQTT-Anfrage ("Listener-Mode"). Die Inhalte werden per MQTT versendet und/oder auf der Kommandozeile ausgeben. Auch das Ändern von Datenpunkten ist möglich. Daten, die von den Viessmann-Geräten untereinander ausgetauscht werden, werden ignoriert. Python.
* [E3onCAN](https://github.com/MyHomeMyData/E3onCAN) hört auf dem CAN-Bus zu und gibt alle oder per Liste spezifizierte Datenpunkte auf der Kommandozeile aus und/oder versendet sie per MQTT. Das Konzept erfasst nur Daten, die von den Viessmann-Geräten von sich aus auf den Bus gesendet werden. Auch die Viessmann Energy Meter E380 und E3100CB werden unterstützt. Eine Begrenzung der Übertragungshäufigkeit ist möglich. Es erfolgt keinerlei aktive Abfrage von Datenpunkten. Auch ein Verändern von Datenpunkten ist nicht möglich. Vorteil ist, man bekommt Änderungen von Datenpunkten sofort mit. Nachteil: Datenpunkte, die sich selten ändern, werden auch nur selten gesendet. Python.
* open3e und E3onCAN sind unabhängig voneinander und parallel nutzbar.
* Der ioBroker-Adapter [e3oncan](https://github.com/MyHomeMyData/ioBroker.e3oncan) **vereint die Funktionalitäten von open3e und E3onCAN** und führt die Datenpunkte in einer gemeinsamen Objektstruktur zusammen. Das Versenden per MQTT ist natürlich mit ioBroker möglich, aber nicht direkt Teil des Adapters. Ein Parallelbetrieb mit open3e und E3onCAN ist möglich, aber nicht sinnvoll. Javascript.

MyHomeMyData, 2024-05-06:
@Wolfgang-03 Zu Deinen Fragen zu [E3onCAN](https://github.com/MyHomeMyData/E3onCAN):

> Wie sicher kann ich dann sein, dass ich alle Änderungen mitbekomme?

Es werden allen Daten erfasst, die auf dem CAN-Bus erscheinen. Ob die Viessmann-Geräte jede (relevante) Änderung auch per CAN versenden, kann ich nicht sagen. Meine Erfahrung ist aber, dass es mindestens nahe bei 100% liegt.

> Was passiert mit den Datenpunkten die sich ständig oder in sehr kurzer Folge ändern? Liefern diese dann unter Umständen mehrmals pro Sekunde neue Daten?

Das kann limitiert werden. Die Energy Meter liefern z.B. manche Werte zweimal pro Sekunde. Es ist eine Mindest-Zeitdifferenz zwischen zwei MQTT-Messages einstellbar.

> Kann ich Daten ausblenden die mich nicht interessieren?

Du kannst die gewünschten spezifizieren (wie bei open3e). Du kannst auch mehrere E3onCAN-Instanzen parallel betreiben, z.B. bei mehreren E3-Geräte für jedes Gerät eine Instanz.

> Kann ich Datenpunkte trotzdem proaktiv auslesen (Zum Beispiel bei einem Neustart von HA, damit die Entitäten initialisiert werden)?

Nein. E3onCAN hört generell nur zu. Aber Du kannst dafür open3e parallel nutzen.

Wolfgang-03, 2024-05-06:
@MyHomeMyData: Verstehe, also wäre das Ziel es so zu machen wie der ioBroker-Adapter, lauschen mit E3onCAN was sich gerade ändert und wenn ich spezifische Daten sofort benötige, dann rufe ich die wie gehabt über bzw. wie Open3E ab. Im HA dann natürlich komplett als Python und nicht als Javascript. Richtig? Ich muss mir dringend mal E3onCAN anschauen.

fabian727, 2024-05-06:
> @fabian727 Ich fürchte, das trifft es nicht wirklich. Ich versuche eine Sortierung:
>
>     * open3e fragt aktiv Datenpunkte bei den Viessmann-Geräten ab, die in einer Liste spezifiziert sind. Einmal [...]

Kannst du das in die readme eintragen? Für Neu-Ankömmlinge wie mich? Dann weiß man direkt zu 10% mehr bescheid ;)

MyHomeMyData, 2024-05-06:
@Wolfgang-03 Ja, für Python gibt es eine [Bibliothek](https://github.com/pylessard/python-udsoncan), die [UDSonCAN](https://www.csselectronics.com/pages/uds-protocol-tutorial-unified-diagnostic-services) beherrscht. open3e nutzt sie für die CAN-Kommunikation. Leider weicht das Protokoll, das die Viessmann-Geräte untereinander verwenden, davon ein wenig ab. Deshalb ist das in E3onCAN direkt implementiert.

pr0vieh, 2024-05-07:
@Wolfgang-03  
> Aktuell habe ich meine Vitocal 250-A, zu 90% im Griff, damit sie macht, was ich möchte. Zwar noch mit ein paar Einschränkungen, die mir nicht gefallen, aber daran kann man noch arbeiten. Zum [...]

Kann es sein das deine Pumpe kein PWM kann oder es nicht eingestellt ist
Ich habe eine Solar Pumpe wo die Solar Typen meinten das Steuergerät (ist 10 Jahre alt) kann nur An/Aus und haben die Pumpe im Normal modi da hingehangen und waren fertig... 
Die folge ständiges Toggeln grausig (da war das Wasser wenn es ankam schon wieder kalt)
ich habe die Pumpe dann in den PWM modus versetzt und im Steuergeräte auch auf PWM umgestellt und siehe da das ding kann langsam und schnell ;) und dann liefs

Wolfgang-03, 2024-05-11:
@pr0vieh 

PWM = Pulsweitenmodulation?
Ich muss zugeben, ich bin kein Heizungsbauer, aber Elektriker. Ich habe heute mal wieder die VC250 komplett auseinander genommen. So wie ich das sehe, ist die bei mir verbaute Pumpe nicht für eine externe Steuerung vorgesehen. 

[image]

Laut Anleitung: SC = Self Control

Die Pumpe hat, soweit ich das sehen konnte, keine weiteren Eingänge für eine Steuerleitung. Das einzige Kabel das zur Pumpe geht, ist in der VC250 auf P1 angeschlossen und das ist, so wie ich das sehe, nur eine 240 V Stromversorgung. 

[image]

Aber auch in der Montageanleitung habe ich nichts gefunden, wie die ansonsten angeschlossen werden sollte, damit die Drehzahl von der VC250 gesteuert werden kann. Aber irgendwie sollte das eigentlich möglich sein, sonst würde ja der DID 1102 keinen Sinn ergeben, oder?

Philip-Wiege, 2024-05-11:
Gibt es von der Vitocal 250-A so unterschiedliche Revisionen? Meine ist im Januar 2024 geliefert worden und hat eine Grundfos UPM4 PWM Pumpe in der Inneneinheit eingebaut. Es ist die Version ohne integrierten Wasserspeicher.

Aber auch dort erfolgt keine wirkliche Regelung. Viessmann gibt dort auch nur eine feste Leistung = Differenzdruck vor und die Pumpe dreht dann so hoch, wie sie muss, um den Differenzdruck zu erreichen.

Daher haben Anlagen mit wenig Druckverlust auch zwischen Soll und Istwert für die Pumpe kaum Abweichung, weil die Pumpe hoch drehen muss für den Differenzdruck. Bei Anlagen mit hohen Druckverlusten erreicht die Pumpe schon bei viel weniger Drehzahl den Differenzdruck. Daher sieht man dort sowas wie Sollwert 80% und Istwert 50%.

Wolfgang-03, 2024-05-12:
> Gibt es von der Vitocal 250-A so unterschiedliche Revisionen?

Offensichtlich, meine wurde in Oktober 2023 geliefert. Die Zirkulationspumpe(n) für den Heizkreis sind bei mir nicht in der Inneneinheit, vermutlich weil ich 2 Heizkreise habe. Hierzu ist neben der Inneneinheit eine Divicon mit 2 Heizkreisen und damit auch 2 Pumpen (siehe oben) installiert. Wobei der Heizkreis 1 (Heizkörper) kein Mischventil hat. Für Heizkreis 2 (Fußbodenheizung) gibt es zusätzlich ein Mischventil, damit dieser Heizkreis mit einer niedrigeren Vorlauftemperatur gefahren werden kann. HK2 ist allerdings noch nicht in Betrieb. Die Fußbodenheizung im Wohnzimmer muss erst noch gebaut werden...

TheSmartGerman, 2024-05-12:
Also meine 250 hat 2 Heizkreise, es gibt ja die Inneneinheit mit 1 oder 2 Heizkreisen (glaube 2C) in der Bezeichnung. Ich habe die Viessmann Vitocal 250-A AWO-E-AC 251.A13 2C Luft/Wasser-Wärmepumpe 2 Heiz-/Kühlkreise.

Bei mir ist ein Heizbalken mit 2 davivon installiert, einmal für Heizkörper und einmal für Fußbodenheizung. 
Als Pumpen habe ich die Wilo (sind ohne PWM).

Installiert 07/23.

Wenn's was zum testen/prüfen gibt, kann ich Nachschauen.

fabian727, 2024-05-13:
BackToTopic:
Die Integration von Hass

Ich war mal so frei und habe mir von @Wolfgang-03 die vitocal-250.yaml "geliehen" und in meinen Fork integriert. Ich hoffe dies war ok.
[fork](https://github.com/fabian727/open3e/tree/vitocalOnHomeAssistant)

Ruft man open3e mit dem zusätzlichen Parameter "--homeassistant" auf, wird die yaml als mqtt gepublished und in HomeAssistant integriert. Es ist somit kein manuelles kopieren notwendig.

Es werden aktuell die unique_id's und object_id's der entities mit einem "auto_" am Anfang ergänzt.
Die serielle Number wird mit "unknown_serial" ergänzt.

Im nächsten Schritt will ich alle dids die gefunden werden als "blinde" nummern in hass reinbringen.
Im Schritt danach wird verglichen welche dids in der yaml existieren und nur diese werden an hass weitergegeben.

Evtl. kann man die yaml in die json's integrieren. Ich wollte sie jedoch erstmal 1:1 erhalten, nachdem es möglich war, um alle Optionen offen zu halten.

Es müsste jetzt auch möglich sein das Skript mit 1 Aufruf dauer laufen zu lassen. Zuerst scanall, danach listen (ungetestet)