open3e/open3e #97: Dekodierung Status Legionellenprogramm

Philip-Wiege, 2024-05-02:
Hallo zusammen,

meiner Meinung nach wäre es eine Verbesserung den DID 873 : RawCodec(2,"LegionellaProtectionActivation") bereits in open3e zu dekodieren.

Ich weiß leider nicht, in welcher Reihenfolge die Bytes in open3e ausgegeben werden. Im MQTT explorer habe ich folgende Übersetzung rausgeforscht:

0000 = Legionellenschutz deaktiviert / nicht aktiv
0100 = Legionellenschutz aktiviert / nicht aktiv
0101 = Legionellenschutz aktiviert / aktiv

Erstes Byte also quasi Mode und zweites Byte State.

Wie ist eure Meinung dazu und wie könnte das umgesetzt werden?

Grüße
Philip

AnHardt, 2024-05-02:
```
        873 : O3EComplexType(2, "LegionellaProtectionActivation",[O3EByteVal(1,"Mode"),O3EByteVal(1,"State")]),    
```
Wäre eine Möglichkeit.
Ginge natürlich, wenn bestimmt nur '0' und '1', auch als bools. (Das macht die (binary-)Sensoren oder Switches für HA aber auch nicht einfacher.)

Philip-Wiege, 2024-05-02:
Diese Modifikation würde ich bei mir lokal mal ausprobieren. In Homeassistant müsste man das 0 und 1 dann relativ leicht in andere Typen casten können. Ich werde mir einen binären Sensor anlegen für "Legionellenprogramm freigegeben" und wenn das Programm aktiv läuft, springt meine Zustandsvariable dür die Vitocal auf "DomesticHotWaterHygiene".

[image]

philippoo66, 2024-05-03:
ausprobieren klingt gut! und wenn es gut aussieht nicht lang fackeln und rein in den develop branch. dann probieren es mehr und entweder profitieren mehr oder wir können es anhand Rückmeldungen verbessern.

Philip-Wiege, 2024-11-05:
Ich habe den Commit https://github.com/open3e/open3e/commit/624c7f9a955cf98921961927a17e97bcefb40d99 in Develop eingereicht. Funktioniert auf meiner open3e Instanz. Das Thema kann geschlossen werden. Danke an alle!

closed 2024-11-05 by Philip-Wiege