open3e/open3e #98: Virtual Environment

fabian727, 2024-05-05:
when I tried to install the requirements. Python reminded me to use a virtual environment.
It's pretty easy.

go to the desired directory /my/home/open3e
creation
´python -m venv´
starting
´source bin/activate´
afterwards install the requirement and start the scripts

with ´deactivate´ you can leave it again.

the systemd service must not call the global python, but the one within the virtual environment.
so change 'python /my/home/open3e/*.py´ to ´/my/home/open3e/venv/python /my/home/open3e/*.py´
that's it

philippoo66, 2024-05-25:
@fabian727 danke für deinen Beitrag! ich habe ihn auch noch im Wiki [verlinkt](https://github.com/open3e/open3e/wiki/030-Installation-und-Inbetriebnahme-von-open3E#systemd-und-virtual-environment). eventuell wäre er da sogar besser aufgehoben...

Nelson666, 2024-05-28:
Ich habe die ganze Installation mit Python 3.12 gemacht, da schaut Start Befehl etwas anders aus. 
`ExecStart=/bin/bash -c 'cd /home/pi/open3e/ && /home/pi/open3e/bin/open3e @myargs'`

Den Pfad, den Ordner und den Names des Parameterfiles müsst ihr natürlich anpassen.
Das müsste auch funktionieren.
`ExecStart=/bin/bash -c 'cd /home/pi/open3e/ && ./bin/open3e @myargs'`

closed 2024-08-26 by MyHomeMyData