open3e/open3e #100: Vitodens 343F supported?

wunderbaum, 2024-05-13:
Congratulations to this nice and versatile project!

I am an owner of a Vitodens 343F gas and solarthermie-system since a few years.
Is my system having canbus or not?

What I found so far: Its sending via MQTT encrypted data to the viessmann cloud where I can get my own data back with an API-key. But this is not _the yellow from the egg_.

philippoo66, 2024-05-13:
good morning @wunderbaum !

pls have a look if there is an Optolink interface on your device

[image]

if it is, then you require [someting else](https://github.com/philippoo66/optolink-splitter). If there is none, pls post the first 7 digits of the manufacturer number / SN (to be found on the label on top of the device) so that we can have a look.

greetings!
Phil

wunderbaum, 2024-05-13:
Good evening!
I do not have such a V-connector on my system. Or can this be inside the housing?
I think I have kind of this: https://www.manualslib.de/manual/456913/Viessmann-Vitodens-Series.html?page=27#manual
Vitodens 343-F, Typ B3UA

philippoo66, 2024-05-13:
are you sure that you don't have the V? slide away the left cover...

[image]

wunderbaum, 2024-05-14:
Thanks for your patience!
My system do not have a left/right cover it's full frame glass front. But I managed to take the display off and behind it, I found two plugs (I made a picture).
One is  the connection for the display (orange). The other one is unused (green).
[image]

Is it worth it do dig deeper and remove more from the housing? Seems to be very strong closed.

philippoo66, 2024-05-14:
this is strange. please post the first 7 digits of Herstellernummer so that we can take a deeper look. pleas also post a foto of the display unit. thx

wunderbaum, 2024-05-14:
Here we go:
[image]

Number is (on the right side): 7722304

philippoo66, 2024-05-14:
you have a CAN Bus and can use open3e :-)

[image]

find your service manual [here](https://static.viessmann.com/resources/product_media/5855198VSA00010_1.PDF)

philippoo66, 2024-05-14:
[image]

wunderbaum, 2024-05-14:
What great news! This makes me happy. Many thanks for your absolute-beginner support!
I think I need to take off the housing to find this connector inside.

Now it's worth for me to read all open3e manuals and get into the system.

philippoo66, 2024-05-14:
have fun! we are always willing to support, there are no stupid questions... ;-)

best to start proceeding like described in the [README](https://github.com/open3e/open3e/blob/master/README.md). Also usefull the ['absolute beginner storry'](https://github.com/open3e/open3e/discussions/5). Regarding hardware see [here](https://github.com/open3e/open3e/wiki/020-Inbetriebnahme-CAN-Adapter-am-Raspberry).