open3e/open3e #124: How to set parameter 504.4?

m0baer, 2024-09-10:
Hi,

thanks to this great project I could access the values of my heatpump as I expect that as an IT guy.
There seems to be a difference or a missing part for a vcal 252 like mine. I can only get Metadata/binarydata for the parameter 504. I would like to get/set 504.4 as that is the max value for hot water in SG Ready mode 4 (force).
My guess is, that is set to around 60 degrees at the moment. The value I get from 504 is 
`vcal 504 DomesticHotWaterSetpointMetaData "64006400f4015802b60300002602"` at the moment.
So my guess is that 504.4 is the `5802` (the fourth packet of four digits) which is dec 600 (little/big endian)-> 60.0. 
Am I right with this thought?
I did not look deep enough into the sources as I wonder if it breaks something for other vcals if I try to build the 504.4 parameter into open3e? Has `DomesticHotWaterSetpointMetaData` another meaning on other vcals or is it just an unknown parameter at the moment?

Thanks,

Jochen

MyHomeMyData, 2024-09-10:
Yes, when we assume 504.1 ... 504.3 all are 16-bit values, this is correct. Where does the information about 504.4 come from?

According to [Service-Anleitung](https://github.com/open3e/open3e/wiki/050-Viessmann-Dokumente) we have:
504.1: MinimumSetpoint
504.3: MaximumSetpoint

I could not find other information about 504 there.

Thanks for supporting our project!
Jürgen

bu-na, 2024-09-10:
Hi all,

here is my guess:

 504 : O3EComplexType(14, "DomesticHotWaterSetpointMetaData", [O3EInt16(2,"Unknown504.0"), O3EInt16(2, "MinimumBufferTemperatureSetpoint", signed=True), O3EInt16(2, "Unknown504.2"),O3EInt16(2, "MaximumBufferTemperatureSetpoint", signed=True),O3EInt16(2, "Unknown504.4"), O3EInt16(2, "EfficiencyLowerLimit", signed=True), O3EInt16(2, "EfficiencyUpperLimit",signed=True)]), #250xx Unit °C

Some additional Information from the Service Guide:       
[image]

I'm not sure what's behind 504.0, 504.2 and 504.4...
My values for these parameters are 10, 50, 95 °C.

Regards Bu-Na

m0baer, 2024-09-11:
Thanks Jürgen and Bu-Na,

the v-community has been my source, too. I didn't know if it's okay to link directly to that thread.
The values from the first post were my actual values for 504: `64006400f4015802b60300002602`

So there is as decimals:
| Parameter | Value | Manual | 252-A guess |
|--------|--------|--------|--------|
| 504.1 | 100 (?) | minimum | no clue |
| 504.2 | 100 (?) | ? | no clue |
| 504.3 | 500 | maximum | Seems to be the Setpoint (50°C)|
| 504.4 | 600 | see forum | SG Ready mode 4 maximum (60°C) |
| 504.5 | 950 | lower efficiency | does not fit | 
| 504.6 | 0 | higher efficiency | does not fit |
| 504.7 | 550 | ? | seems to be the maximum (55°C) | 

Or does the index of a parameter start at 0? My guess was it starts at 1 so the 60 degrees of 504.4 match to my observations in SG Ready mode 4. The column 252-A guess has my guesses that fit to my settings at the moment.
I'm still waiting for a first firmware update, the inner part of the heatpump was delivered december 22 and never had an update. Did some of the parameters change during updates?

Thank you,

Jochen

bu-na, 2024-09-11:
The index of a parameter start at 0
  
[image]

m0baer, 2024-09-11:
Thanks for the input, my HMI does not have the option to see or set the parameters in that way the only thing I can set is the setpoint and the mod (eco/comfort).
That looks like the info from Flo is wrong and 504.3 is the parameter I need.

| Parameter | Value | Manual | 252-A guess |
|--------|--------|--------|--------|
| 504.0 | 100 | ? | no clue |
| 504.1 | 100 | minimum | ... |
| 504.2 | 500 | ? | Not the Setpoint, does not change when changing the setpoint on the device |
| 504.3 | 600 | max | SG Ready mode 4 maximum (60°C) |
| 504.4 | 950 | ? | does not fit | 
| 504.5 | 0 | efficiency low | ... |
| 504.6 | 550 | efficiency high | max in the default mode |

MyHomeMyData, 2024-09-11:
From my point of view proposal of @bu-na presently is best guess. I updated data point 504 in branch develop accordingly, but used `RawCodec` for the `Unkown` parts.

```
504 : O3EComplexType(14, "DomesticHotWaterSetpointMetaData", [RawCodec(2,"Unknown504.0"), O3EInt16(2, "MinimumBufferTemperatureSetpoint", signed=True), RawCodec(2, "Unknown504.2"),O3EInt16(2, "MaximumBufferTemperatureSetpoint", signed=True),O3EInt16(2, "MaximumDomesticHotWaterTemperature", signed=True), O3EInt16(2, "EfficiencyLowerLimit", signed=True), O3EInt16(2, "EfficiencyUpperLimit",signed=True)]), #250xx Unit °C
```

Please check. Feedback is welcome.

m0baer, 2024-09-12:
Thank you, Jürgen,

I will move my build pipeline to the development branch this evening to test the settings.

Meanwhile I have another guess for 504.2. Maybe this is the point where the SG Ready mode re-activates the heatpump.

Example: I had SG mode 4 (forced) active. The heatpump stops when 504.3 is reached. Once the hot water temperature got below 504.2 the heatpump starts again.

It's just a guess that fits to my observations, maybe somebody can verify?

bu-na, 2024-09-12:
Hi all,
according to the Service Manual  (just found in the net, see my edited comment on top):
Parameter 504.0 , 504.2 and 504.4 describes the **Setting Range** of 504.1. and 504.3.

All Values are Temperatures.

Regards Bu-Na

(It will probably work the same Way for other Parameters)

m0baer, 2024-09-13:
That's great!

switched to development branch yesterday and readings look good (on my Vitocal 252-A), thank you.

As my HMI has no option (yet, still waiting for a newer firmware) to set these parameters I have to use open3e to adjust 504.3.
Do I have to set the whole 504 parameters as one entry and adjust the 4th 16 bit value to my needs while leaving the other bits as they are or is there a way to address 504.3 directly for writing?

bu-na, 2024-09-13:
@MyHomeMyData 
My updated naming suggestion for 504.(0-4):

504.0 --> LowerBufferLimitTemperature
504.1 --> MinimumBufferTemperature
504.2 --> DefaultBufferTemperature
504.3 --> MaximumBufferTemperature
504.4 --> UpperBufferLimitTemperature

MyHomeMyData, 2024-09-13:
Find ich klasse. Eben in develop eingebaut:
```
504 : O3EComplexType(14, "DomesticHotWaterSetpointMetaData", [O3EInt16(2,"LowerBufferLimitTemperature", signed=True), O3EInt16(2, "MinimumBufferTemperature", signed=True), O3EInt16(2, "DefaultBufferTemperature", signed=True), O3EInt16(2, "MaximumBufferTemperature", signed=True), O3EInt16(2, "UpperBufferLimitTemperature", signed=True), O3EInt16(2, "EfficiencyLowerLimit", signed=True), O3EInt16(2, "EfficiencyUpperLimit",signed=True)]), #250xx Unit °C
```

Aus meiner Sicht kann der Issue damit geschlossen werden.

m0baer, 2024-09-14:
> Aus meiner Sicht kann der Issue damit geschlossen werden.

Hmmm wäre schade. Die Frage war ja nach set/write und wurde mir bisher noch nicht beantwortet.
Kann ich jetzt einen beschreibbaren Teilparameter wie 504.3 mit Open3e direkt beschreiben, oder muss ich mir den kompletten 504 Parameter aus einzelnen 16 Bit little endians manuell zusammensetzen und dann in einem Rutsch schreiben?

SanMiggel, 2024-09-15:
Complex dids können auch als json geschrieben werden. Allerdings müssen im alle Felder definiert werden. Also z.b.

 open3e -c can0 -dev vcal -v -w '424={"Comfort": 21.0, "Standard": 20.0, "Reduced": 20.0, "Unknown2": "0000"}'

Wenn du nur einen Wert ändern willst musst du lesen, anpassen, schreiben. Das geht auch über das mqtt Protokoll.

m0baer, 2024-09-16:
> Complex dids können auch als json geschrieben werden. Allerdings müssen im alle Felder definiert werden. Also z.b. open3e -c can0 -dev vcal -v -w '424={"Comfort": 21.0, "Standard": 20.0, "Reduced": [...]

Danke für die Rückmeldung, ich habe es befürchtet. Ich bekomme immer den Fehler 0x22 bei einem Schreibversuch:
`# open3e -c can0 -dev vcal -v -w '504={"LowerBufferLimitTemperature": 10.0, "MinimumBufferTemperature": 10.0, "DefaultBufferTemperature": 50.0, "MaximumBufferTemperature": 56.0, "UpperBufferLimitTemperature": 95.0, "EfficiencyLowerLimit": 0.0, "EfficiencyUpperLimit": 55.0}'
write: 1664.504 = {'LowerBufferLimitTemperature': 10.0, 'MinimumBufferTemperature': 10.0, 'DefaultBufferTemperature': 50.0, 'MaximumBufferTemperature': 56.0, 'UpperBufferLimitTemperature': 95.0, 'EfficiencyLowerLimit': 0.0, 'EfficiencyUpperLimit': 55.0}
Traceback (most recent call last):
  File "/usr/local/bin/open3e", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.12/site-packages/open3e/Open3Eclient.py", line 394, in main
    succ,code = dicEcus[ecu].writeByDid(didkey, didVal, raw=False)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/open3e/Open3Eclass.py", line 162, in writeByDid
    response = self.uds_client.write_data_by_identifier(did, val)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/udsoncan/client.py", line 167, in decorated
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/udsoncan/client.py", line 521, in write_data_by_identifier
    response = self.send_request(req)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/udsoncan/client.py", line 2258, in send_request
    raise NegativeResponseException(response)
udsoncan.exceptions.NegativeResponseException: WriteDataByIdentifier service execution returned a negative response ConditionsNotCorrect (0x22)`

Das selbe auch, wenn ich per RAW versuche den neuen Wert zu setzen oder auch wenn ich einfach die ausgelesenen Werte nochmal schreiben will:

`/open3e # open3e -c can0 --raw -r 0x680.504
64006400f4015802b60300002602
/open3e # open3e -c can0 --raw -w 0x680.504=64006400f4015802b60300002602
write raw: 1664.504 = 64006400f4015802b60300002602
Traceback (most recent call last):
  File "/usr/local/bin/open3e", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.12/site-packages/open3e/Open3Eclient.py", line 386, in main
    succ,code = dicEcus[ecu].writeByDid(didkey, didVal, raw=True)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/open3e/Open3Eclass.py", line 162, in writeByDid
    response = self.uds_client.write_data_by_identifier(did, val)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/udsoncan/client.py", line 167, in decorated
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/udsoncan/client.py", line 521, in write_data_by_identifier
    response = self.send_request(req)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/udsoncan/client.py", line 2258, in send_request
    raise NegativeResponseException(response)
udsoncan.exceptions.NegativeResponseException: WriteDataByIdentifier service execution returned a negative response ConditionsNotCorrect (0x22)`

Kann mich da jemand auf den richtigen Pfad stupsen, ich finde bei udsoncan den Fehler 0x22 nur in Verbindung mit read, aber nicht mit write.

SanMiggel, 2024-09-18:
Hmm, soweit ich weiß heißt dieser Fehler, dass die Firmware nicht mag dass der Parameter geschrieben wird. Aus welchen Gründen auch immer. Da gab es hier schon lange Diskussionen. 
@philippoo66 oder übersehe ich etwas?

MyHomeMyData, 2024-09-18:
Ja, so interpretiere ich das auch. Prinzipiell kann der Datenpunkt geschrieben werden, 504 steht in der [Liste](https://github.com/open3e/open3e/blob/master/src/open3e/Open3Edatapoints_writables.json) der schreibbaren Datenpunkte. NR mit 0x22 besagt aber, dass nicht alle notwendigen Randbedingungen geben sind. Vermutlich sind dazu höhere Rechte notwendig und das Verfahren ist per Seek&Key geschützt. Zumindest bisher mit open3e nicht machbar.

m0baer, 2024-09-18:
Dann wird das wahrscheinlich auch an meiner alten Firmware liegen. Ich weiß nichtmal ob meine FW 2244 überhaupt ein OTA Update kann. Ich kann die Parameter ja auch nicht am HMI einstellen, bei neuerer Firmwareversion scheint das ja zu gehen.
Mal sehen wie ich da ein Update anleiern kann.
Danke für die Rückmeldungen, ich schließe den Vorgang dann mal, das scheint ja nur mich (bzw. alte Firmwarestände) zu betreffen.

MyHomeMyData, 2024-09-18:
Meine Vitocal ist auf FW 2323 und da geht es auch nicht. Es kommt auch NR 0x22. Ich würde also eher auf geschützten Parameter tippen.

m0baer, 2024-09-19:
Aber kann man da in FW 2323 wenigstens im HMI Änderungen einstellen bei dir @MyHomeMyData ? Ich habe als Endkunde hier im Augenblick gar keine Möglichkeit was zu ändern, auch nicht über viguide Web.

MyHomeMyData, 2024-09-19:
@m0baer Ja, im HMI kann ich 504.x Datenpunkte ändern und sehe die Änderungen dann auch über open3e. Werde mir mal anschauen, ob der Änderungsvorgang auf dem CAN-Bus sichtbar wird.

MyHomeMyData, 2024-10-14:
@m0baer Nun kann der Issue aber geschlossen werden, oder?

m0baer, 2024-10-14:
Besten Dank @MyHomeMyData ,

ich konnte es noch nicht austesten, weil ich noch am "externen" Bus hänge, werde das aber am kommenden Wochenende nachholen.

Ja, ich schließe das Issue mit dem Post, Danke!

closed 2024-10-14 by m0baer