open3e/open3e #132: DID 2529: Name Energiewert für Woche

Philip-Wiege, 2024-10-09:
Hallo zusammen,

kann es sein, dass im DID 2529 GeneratedCoolingOutput der Wert für die letzten 7 Tage noch "Week" heißt und nicht "Past7Days" wie bei den anderen zwei DIDs für Warmwasser und Kühlung?

Ist zwar nur ne kleine Sache, aber vielleicht sollten wir das gleichziehen?

1391 GeneratedDomesticHotWaterOutput
1211 GeneratedCentralHeatingOutput

Grüße
Philip

MyHomeMyData, 2024-10-09:
Stimmt. Und bei 2517 und 2526 genauso. Ich werde das anpassen.

Philip-Wiege, 2024-10-09:
Super, danke für deinen Einsatz! Ich komm aktuell einfach nicht dazu mal ruhig am PC zu sitzen und das selbst zu machen.

MyHomeMyData, 2024-10-09:
Ist in develop drin. Ich werde demnächst wieder einen Merge develop nach master machen.

flecke-m, 2024-10-09:
@MyHomeMyData 
kannst du dann bitte auch die Version wieder updaten? -> pyproject.toml | Updated version to 0.2.0

Ich habe letztens das Docker image mit dem aktuellsten main-branch erstellt und die Version 0.2.0 gemacht, damit es synchron wird.

Es wäre gut wenn die Versionierung dementsprechend weiter geht, so dass ich das Docker image entsprechend benennen kann.

Danke

MyHomeMyData, 2024-10-09:
@flecke-m Danke für den Hinweis. Habe es auf 0.2.1 gesetzt.

closed 2024-10-11 by MyHomeMyData