open3e/open3e #141: write to did throws NegativeResponseException

bomilkar, 2024-11-03:
I'm new to this and got it to work, almost.
I have the did list and can read values from my Vitocal-250A:
```
open3e -c can0 -dev vcal -r 1643 -v
vcal 1643 MixerOneCircuitCurrentTemperatureSetpoint 21.0
closing 0x680 - bye!
```
But when I try to write, it throws the exception:
```
open3e -c can0 -dev vcal -raw -w 1644=d200
write raw: 1664.1644 = d200
Traceback (most recent call last):
  File "/home/pi/.local/bin/open3e", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 386, in main
    succ,code = dicEcus[ecu].writeByDid(didkey, didVal, raw=True)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/open3e/Open3Eclass.py", line 162, in writeByDid
    response = self.uds_client.write_data_by_identifier(did, val)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/udsoncan/client.py", line 167, in decorated
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/udsoncan/client.py", line 521, in write_data_by_identifier
    response = self.send_request(req)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/udsoncan/client.py", line 2258, in send_request
    raise NegativeResponseException(response)
udsoncan.exceptions.NegativeResponseException: WriteDataByIdentifier service execution returned a negative response ConditionsNotCorrect (0x22)
```

Can anyone help me?

Not sure why it echos this:
`write raw: 1664.1644 = d200`
Why toes it clobber the did ?

MyHomeMyData, 2024-11-03:
The meaning of "negative response ConditionsNotCorrect (0x22)" is: Device (Vitocal) rejected the write access. There are many data points we cannot change. Typically we don't know the reason. But this is not a malfunction of open3e.

In some cases an extended write service "sid 77" can do the trick. But not in this case - I just tried it on my vitocal.

Philip-Wiege, 2024-11-03:
It seems as if you wanted to change the setpoint for central heating. The DID you are trying to write gives the current setpoint which is read only.

In my understanding the setpoints must be controlled via the setpoints for reduced, normal and comfort mode. These setpoints in DID 424 should by writeable.

bomilkar, 2024-11-04:
Thanks a lot for the quick help! So that gets me one step further. But I'm not quite there, yet.
1. is there a list for DIDs which allow write access? I guess that list would be device specific.
2. when I read DID 424 I get this:
```
open3e -c can0 -dev vcal -r 424 -v
vcal 424 MixerOneCircuitRoomTemperatureSetpoint {"Comfort": 25.0, "Standard": 21.0, "Reduced": 3.0, "Unknown2": "0000", "Unknown1": 0}
closing 0x680 - bye!
```
How can I change a single value in this dictionary? (I assume it's a Python dictionary.)

@MyHomeMyData : what exactly do you mean by "extended write service 'sid 77'"

Please bare with me, I'm new to this. I promise to write up my learning and post it here.

Philip-Wiege, 2024-11-04:
Ich glaube, wir können laut deinem Profil auch auf Deutsch schreiben.

Soweit ich das den Beispielen auf der Titelseite von open3e entnehmen kann, kann man diese "Komplexen" DIDs entweder raw schreiben, also alle Bytes explizit auf einmal oder über JSON als Dictionary. Die Temperaturen natürlich nach Wunsch anpassen.

open3e -c can0 -dev vcal -w 424='{"Comfort": 25.0, "Standard": 21.0, "Reduced": 3.0, "Unknown2": "0000", "Unknown1": 0}'

@MyHomeMyData Stimmt das? Kann man so einfach genau den Text, den man beim Lesen bekommt, modifizieren und wieder zurückschreiben? Muss man die beiden Unknown Felder mit übermitteln? Gibt es eigentlich schon eine Methode, einzelne Sub-DIDs in den komplexen DIDs zu schreiben?

MyHomeMyData, 2024-11-04:
> How can I change a single value in this dictionary? (I assume it's a Python dictionary.)

Pls. refer to [readme](https://github.com/open3e/open3e?tab=readme-ov-file#using-json-data-format): `open3e -c can0 -w 538='{"Comfort": 25.0, "Standard": 21.0, "Reduced": 18.0, "Unknown2": "0000", "Unknown1": 0}'` 

> is there a list for DIDs which allow write access? I guess that list would be device specific.

Yes, such a [list](https://github.com/open3e/open3e/blob/master/src/open3e/Open3Edatapoints_writables.json) exists. However, for some of those data points you will run in "negative response". We don't have a list for that part, yet.

> what exactly do you mean by "extended write service 'sid 77'"

It's an additional approach using another service of UDSonCAN. Presently implemented in branch develop of open3e, described [here](https://github.com/open3e/open3e/tree/develop?tab=readme-ov-file#extended-writing-service-internal-can-bus-only-experimental). This service is available on **internal** CAN bus only and ment for specific use cases only. You don't need it for did 538.

MyHomeMyData, 2024-11-04:
@Philip-Wiege
> Gibt es eigentlich schon eine Methode, einzelne Sub-DIDs in den komplexen DIDs zu schreiben?

Verstehe die Frage nicht. Bitte ein Beispiel.

Philip-Wiege, 2024-11-04:
@MyHomeMyData Als Sub-DID würde ich zum Beispiel die Komfort Temperatur innerhalb des DID 424 bezeichnen. Interessant wäre es, wann man quasi 424.0 lesen und schreiben könnte und somit direkt an den Dictionary Wert für "Comfort" rankäme.

MyHomeMyData, 2024-11-04:
@Philip-Wiege Verstehe. Das können wir in open3e implementieren, habe aber Zweifel, ob es die Bedienung vereinfacht. Es kommt ja dann eine weitere Adressierung hinzu (424.0), die stimmen muss und die verstanden werden muss.

Der interne Ablauf wäre zudem komplizierter. Da immer nur die komplette Did geschrieben werden kann, müsste die Did zuerst gelesen werden, das gewünschte Elemente innerhalb der Struktur geändert werden und dann die gesamte Struktur geschrieben werden.

Gehen tut das, im ioBroker-Adapter mache ich es im Prinzip so. Allerdings nutzte ich die in der Objektstruktur von ioBroker bereits vorhandenen Daten, um die komplette Did-Struktur zusammenzubauen. Und die Benutzer-Schnittstelle ist halt viel einfacher.

Philip-Wiege, 2024-11-04:
@MyHomeMyData Das stimmt, das macht es komplizierter. Ich überlege noch, ob mein Home Assistant Code das übernehmen soll. 
@bomilkar Ich glaube, du müsstest die benötigten Infos haben :) Kurzes Feedback wäre schön, dann könnten wir hier erfolgreich schließen.

bomilkar, 2024-11-04:
Vielen Dan an @MyHomeMyData und @Philip-Wiege !
Das beantwortet meine Fragen, zum jetzigen Zeitpunkt.

Ich habe die Vitocal-250A seit 2 Wochen und bin ziemlich enttäuscht über die Steuerung der Anlage.
1. es gibt keinen Raumtemperatur-Sensor und ich frage mich, wie will das Ding die Temperature regeln, wenn es diese nicht kennt [im Forum](https://community.viessmann.de/t5/Waermepumpe-Hybridsysteme/Kein-Raumtemperatur-Sensor-kann-das-funktionieren/td-p/479161)
2. der Pufferspeicher hat den Nachteil, dass er nach der Nachtabsenkung das Aufheizen der Heizkreisläufe verzögert. Warum heizt er nicht zuerst die Heizkreisläufe bevor es (überschüssige Energie) in den Pufferspeicher packt.
3. zum Begin der Nachtabsenkung ist der Pufferspeicher warm. Warum versucht die Anlage nicht, die Energie im Pufferspeicher zu halten? Stattdessen laufen die Umwälzpumpen weiter und verdaddeln die Energie. Es sei denn, die Soll-Temperatur während der Nachtabsenkung wird auf 3° gesetzt.

Alles ziemlicher Murks. Deswegen habe ich angefangen, in die Regelung einzugreifen. Mal sehen, wie gut mir das gelingt.

closed 2024-11-04 by bomilkar

Philip-Wiege, 2024-11-04:
> Vielen Dan an @MyHomeMyData und @Philip-Wiege ! Das beantwortet meine Fragen, zum jetzigen Zeitpunkt.
>
> Ich habe die Vitocal-250A seit 2 Wochen und bin ziemlich enttäuscht über die Steuerung der [...]

Ich bin auch etwas enttäuscht und bin aktuell dabei, mit Home Assistant eine übergelagerte Steuerung aufzubauen. Zu deinen Punkten:

1. Naja, das ist halt eine klassische Steuerung statt einer Regelung. Stichwort "Open Loop". Der Wärmeverlust korreliert mit der Außentemperatur, welche gemessen wird. Die Vorlauftemperatur wird proportional angepasst, welche mit der Wärmeabgabe korreliert. Muss halt genau eingestellt werden. Es gibt übrigens von Viessmann ein Raumthermostat welches per Funk an die Vitocal 250 angebunden werden kann. Dann kann man die Anlage raumgeführt "geregelt" laufen lassen.

2. Der Puffer sollte ja Nachts noch warm genug bleiben, um am morgen wieder aufheizen zu können. Wenn der Puffer hydraulisch gut eingebunden ist, fließt der Volumenstrom morgens direkt von der WP in den Heizkreis. Nur überschüssiger Volumenstrom geht durch den Speicher und wärmt diesen langsam auf.

3. Ich hab bei uns (kein Pufferspeicher, nur FBH, PV Anlage) Nachts in der Übergangszeit auf "reduziert 3 Grad" gestellt. Außerdem habe ich die Anlage so kodiert, dass sie wirklich erst unter 3 Grad die Umwälzpumpe laufen lässt. Somit ist überhalb von 3 Grad die Pumpe wirklich Nachts aus.

bomilkar, 2024-11-04:
@Philip-Wiege : wie ich im Forum schon beschrieben habe, gibt es bei mir große Fenster, durch die die Sonne mitheizen kann (wenn sie denn scheint). Es stimmt, dass der Wärmeverlust mit der Außentemperatur korreliert, aber wenn eine Energiezufuhr (Sonne, Ofen, ...) nicht einkalkuliert wird, geht die Steuerung in die Binsen. Den Vitotrol 100-E hat mein Heizungsbauer bestellt. Mal sehen was er tut. Denn dokumentiert ist da nix.

Du schreibst "Wenn der Puffer hydraulisch gut eingebunden ist, fließt der Volumenstrom morgens direkt von der WP in den Heizkreis". Das klingt plausible, aber meine Anlage verhält sich nicht so. Liegt vielleicht daran, dass 2 von den Anschlüssen des Pufferspeichers nicht so angeschlossen sind, wie das in der Montageanleitung beschrieben ist. Die Frage muss mir der Heizungsbauer noch beantworten.

Übrigens: ich glaube, wir habe schon bei XCSoar und OpenVario miteinander zu tun gehabt. :-)