open3e/open3e #145: Falsche Dekodierung des OpState im DID 1415 MixerOneCircuitOperationState

Philip-Wiege, 2024-11-06:
Hallo zusammen,

im DID 1415 MixerOneCircuitOperationState gibt es ja den OpState und den OpMode. Meiner Beobachtung nach stimmt die Dekodierung des OpState nicht. Meine Vitocal 250-A ist aktuell im Modus Heizen, reduziert und zusätzlich noch im Sparmodus, weil es draußen zu warm ist. Der numerische OpState ist 6, was als Antifreeze dekoriert wird. Meiner Meinung nach müsste da aber energy saving oder energy saving reduced rauskommen.

Ich habe das Gefühl, wir müssten da nochmal alle Kombinationen aus Heizen / Kühlen + reduziert / normal / komfort + Sparmodus an / aus durchspielen. Hoffentlich kommt da überhaupt eine Eindeutige Zuordnung raus, welche nicht vom aktuellen Heizmodus abhängt.

Grüße
Philip
[image]
[image]

Philip-Wiege, 2024-11-06:
Ich habe nochmal alle Kombinationen an meiner Vitocal durchgespielt und meine Aufzeichnungen in HomeAssistant durchgeschaut. Ich komme auf die folgende Dekodierung:

MixerOneCircuitOperationState

**Mode**

0: Off
1: Heating
5: Cooling
255: Heating / Cooling

**State**

0: Off
1: Heating Reduced (Heating or Heating + Cooling)
2: Heating Normal (Heating or Heating + Cooling)
3: Heating Comfort (Heating or Heating + Cooling)
4: ?
5: ?
6: Energy Save Heating Reduced / Energy Save Cooling Reduced / Energy Save Heating + Cooling Reduced
7: Energy Save Heating Normal / Energy Save Cooling Normal / Energy Save Heating + Cooling Normal
8: Energy Save Heating Comfort / Energy Save Cooling Comfort / Energy Save Heating + Cooling Comfort
9: Cooling Reduced (Cooling or Heating + Cooling)
10: Cooling Normal (Cooling or Heating + Cooling)
11: Cooling Comfort (Cooling or Heating + Cooling)

Zusammen einige Zustände fallen zusammen. Zum Beispiel ist 6 immer Sparmodus reduziert, egal ob Heizen, Kühlen oder Heizen + Kühlen. Zusammen mit der Info aus des Modbus Gateway Dokument schlage ich folgende neue Dekodierung vor:

MixerOneCircuitOperationState

Mode

0: Off
1: Heating
5: Cooling
255: Heating / Cooling

State

0: Off
1: Heating Reduced
2: Heating Normal
3: Heating Comfort
4: Fixed Flow Temperature
5: Frost Protection
6: Energy Saving Reduced
7: Energy Saving Normal
8: Energy Saving Comfort
9: Cooling Reduced
10: Cooling Normal
11: Cooling Comfort

[image]

Philip-Wiege, 2024-11-06:
Welche Source Dateien hast du denn verwendet? Die aus der Android App?

bu-na, 2024-11-08:
Hier mal als Ergänzung zur Kundendatenpunktliste die Auflistung der möglichen OperatingPrograms aus der API:
( aktueller Wert --> viessmannapi.0.xxxxxxx.0.features.heating.circuits.0.operating.programs.active.properties.value.value)

viessmannapi.0.xxxxxxx.0.features.heating.circuits.0.operating.programs:
- active
- comfort
- comfortCooling
- comfortCoolingEnergySaving
- comfortEnergySaving
- comfortHeating
- eco
- fixed
- forcedLastFromSchedule
- frostsprotection
- normal
- normalCooling
- normalCoolingEnergySaving
- normalEnergySaving
- normalHeating
- reduced
- reducedCooling
- reducedCoolingEnergySaving
- reducedEnergySaving
- reducedHeating
- standby
- summerEco

Um das aktuelle Programm abzubilden, muss man vermutlich mehrere DIDs geeignet verknüpfen.
Ich denke, die Abbildung des DID-1415 entsprechend Kundendatenpunktliste ist soweit korrekt.

Philip-Wiege, 2024-11-08:
Wie würdest du denn diese lange Liste auf die numerischen Zustände mappen? Meiner Meinung nach ist die aktuelle Enum auf jeden Fall falsch, weil dort bei mir Antifreeze Protection rauskommt, wenn die Anlage auch bei warmen Tagen in den Sparbetrieb geht.

Grüße
Philip

philippoo66, 2024-11-10:
sorry for my late reply...

> Welche Source Dateien hast du denn verwendet? Die aus der Android App?

ja. aus Vicare apk, ist aber schon ne Weile her

> Meiner Meinung nach ist die aktuelle Enum auf jeden Fall falsch,

vielleicht ist sie in der neuen apk geändert? ich guck nachher mal es sein denn jemand von euch wäre schneller ;-)

wangnick, 2025-02-05:
Was ist denn mit State.ID 255, wie sie meine Vitocal 250-A SW 2440 für nicht genutzte Heizkreise liefert? Das ist ja auch nicht schön ...

```
    "1417 MixerThreeCircuitOperationState": {
        "Mode": {
            "ID": 0,
            "Text": "Off"
        },
        "State": {
            "ID": 255,
            "Text": "not found in OpStates"
        }
    },
```

philippoo66, 2025-02-06:
moin @wangnick !

du hast ne Menge 'Ungereimtheiten aufgedeckt' in letzter Zeit - danke erstmal dafür!! Ich antworte hier mal 'stellvertretend'. 

Die Codecs sind ja 'empirisch' zugewiesen, teilweise auch nur vermutet. Genauso die zugehörigen Listen bei Aufzählungstypen. Die Enums stammen großteils aus der Viguide APK.

Wir können gern Enums hinzufügen und die dann entsprechend zuordnen. Überhaupt können wir auch die Codec Zuordnungen ändern. Wir haben es allerdings auch schon erlebt, dass Änderungen gemäß eines 'Modells' dann dazu geführt haben, dass es danach auf einem anderen Modell falsch war.

Leider haben wir kein Logbuch geführt, welche DID Codierungen auf welchen Anlagen verifiziert wurden, weche nur geraten sind, und so weiter.

Alles was wir tun können, ist glaubich, dass du deine Vorschläge in die datapoints.py einpflegst, abei die bestehende Codierung bitte nur auskommentierst und nicht löschst, entsprechend markierst (einen Kommentar daneben mit eindeutiger 'Quellenangabe', z.b. `#wangnick Vitocal 250-A SW 2440` und wir das dann erstmmal in den develop branch übernehmen.

Das Zauberwort zur 'echten' Lösung heisst glaubich ODX. Ich habe vor 10 Tagen meine Viessmann Kontakte aus den Online Konferenzen angemailt und ihnen dargelegt, dass es auch für Viessmann von Vorteil wäre, wenn wir (bzw. unsere Projekte) die ODX Inhalte nutzen könnten (schliesslich verweist inzwischen sogar der Viessmann Kundendienst auf open3e etc, wenn Kunden fragen, wie sie ihre Anlage in eine Gebäudeautomatisierung einbinden können). Aber bisher keine Antwort...

wangnick, 2025-02-06:
[WILCO](https://de.wikipedia.org/wiki/Wilco_(Funk))

MyHomeMyData, 2026-10-25:
We finally tracked this down. Thanks to everyone who dug into this over the past two years, especially Philip-Wiege's systematic testing and wangnick's observation about value 255.

The root cause: our `OpStates` enum was taken from an older viGuide app version, which apparently had a different (and apparently wrong) numbering than the current one. We decompiled the current viGuide app and found the explicitly-valued `DOP_1ByteTextTableUint8OperaState` enum, which matches **exactly** what Philip-Wiege found through field testing, including the value 255 for "no demand" that wangnick observed on unused circuits.

Corrected decoding, now in `develop` (v0.7.11):

```python
"OpStates" : {
    0: "Off",
    1: "Heating Reduced",
    2: "Heating Normal",
    3: "Heating Comfort",
    4: "Fixed Flow Temperature",
    5: "Frost Protection",
    6: "Energy Saving Reduced",
    7: "Energy Saving Normal",
    8: "Energy Saving Comfort",
    9: "Cooling Reduced",
    10: "Cooling Normal",
    11: "Cooling Comfort",
    255: "No request",
},
```

Value 6 is now "Energy Saving Reduced" instead of "Antifreeze protection", matching Philip-Wiege's original report, and 255 is defined instead of falling through to "not found in OpStates".

Closing this one out.

closed 2026-10-25 by MyHomeMyData