open3e/open3e #147: doip: Connection Refused

liborw, 2024-11-07:
Hi, I'm trying to connect to my Vitodens 200-W B2KF using doip:

```
> open3e_depictSystem -d 192.168.1.23 -s
read DID enums ...
3038 DIDs listed.
scan COB-IDs 0x680 to 0x6ff ...
# ECU 0x680: [Errno 111] Connection refused
# ECU 0x681: [Errno 111] Connection refused
# ECU 0x682: [Errno 111] Connection refused
...
0 responding COB-IDs found.
write devices.json ...
done.

configuration:
{}

run open3e with -mqtt and -a to get EVERYTHING on your MQTT app.
```

Is there some configuration to enable or am I missing someting?

liborw, 2024-11-13:
Ok, then I would loose ViCare and other connections...

Philip-Wiege, 2024-11-13:
Exactly. The only way to have ViCare cloud connection and local access via Open3E at the same time is to use the CAN bus.

Philip-Wiege, 2024-11-13:
Issue can be closed. No software problem. DoIP can only be accessed if device is in access point mode.

closed 2024-11-13 by Philip-Wiege