open3e/open3e #149: Feature Request: Writing to complex DIDs

Philip-Wiege, 2024-11-10:
As far as I know there is currently no way to write complex DIDs without using raw values.

It would improve open3e if individual entries within complex DIDs can be written directly using the Codecs. Open3e would read the complex DID change only the accessed sub-DID and write back the new data.

I will try to implement if I find some spare time and maybe create a branch / fork for this. Feedback is welcome in german or english.

MyHomeMyData, 2024-11-10:
It is possible to write complex data using JSON format, described in [readme](https://github.com/open3e/open3e?tab=readme-ov-file#using-json-data-format).

However, it's not possible to access a single element out of a strcuture, e.g. 538.0=0. I'm not sure if there really would be a big benefit compared to json based solution. For sure we would significantly add complexity to open3e code base.

Philip-Wiege, 2024-11-11:
@MyHomeMyData I get your point. In my opinion external components like HomeAssistant would benefit from the possibility to write to single Sub-DIDs without handling the other Sub-DIDs in a complex DID. I suggest that I will try to implement the functionality inside of open3e and maybe we can review once it is finished.

MyHomeMyData, 2024-11-11:
Ok, understood. Let's go ahead. Best would be to do this on a new branch.

Philip-Wiege, 2024-11-11:
Do I need to fork this repo or do I have the permission to create a branch in this repo?

Philip-Wiege, 2024-11-11:
I created the branch: https://github.com/open3e/open3e/tree/149-feature-request-writing-to-complex-dids
It is my first time so I hope I didn't mess up anything. As soon as I have a working version we could review.

MyHomeMyData, 2024-11-11:
@Philip-Wiege Did you use master as base for this branch? We better should derive it from develop to simplify merging later on.

Philip-Wiege, 2024-11-11:
Yes, I branched off from master which was maybe not the best idea. I will try to fix that and rebranch from develop.

MyHomeMyData, 2024-11-11:
Ok. Simplest way may be to delete the branch ("View all branches") and the to create it again.

Philip-Wiege, 2024-11-26:
@MyHomeMyData Falls du Lust hast, kannst du dir meinen Branch mal per Diff anschauen. Im Grunde habe ich zwei Funktionen in der Open3Eclass hinzugefügt, mit denen man Sub-DIDs aus komplexen DIDs lesen und schreiben kann. Das beste Feature daran ist eigentlich, dass man einzelne Sub-DIDs ändern kann, ohne sich um die anderen Sub-DIDs zu kümmern.

So kann man zum Beispiel die Komfort Soll-Temperatur einfach einzeln ändern. Die Funktion kümmert sich darum, dass die anderen Bytes unverändert wieder zurückgeschrieben werden:

OEClassHPMU.writeByComplexDid(did=424, subDid=0, val=24.0, raw=False, simulateOnly=False, verbose=True)

MyHomeMyData, 2024-11-26:
@Philip-Wiege Wow, das sieht super aus. Habe die Details noch nicht durchdrungen, aber der Ansatz ist klar und gut verständlich. Würde das gerne mal ausprobieren. Wie ruft man das auf? Wenn ich die Komfort Soll-Temperatur ändern will, wie muss dann die Kommandozeile aussehen?

MyHomeMyData, 2024-11-26:
Ah, verstehe. Ich mach mir mal ein paar Gedanken.

Ich arbeite aber grade an einem anderen, handwerklichen, Projekt und das will ich vor Weihnachten in eine vorzeigbare Form bringen. Natürlich ist wieder alles komplizierter und aufwändiger als gedacht. Morgen steht der nächste Ausflug in den Baumarkt auf dem Programm ...

Was ich sagen will: Hab grade keinen Kopf, ein open3e-Thema anzugehen. Kann aber gerne Fragen beantworten oder Hinweise geben.

philippoo66, 2024-11-26:
@Philip-Wiege  kannst du beim nächsten mal bitte eine 'handlichere' branch-Benennung erwägen?! das unfallfrei einzutippen ist schon eine echte Herausforderung... ;-)

philippoo66, 2024-11-27:
hübsch! ich hätte aber 2 "Wünsche":

1. statt dem endIndexSubDid +1 und -1 Gewiggel das Ding stopIndexSubDid zu nennen (analog zu startIndexSubDid) und das Ganze +1 und -1 weg zu lassen.

2. die Unterscheidung
```
                    if (subDid == numSubDids-1): #if is last sub DID
                        rawDidDataNew = rawDidData[:startStringIndexSubDid] + encodedDataHexString
                    elif(subDid == 0): # if is first sub DID
                        rawDidDataNew = encodedDataHexString + rawDidData[endStringIndexSubDid:]
                    else:
                        rawDidDataNew = rawDidData[0:startStringIndexSubDid] + encodedDataHexString + rawDidData[endStringIndexSubDid:]
```
weglassen und einfach nur
```
rawDidDataNew = rawDidData[:startStringIndexSubDid] + encodedDataHexString + rawDidData[endStringIndexSubDid:]
```
zu schreiben.

Schön wäre auch, wenn wir wie bei den ganzen Codecs mit string_ascii, string_bin und string_len arbeiten könnten - das würde das Verständnis erleichtern, für wer sich schon mit den Codecs befasst hat.

Und nahezu perfekt wäre es, wenn wir das readByComplexDid in dem writeByComplexDid 'wiederverwenden' würden, statt es erneut zu programmieren ;-)

philippoo66, 2024-11-27:
> Leider gibt es noch keine Anbindung an das "open3e" Hauptskript

hier sollten wir einfach die Komplexe Adressierung erweiten! Also z.B. -r 0x680.256.7 für die VIN. Dann funktioniert automatisch alles wie gehabt. Sollte auch mit dem Listen Mode so sein.

philippoo66, 2024-11-27:
@Philip-Wiege würdest du dir bitte mal das readByComplexDid_p angucken?! ich find das so in der Art deutlich klarer und übersichtlicher (und auch kompakter, wobei ich nich immer Freund von Kompaktem bin, wenn dadurch die Lesbarkeit leidet, aber das ist in diesem Fall glaubich nicht der Fall).

Die verbose Info gibt's dabei zwar nur noch für den ausgewählten Sub, aber ich denke das schadet nicht, sobald es läuft. 

Damit es in's Schema passt, abe ich den return noch erweitert (alle read... returnen den Wert und den Bezeichner).

Ich habe es (aus bekannten Gründen....) nicht getestet, aber es sollte eigentlich so funktionieren.

MyHomeMyData, 2024-11-27:
> kannst du beim nächsten mal bitte eine 'handlichere' branch-Benennung erwägen?! das unfallfrei einzutippen ist schon eine echte Herausforderung... ;-)

@philippoo66 Wo musst Du das eintippen? Der Befehl git unterstützt Vervollständigung per Tab-Taste. Z.B. `git checkout 149` Tab-Taste vervollständigt den Branch-Namen.

Philip-Wiege, 2024-11-27:
@philippoo66 Deine Anmerkungen sind korrekt ;) Der Code ist noch wenig optimiert aber er läuft immerhin. Deine Optimierungen sehen ganz gut aus und ich würde sie mal zusammenführen und bei meiner Anlage testen. Raw Schreiben für Sub DIDs geht auch schon. Die Funktion kann halt "nur" Sub-DIDs schreiben und nicht den ganzen komplexen DID. Eventuell bekommt man es hin, dass man wenn kein Sub-DID übergeben wird, automatisch den ganzen komplexen DID schreiben muss. Das MUSS dann aber raw sein. Aktuell nimmt sie, wenn man keinen Sub-DID Index übergibt einfach den Index 0.

@MyHomeMyData du würdest dir also sowas in der Art Wünschen?

-r 0x680.424.0 oder -r 0x680.424.Comfort
-w 0x680.424.0=24.0 oder -w 0x680.424.Comfort=24.0
-raw -w 0x680.424.0=FA00 oder -raw -w 0x680.424.Comfort=FA00
-raw -w 0x680.424=FA00FA00FA00

MyHomeMyData, 2024-11-27:
Ja, so stelle ich mir das vor. Sehe ich aber als nice-to-have. Der wesentliche Fortschritt wird sein, überhaupt Sub-Dids adressieren zu können.

philippoo66, 2024-11-27:
> Wo musst Du das eintippen? Der Befehl git unterstützt Vervollständigung per Tab-Taste

das hatte ich für MS Terminal erinnert und mir die Haare gerauft warum is es nicht hinbekomme ;-) ok, werde es mal in git probieren.

> Zusätzlich wäre aber klasse, wenn auch -r 0x680.256.VIN funktionieren würde.

dann müsste ja eigentlich auch -r 0x680.BusIdentification.VIN funktionieren... naja -r myVito.256 geht ja schon, auf die Weise können wir dann die andren 'Klartexte' auch noch hinbekommen. Ich guck mir diesbezüglich mal das Komplexe Addressieren an, das war nicht ganz trivial, weil es ja auch noch die Sache mit den Kommas gibt (-r 0x680.257,258)

philippoo66, 2024-11-27:
> Eventuell bekommt man es hin, dass man wenn kein Sub-DID übergeben wird, automatisch den ganzen komplexen DID schreiben muss. Das MUSS dann aber raw sein. Aktuell nimmt sie, wenn man keinen Sub-DID Index übergibt einfach den Index 0.

die Unterscheidung, ob readByDid() oder readByComplexDid() aufgerufen wird, wird dann von der komplexen Adressierung getroffen, das ist soweit kein Problem. Aber das raw Flag ist ein anderer Mechanismus / eine andere cmdln Option, von daher sollte es eben funktionieren. Aber das kann nicht viel sein, weil ja das raw Flag schon Parameter im Aufruf ist. ggf. müssen wir den einfach nur 'nach unten durchreichen'. Das ist eben der Vorteil, wenn man schon existierende Bausteine nutzt wo immer es geht ;-)

philippoo66, 2024-11-27:
@MyHomeMyData  ich hab `git switch 14` eingegeben und die Tab Taste bewirkt garnix - was mache ich falsch?

philippoo66, 2024-11-27:
ich hab mir grad mal git diff develop 149-feature-request-writing-to-complex-dids angeguckt - das wird ne echte Würgerei wenn wir das zusammenführen wollen

@Philip-Wiege ich denke die Additions/Änderungen in 149-feature-... sind nicht so viele? könntest du vielleicht den Brannch noch mal neu machen basierend auf develop und die paar aktuellen Änderungen dann da rein kopieren?!

MyHomeMyData, 2024-11-27:
> ich hab mir grad mal git diff develop 149-feature-request-writing-to-complex-dids angeguckt - das wird ne echte Würgerei wenn wir das zusammenführen wollen

Github kann das mergen:
[image]

MyHomeMyData, 2024-11-27:
> ich hab git switch 14 eingegeben und die Tab Taste bewirkt garnix (mit checkout das Gleiche) - was mache ich falsch?

Was für ein Terminal verwendest Du? Unter Windows oder Linux?

philippoo66, 2024-11-27:
doch Terminal? unter Windows. Ich hatte jetzt zwischendurch gedacht, das wäre für das Terminal in VS Code?!? jetz bin icht total verwirrrt....

MyHomeMyData, 2024-11-27:
In VS Code mache ich das über das 3-Punkte-Menü bei "Source-Control", dann "Checkout-to" und aus der Liste den gewünschten Branch anklicken.

philippoo66, 2024-11-27:
Mist, ich wollte grad mal den Sub und die Klartextsache in die Komplexe Adressierung einbauen, aber die DID und Sub Klartexte stehen in der ECU Klasse, die ECU Liste und die Komplexe Adressierung aber in dem Client :-/ Das müssen wir dann ausgiebiger testen fürchte ich. 

Ich mach es jetzt so, dass das komplexe Evaluieren immer ein 3er Tuple liefert und statt read/writeByDid immer read/writeByComplexDid aufgerufen wird, das dann bei None im 3. Teil read/writeByDid aufruft, ok?!

@Philip-Wiege  wozu hast du jetzt die read/writeGenericDid() eingeführt?

philippoo66, 2024-11-28:
so, complex addressing ist erweitert auf subs, alle 'Arten' könnten funktionieren, named addressing auch:
```
    # complex addressing: 0x680.[257,258,259] or 0x680.256 or 257,259,261 or 256 ...
    # also possibel 0x680.[257.0,258.1,259]  or 0x680.256.1 ...
    # also Vcal.BusIdentification.BusAddress or Vcal.256.BusAddress ...
    # if sub gets addressed, ecu must be given, otherwiese not clear if ecu.did or did.sub
```
zumindest per command line ist zumin das Lesen auch von subs und named eingebaut.

Ich hab natürlich noch nix getestet, ich müsste jetzt doch mal wieder meine virtualE3 an den Start bringen, vielleicht brauch ich da noch mal Hilfe. @MyHomeMyData was war noch mal der clue, den wir letztens am Telefon 'identifiziert' hatten? und muss ich erstmal eine pip Installation machen?

Philip-Wiege, 2024-11-28:
Read/Write Generic sollen Funktionen werden, welche sämtliche Arten von DIDs in einer Funktion bündeln. Das Ausgabeformat ist identisch zu den ursprünglichen Funktionen, lösst sich also einfach einbinden in das bestehende System.

Die ReadGenericDid Funktion kann aber alles lesen: Simple, Complex, Sub-DIDs, Raw, Pure

philippoo66, 2024-11-28:
supi, danke!

Wir sollten nochmal überlegen, ob wir wirklich den Punkt als 'Trenner' für Subs nehmen wollen!

bei `ecu.did.sub` ist das ja kein Problem, aber `ecu.did` und `did.sub` sind so nicht zu unterscheiden.

wenn wir bsw einen Doppelpunkt für den Subindex nähmen, gige `ecu.did:sub`, `ecu.did` und `did:sub` !?

Philip-Wiege, 2024-11-28:
Hui, hier geht es ja gut vorwärts :)

Cool fände ich für die Adressierung ecu:did.sub Einfach nur, weil Viessmann in seinen Service Dokumenten auch DID.Sub verwendet. Das wäre aber genau die Option, welche nicht rückwärtskompatibel ist?

Wenn ich es richtig verstehe, hat @philippoo66 quasi meine readByComplexDID und writeByComplexDID Funktion an das Open3E Format angepasst und bereits in die Infrastruktur der Kommandozeile eingebunden? Hast du dort auch schon die Befehle per MQTT eingebunden?

Ich habe jetzt die beiden Funktionen readGenericDid und writeGenericDid beigesteuert. Diese beiden Funktionen sind so universell, dass man damit alle DIDs gleich behandeln kann. Man muss nicht mehr zwischen komplex und normal unterscheiden. Beide Funktionen benutzen unter der Haube noch die "alten" Funktionen readPure, readByDid und readByComplexDid.

Mein Wunschzustand wäre, dass Open3E nur noch über die beiden generic Funktionen geht (mit den korrekten Parametern) und die beiden Funktionen somit die einzigen Interfaces für UDS Schreiben und lesen sind. Somit wären die Schnittstellen stark reduziert.

MyHomeMyData, 2024-11-28:
> wenn wir bsw einen Doppelpunkt für den Subindex nähmen, ginge ecu.did:sub, ecu.did und did:sub !?

Ich bin dafür, generell bei Punkten zu bleiben. Dafür würde ich in Kauf nehmen, für die Spezifizierung von Subs immer die ecu angeben zu müssen. Wenn wir zusätzlich Doppelpunkte zur Trennung einführen, wird es aus meiner Sicht zu verwirrend.

philippoo66, 2024-11-28:
@Philip-Wiege  ich habe bisher nur das Lesen und das nur per command line arg eingebaut (und bisher noch nicht probiert), der Rest kommt noch. Mit MQTT/Listen ist nicht ganz so trivial.

Ich hab mir noch mal die generic read angeschaut - soweit so gut, im weiten Zügen ist sie gleich der readByDid, die ich oben eingesetzt habe. Mich stören aber ein paar Sachen:

- die große Verschachtelungstiefe. Warum nicht gleich returnieren wenn angesagt? das macht die Sache viel übersichtlicher.
- Parameter sind da optional, die aufgrund der gesamten Programm-Struktur aber nicht optional sein dürfen.
- ein Defaultwert -1 um None 'zu verkörpern' macht keinen Sinn.
- die for Schleife, die mittels Vergleich in jedem Durchlauf auf einen Index gebraket wird, statt sie einfach nur bis zum Index laufen zu lassen, ist 'Unsinn'.
- beim Bytes Zählen bei jeder Summenbildung *2 zu machen, statt wirklich die bytes zu zählen und am Ende im erkennbaren Zusammenhang mit dem string_ascii ein *2 zu machen, ist nicht schön, nicht klar und nicht straight 
- die Funktion ist mit der Parameterreihenfolge nicht kompatibel mit dem Code (ok, am Ende könnt man sowas anpassen, aber für zwischendrin ist das blöd).
- warum müssen die Parameter jetzt anders heissen? das gibt Probleme bei namentlicher Übergabe.
- warum muss readPure explizit aufgerufen werden, wenn readByDid bei gegebenen Anlass sowieso schon auf sie zurück fällt?
- ...

guck dir bitte die jetzige readByDid an. die macht eigentlich genau das, was deine generic macht (ich hab sie ja eigentlich von dir abgeschrieben), aber ich halte sie an vielen Stellen für sauberer.

Sei mir nicht böse, aber ich habe mindestens 2 Wochen lang Tage und Nächte konzentriert diese Struktur erschaffen so wie sie ist, und weiss genau, was ich warum getan habe. Es sind ja nur ein paar 'Kleinigkeiten', aber die führen zu Inkompatibilität oder mindestens zum Bruch mit 'Leitlinien'.

Wie gesagt, eigentlich habe ich dein Werk nur in die bestehende 'Form' eingepasst. Vielleicht übersehe ich was, bitte sag, was du warum an deiner readGeneric für besser hälst / was in der jetzigen readByDid dergegenüber fehlt.

das Write kommt dann...

Philip-Wiege, 2024-11-28:
Wenn deine Funktionen diese Features alle schon zusammenführen, kannst du die "überflüssigen" generic Funktionen gerne entfernen! Ich betrachte meinen Beitrag als getan und du hast die Struktur von Open3E besser im Blick.

Sobald die Anbindung an die Command Line läuft, können wir ja gut testen. Sag Bescheid, wenn ich eine Python Funktion auch ohne Anbindung an Command Line oder MQTT testen soll. Im IDE kann man die Funktionen ja auch live testen ohne dass sie bereits in den Open3eClient eingebaut sind.

philippoo66, 2024-11-28:
oki. 

oh, ich hab grad doch was gefunden - ich glaube in der jetzigen Version funktioniert raw beim sub lesen nicht? es wird immer decodet. das müsste man noch ändern. (das decode nur wenn nicht raw)

Wenn man es sich genau anguckt, passiert hier (bei nicht raw) natürlich ein mehrfaches string_bin -> string_ascii -> sub_string_ascii -> sub_string_bin -> decode...

Ist das bei deiner neuen (generic) Version effizienter? da wird glaubich nur einmal decodet, dafür aber dann alle subs, von denen dann nur einer gebraucht wird, richtig?

Hast du das eigentlich auch mit O3EArray, -List und -Enum ausprobiert?

Philip-Wiege, 2024-11-28:
Ich habs mit den Enum, List, Array noch nicht ausprobiert. Fallen dir unkritische DIDs ein zum Testen?

Ich glaube nicht, dass meine Generic Funktionen da wesentlich effizienter sind.

philippoo66, 2024-11-28:
Lesen sollte doch immer unkritsch sein, oder?

ok, ich werde zusehen, da auch 'your way' zu übernehmen und das Mehrfach-Hin-Her zu vermeinden.

MyHomeMyData, 2024-11-28:
Zum Testen empfehle ich Phils virtuelle E3. Verwende ich ständig für Testen meines ioBroker-Adapters. Habe dafür einen Fork von Phils Repo angelegt. Es geht so:
* Virtuelle Umgebung anlegen und aktivieren, VirtualE3 installieren: `git clone https://github.com/MyHomeMyData/vitualE3`
* Abhängigkeiten installieren: `cd vitualE3 && pip install -r requirements.txt`
* Virtuellen CAN vcan0 aufsetzen und starten
* Virtuelle E3 starten mit `python3 virtualE3.py -c vcan0 -cnfg devices.json`

Jetzt kann open3e über `vcan0` mit der virtuellen E3 kommunizieren und gefahrlos Daten lesen und schreiben.

philippoo66, 2024-11-29:
Hilfe! bitte...

[image]

wo zur Hölle muss das devices.json jetzt stehen?

ich habe mit `pip install --editable .[dev]` installiert. der Aufruf open3e ... benutzt jetzt die Dateien in /home/pi/open3e/src/open3e. Genau da steht auch das devices.json, aber open3e findet es nicht?!?

MyHomeMyData, 2024-11-29:
Es muss da stehen, wo Du den Befehl `open3e` absetzt. Natürlich müssen da auch die anderen von depictSystem erzeugten Dateien stehen. Eigentlich musst Du nur unsere [Anleitung ](https://github.com/open3e/open3e/wiki/030-Installation-und-Inbetriebnahme-von-open3E#open3e-installieren) befolgen.

philippoo66, 2024-11-29:
ah, ok. nachdem ich in /home/pi/open3e den `open3e_depictSystem -c vcan0` gemacht habe funktioniert in /home/pi/open3e auch `open3e -cnfg devices.json ...`

die virtual3E ist aber zwischendruch mal abgestürzt mit
```
Traceback (most recent call last):
  File "/home/pi/open3e/virtualE3/virtualE3.py", line 481, in <module>
    rdbiRequestReceived(ecu, did, msg)
  File "/home/pi/open3e/virtualE3/virtualE3.py", line 221, in rdbiRequestReceived
    buffer = bytes([0x10, dlen+3, 0x62, msg.data[2], msg.data[3]]) + txdata[:3]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: bytes must be in range(0, 256)
```

sagt dir das was @MyHomeMyData ?

MyHomeMyData, 2024-11-29:
Nein, sagt mir nix. An der Zeile habe ich auch nicht rumgefummelt, oder? Bei welcher Did passiert das?

philippoo66, 2024-11-29:
ne die is original mein ich. ich muss mal schauen ob ich das rekonstruiert bekomme. eigentlich sind die msg.data ja auch nur Bytes und können eigentlich nicht größer als 255 sein.

Das ist beim Scannen von 6E0 passiert, mehrfach. ich schau noch mal, aber erstmal die andere Baustelle...

philippoo66, 2024-11-29:
so, das subs lesen auch mit namentlich funktioniert schon mal.

[image]

wer Lust hat, kann es ja gern intensiv testen. An's Schreiben mache ich mich nachher.

philippoo66, 2024-11-29:
so, das Schreiben funktioniert an der command line auch erstmal:

[image]

ABER jetzt bin ich etwas verwirrt:

Im raw Mode kann ich ja nur raw Werte vorgeben (hex str). Wenn ich aber raw abschalte, um z.B. 

`-w 0x680.BusIdentification.bustype=CanInternal`

zu schreiben, will er automatisch den ganzen/kompletten json Brocken haben. 

Damit erledigt sich das Schreiben von Subs ja eigentlich, oder??

Philip-Wiege, 2024-11-29:
Also die reinen Funktionen, welche ich geschrieben hatte, können einzelne sub DIDs beschreiben. Ich habe aber nur numerische Werte probiert. Das CanInternal ist ja ein String. Das sollte aber durch die encode Funktion trotzdem gehen? Ist das CANInternal nicht aus einem Enum? Können die Enum Codecs auch encoden?

philippoo66, 2024-11-29:
na klar geht das. macht es ja auch. aber ich kann es nicht 'eingeben', weil es bei raw einen hex Wert erwartet und ohne raw das json Ding.

Wenn wir das jetzt ändern haben wir einen Kompatibilitätsbruch. ich überlege aber trotzdem, dazu die -j Option 'heranzuziehen'... 

Aber was ich meinte ist, dass man damit (mit dem json) ja eigentlich schon ganz bequem die Sub's schreiben kann. Sogar mehrere in einem Rutsch wenn man will...

philippoo66, 2024-11-29:
funktioniert das überhaupt?

Wenn ich den json lese, bekomme ich `{'BusAddress': 1, 'BusType': {'ID': 2, 'Text': 'CanInternal'}, 'DeviceProperty': {'ID': 31, 'Text': 'HPMUMASTER'}, 'DeviceFunction': {'ID': 2, 'Text': 'HMUMASTER'}, 'SW-Version': '20.509.2305.192', 'HW-Version': '2.612.101.4', 'VIN': '7470763218578124'}`

Was muss ich denn jetzt 'eingeben', um den Bustype zu ändern? (klar keiner wird den BusType ändern, aber mit der virtualE3 geht alles...)

geht das auch mit 'benannten' Werten?

philippoo66, 2024-11-29:
wenn ich eingebe 

`open3e -c vcan0 -cnfg devices.json -w 0x680.BusIdentification='{"BusAddress":
 1, "BusType": 4, "DeviceProperty": 31, "DeviceFunction": 2, "SW-Version": "20.509.2305.192", "HW-Version": "2.612.101.4", "VIN": "7470763218578124"}'`

bekomme ich 
```
  File "/home/pi/open3e/src/open3e/Open3Ecodecs.py", line 368, in encode
    raise ValueError("Ivalid input for OEEnum")
ValueError: Ivalid input for OEEnum
```

philippoo66, 2024-11-29:
wenn ich eingebe

`'{"BusAddress": 1, "BusType": "CanExternal", "DeviceProperty": "HPMUMASTER", "DeviceFunction": "HMUMASTER", "SW-Version": "20.509.2305.192", "HW-Version": "2.612.101.4", "VIN": "7470763218578124"}'`

kommt

```
  File "/home/pi/open3e/src/open3e/Open3Ecodecs.py", line 181, in encode
    raise Exception("not implemented yet")
Exception: not implemented yet
```

Philip-Wiege, 2024-11-29:
> na klar geht das. macht es ja auch. aber ich kann es nicht 'eingeben', weil es bei raw einen hex Wert erwartet und ohne raw das json Ding.
>
>
>
> Wenn wir das jetzt ändern haben wir einen [...]

Den Hauptvorteil der von mir vorgeschlagenen Logik zum Sub-DID Schreiben liegt darin, dass man sich um die restlichen Sub DIDs nicht kümmern muss. Aktuell muss man ja das JSON selber konstruieren und die gewünschten Werte selber modifizieren und den Rest unverändert reinkopieren.

philippoo66, 2024-11-29:
im README steht ja, man soll lesen, das Gewünschte ändern und damit schreiben. aber

`open3e -c vcan0 -cnfg devices.json -w 0x680.BusIdentification='{"BusAddress": 1, "BusType": {"ID": 3, "Text": "CanExternal"}, "DeviceProperty": {"ID": 31, "Text": "HPMUMASTER"}, "DeviceFunction": {"ID": 2, "Text": "HMUMASTER"}, "SW-Version": "20.509.2305.192", "HW-Version": "2.612.101.4", "VIN": "7470763218578124"}'`

funktioniert auch nicht...

scheint schon so, als ob es Sinn machen würde, das Explizite Schreiben einzubauen. 

Aber dann muss man genau wissen, was bsw. der O3EEnum Codec bei dem Wert liefern _würde_, auf den man setzen will. das ist auch nicht ohne!

philippoo66, 2024-11-29:
und das encode() muss bei allen Codecs gehen (zuindest bei den in Frage kommenden...)

philippoo66, 2024-11-29:
@Philip-Wiege  hattest du deine write eigentlich mal ausprobiert? bei mir schmieren die auch mit irgendwelchen Codec Problemen ab

philippoo66, 2024-11-30:
ok, `-w 0x680.BusIdentification.bustype=CanInternal` geht jetzt auch.

Wenn man die alte Variante mit json haben will, muss jetzt `-j` in der cmd line beigepackt werden. 
BRUCH DER ABWÄRTSKOMMPATIBILITÄT!

so, mit dem Listen mode müssen wir uns was überlegen. `{"mode": "read", "data":[256], "sub":2}` geht, aber das kann es ja nicht sein! Wir müssen überlegen, wie wir diese komplexe Adressierung verträglich mit dem Bestehenden kombinieren können!

MyHomeMyData, 2024-11-30:
@philippoo66 @Philip-Wiege Das geht ja richtig voran hier - sehr beeindruckend! Sorry, wenn ich mich grade kaum beteilige, ich bin mit anderen Themen grade ziemlich geerdet.

Inzwischen ist mir auch nicht mehr ganz klar, was jetzt funktioniert, bzw. funktionieren soll und was nicht:
* Funktioniert `open3e -c vcan0 -cnfg devices.json -j -w 0x680.BusIdentification='{"BusAddress": 1, "BusType": 4, "DeviceProperty": 31, "DeviceFunction": 2, "SW-Version": "20.509.2305.192", "HW-Version": "2.612.101.4", "VIN": "7470763218578124"}'`? Und brauchen wir das wirklich? Wir könnten das Schreiben (und Lesen) von Subs auch auf unstrukturierte Teile einschränken. Also hier z.B. auf `0x680.BusIdentification.BusAddress=1` usw.
* Beim Listener-Mode sollte das genauso aussehen, wie auf der Kommandozeile, also z.B. `{"mode": "read", "data":[0x680.BusIdentification.BusAddress]}`, Schreiben entsprechend `{"mode": "write", "data":[[0x680.BusIdentification.BusAddress,1]]}`. Für Sub-Strukturen wäre das `{"mode": "write", "data":[[0x680.BusIdentification,{"BusAddress": 1, "BusType": 4, "DeviceProperty": 31, "DeviceFunction": 2, "SW-Version": "20.509.2305.192", "HW-Version": "2.612.101.4", "VIN": "7470763218578124"}]]}`

philippoo66, 2024-11-30:
Hey Jürgen @MyHomeMyData , kein Ding! ich hatte mich ja auch lange Zeit sher 'zurückgehalten'...

ich fass mal zusammen: 

bisher habe ich das Ganze 'minimal invasiv' gehalten:

- die Interpretation von Strings statt Zahlen für DID und Sub ist in Form von kleinen 2 Fuktionen dazu gekommen, die einfach aus einem String eine Zahl machen, und die einfach im readByDid() dazwischen/vor- geschaltet sind (muss in der class sein, weil nur die das dataIdentifiers kennt), 
- das evaluate_complex_...() liefert jetzt statt einer Liste von `[ecu,did]` eine Liste von `[ecu,did,sub]`, wobei ggf. sub=None ist.
- das read/writeByDid() hat den optionalen Parameter sub=None hinten dran bekommen, und wenn nix oder None übergeben wird, fällt es auf das alte read/writeByDid() zurück. Mit etwas im Sub tut es das, was @Philip-Wiege entwickelt hat.

**Damit lies sich das read/write subs mit wirklich minimalsten Änderungen einbauen.**

**'Problematisch' wird es allerdings beim Listen Mode:**

Da der Listen Mode mit der Methodik 'data' als List von dids beim Lesen oder [did=val] beim Schreiben oder nicht unmittebar der Struktur der Komplexen Adressierung folgt, funktioniert hier die Erweiterung auf mögliches sub nich "von selbst".

Wenn man die Beispiele im README betrachtet, wird schnell klar, dass das aus mehreren Gründen nicht einfach harmoniert:

- die Komplexe Adressierung trennt die Bestandteile anhand von '.' (Punkten). Im Listen Mode gibt es Punkte aber bsw. auch bei zu schreibenden float Werten.
- die Liste an 'data' wird 1:1 als dids durchiteriert. Sollten wir das auf Subs erweitern, hätten wir items `did.sub`, die bei der komplexen Addressierung 'verboten' sind wegen nicht möglicher Unterscheidung zu dem bisher verwendeten und aus Kompatibilitätsgründen beizubehaltenden `ecu.did`.

Wir müssen hier also tiefer eingreifen.

Jetzt gilt es, ein System zu entwickeln, dass a) abwärtskompatibel ist und 'trotzdem' b) jetzt möglichst dem System der komplexen Adressierung folgt. 

Deine Beschreibung oben zeigt hier einen Weg auf (der mir gestern nacht mit rauchendem Kopf nicht so klar war): Wenn wir die items der 'data' list so behandeln, als wären es per ',' (Komma) getrennnte Argumente in der Kommandozeile, könnte die Sache ziemlich konsistent funktionieren. Ich schaue mal, das eval_complex_list() für Listenelemente kompatibel zu machen...

philippoo66, 2024-11-30:
ich glaube, mein Problem ist, dass ich mit dem, was ein json Parser tut und was er nicht tut, nicht so firm bin.

ich glaube das beste ist, ich reverte meine Änderungen im Listen Mode und du @MyHomeMyData machst das, wenn du mal irgendwann wieder Ruhe hast. Es funktioniert so lange ja wie gehabt, nur eben ohne explizites Lesen/Schreiben von Subs.

MyHomeMyData, 2024-11-30:
> ich glaube das beste ist, ich revertiere meine Änderungen im listen() und du @MyHomeMyData machst das, wenn du mal irgendwann wieder Ruhe hast. Es funktioniert so lange ja wie gehabt, nur eben ohne explizites Lesen/Schreiben von Subs.

@philippoo66 Moin Phil, ok, ich schaue es mir an. Versuche, es zeitnah zu machen, möchte aber nichts versprechen.

philippoo66, 2024-11-30:
@MyHomeMyData  ich glaube ich hab es schon. war doch nicht so schwierig. evts guggst du noch mal drüber, ich hab's noch nicht getestet, der Hund will erstmal in'n Wald bei dem Wetter... ;-)

[94 more replies]

closed 2025-02-06 by philippoo66