open3e/open3e #153: Open3EdataPoints.py and overlay files need to be synced

Philip-Wiege, 2024-11-14:
Hello,

is it correct, that the overlay files such as Open3EdatapointsVcal.py need to be synced with the Open3Edatapoints.py file?

For example the DID 2612 was added to Open3Edatapoints.py but does not exists in Open3EdatapointsVcal.py. The software says that no codec is defined when dev=vcal is chosen because Open3EdatapointsVcal.py is not up to date.

Does anyone know if it is possible to create an automated check if the overlay files are in sync with the main file?

philippoo66, 2024-11-15:
die `-dev` Option ist und bleibt ein Behelf / Überbleibsel aus anfänglichen Zeiten. Wir können sie nicht 'richtg' machen, weil sie auch immer nur für eine Firmware gilt, und ausserdem es zum Beispiel gerade bei vcal ja ganz viele unterschiedliche Geräte gibt.

Ganz ohne `-dev` sollte eigentlich immer die Codec Definition aus datapoints.py genommen werden. `-dev` ist auch eigentlich nur notwendig, wenn man ein `--scanall` macht (eigentlich müsste das 'readall' heissen, weil es alle DIDs aus der/den zugrundeliegenden Datenpunktliste/n leist; scannen tut depict_system...).

Philip-Wiege, 2024-11-15:
Ah ok, also ohne -dev wird die Hauptdatei verwendet. Nur wenn man das Overlay benutzen will, sollte man -dev angeben?
Trotzdem müsste ja dann die Overlay Datei zumindest etwas gepflegt werden, falls man sie nutzen will.

philippoo66, 2024-11-15:
> Nur wenn man das Overlay benutzen will, sollte man -dev angeben?

korrekt.

> Trotzdem müsste ja dann die Overlay Datei zumindest etwas gepflegt werden,

Irrtum. Wie willst du eine sowohl für Mono- als auch für Split Geräte zutreffende datapoints_vcal machen, und eine, die für 200er vdens gleichermassen passt wie für 300er?

MyHomeMyData, 2024-11-15:
@Philip-Wiege @philippoo66 Wie wäre es, wenn wir etwas radikaler vorgehen würden:
* die Option -dev schaffen wir ab. Nach der Installation von open3e wird immer die generische Liste verwendet. Wer device-spezifisch arbeiten möchte, muss eben zunächst ein depictSystem machen.
* die Overlay-Dateien entfernen wir auch. Die werden von depictSystem zur Laufzeit dynamisch erstellt. @philippoo66 Geht das überhaupt?

Das würde das System einfacher und schlanker machen und uns manche Diskussion zu -dev ersparen.

philippoo66, 2024-11-15:
ja, geht. Aber ich wäre trotzdem dafür, -dev zu behalten (siehe auch hier https://github.com/open3e/open3e/issues/154#issuecomment-2478685113 ff.). Nur sollten wir einen deutlicheren Hinweis in's ReadMe machen. Plane ich für heute NNachmittag, bin grad etwas in Eile

Philip-Wiege, 2024-11-15:
Die Overlay Dateien haben natürlich den Vorteil, dass eine andere Verwendung eines DID zwischen 2 Geräten aufgelöst werden kann. Wenn z.B. eine Vitocal und eine Vitodens den selben DID unterscheidlich codieren, brachen wir die Overlays.

@MyHomeMyData Meinst du mit dem zur Laufzeit generierten Dateien die zum Beispiel DataPoints_68C.py Datei, welche die DIDs von ECU 0x68C enthält? Damit ist auf jeden Fall abgesichert, dass man eine Liste von DIDs erhält, welche von jeder ECU unterstützt werden. Ob man die richtige Codierung hat, ist dadurch leider nicht sichergestellt :(

Ich lese gleich nochmal meine ECUs aus. Es gibt einen DID, der die ODX Datei beschreibt. Normalerweise werden in ODX Dateien die DTCs und ich glaube auch alle DIDs beschrieben. Normalerweise geben ECUs dem Tester via DID die Info, welche ODX Datei verwendet werden muss, um deren Daten korrekt zu interpretieren. Der richtige Jackpot wäre, wenn die ODX Dateien irgendwo in den APKs Sourcen vergraben wären...

Philip-Wiege, 2024-11-15:
Ich kapiers nicht.. Wenn ich den DID 1553 für die ODX Version Abfrage, bekomme ich Daten für den DID 0x3100 zurück und es gibt nen Fehler.

@MyHomeMyData Wie kann ich einen CAN Dump machen, um mal in den rohen CAN Verkehr reinzuschauen, was da wirklich zurückkommt?

MyHomeMyData, 2024-11-15:
> Meinst du mit dem zur Laufzeit generierten Dateien die zum Beispiel DataPoints_68C.py Datei, welche die DIDs von ECU 0x68C enthält? Damit ist auf jeden Fall abgesichert, dass man eine Liste von DIDs erhält, welche von jeder ECU unterstützt werden.

@Philip-Wiege Ja, die meine ich. Und damit funktioniert komplexe Adressierung und somit die Unterscheidung der Devices.

> Wie kann ich einen CAN Dump machen, um mal in den rohen CAN Verkehr reinzuschauen, was da wirklich zurückkommt?

Wenn Du alles sehen willst:
```
candump -c can0 -tA"
```
Wenn Du nur die Kommunikation mit Device-Adresse 0x680 sehen willst:
```
candump -c can0 -tA | grep "  6[89]0  "
```

MyHomeMyData, 2024-11-15:
> ja, geht. Aber ich wäre trotzdem dafür, -dev zu behalten (siehe auch hier https://github.com/open3e/open3e/issues/154#issuecomment-2478685113 ff.). Nur sollten wir einen deutlicheren Hinweis in's ReadMe machen.

@philippoo66 Verstehe ich Dich richtig: Wir brauchen -dev, um die Adresse für --scanall festzulegen? Das könnten wir leicht ändern und die Adresse mit --scanall mitgeben, also z.B. `--scanall 0x680`

Ich bleibe dabei: -dev sollte weg. Das macht nur Ärger.

philippoo66, 2024-11-15:
die sache ist dass das Laden des enstprechenden Modules schon eingebaut ist. Wir können (konnten bis zur pip Installation grrrr) damit jedes Modul Open3Edatapoints_irgendwas.py laden. 

Da ist nicht nur für scanall wichtig, sondern auch bei den Datenpunkten, die sich wie z.B. bei Vitodens von den Codec Definitionen in der generischen datapoints.py unterscheiden.

Aber wenn das Modul nicht gefunden wird, ist es egal, ob wir versuchen, es mit -dev zu laden oder mit scanall - es wird halt nicht mehr gefunden seit pip.

Wer hat nochmal die Umstellung gemacht und kann das wieder in Ordnung bringen?

philippoo66, 2024-11-15:
wenn es wieder geht können wir ja überlegen, ob wir -dev drin lassen. Ich wäre aber immer noch dafür, da man dann alle Möglichkeiten hat. 

Wie würdest du einen von der generischen datapoints abweichende DID Codierung spezifizieren, wenn du eine andere Spezifikation (ein Overlay) nur bei scanall laden kannst? 

Die Reihenfolge passt auch nicht. Das Overlay wird angewand wenn die ECU Instanz erzeugt wird. Ein scannall kann auch erst sehr viel später z.B. im Listener MMode kommen.

Das würde alles durcheinander bringen, wenn wir das mit dem Overlay jetzt plötzlich anders handhaben wollen....

philippoo66, 2024-11-15:
@SanMiggel  könntest du bitte schauen, warum 
- bei Angabe -dev vcal die Open3Edatapoints_Vcal.py gefunden wird
- aber bei Angabe -dev _680 die Open3Edatapoints_680.py nicht gefunden wird

obwohl beide Dateien direkt nebeneinander stehen und das früher funktioniert hat!?
Danke!!

philippoo66, 2024-11-15:
Jürgen @MyHomeMyData das mit dem Overlay ist in Open3Eclass init ab Zeile 55
```
        # load general datapoints table from open3e.Open3Edatapoints.py
        self.dataIdentifiers = dict(open3e.Open3Edatapoints.dataIdentifiers["dids"])            

        # overlay dids if certain device is selected ~~~~~~~~~~~~~~~~~~
        ....
```

philippoo66, 2024-11-15:
dass die Datei gefunden wird. 

ich glaube es ist schlichtweg ein Überbleibsel aus der Suchen&Ersetzen:

```
                if('.py' in dev):
                    didmoduledev = import_path(dev)
                else:
                    module_name = "open3e.Open3Edatapoints" + dev.capitalize()
                    didmoduledev = importlib.import_module(module_name)
```

Mit `-dev _680` soll die Datei Open3Edatapoints_680.py geladen werden. Die hat natürlich den Modulnamen "Open3Edatapoints_680" und nicht "open3e.Open3Edatapoints". Aber ich frage mich, warum es funktioniert, wenn man `-dev vcal` angibt. Ich blick das nicht mit dem "open3e." davor.

das oben mit `if('.py' in dev):` kommt wegen der devices.json, da steht
```
  "0x680": {
    "tx": "0x680",
    "dpList": "Open3Edatapoints_680.py",
    "prop": "HPMUMASTER"
  },
```

SanMiggel, 2024-11-16:
Hmmm. Der obere if-Zweig ist für Lokal erzeugte Data Point Dateien im Arbeitsverzeichnis. Der Else Zweig geht gegen die Dateien die in Open 3e Modul sind. Da liegt ja auch zum Beispiel
https://github.com/open3e/open3e/blob/master/src/open3e/Open3EdatapointsVcal.py

Da die files ja normal in komplett unterschiedlichen Verzeichnissen liegen musste ich die Bedingung einfügen.

Du willst jetzt eine lokale Datei "manuell “ importieren? Hast du schon probiert den kompletten Dateinamen anzugeben? Also
- dev Open3Edatapoints_680.py

Das sollte eigentlich gehen. Wenn nicht kann ich das morgen Mal selber probieren. 

Grüße
Michael

philippoo66, 2024-11-16:
@SanMiggel danke für's Nachschauen und erklären!

ich hab nix probiert, weil ich's ja garnicht am laufen hab weil ich ja keine E3 hab... Ich hatte es nur den Roland @bomilkar gesagt, weil es halt früher mal ging. 

Aber wenn es per `-dev Open3Edatapoints_680.py` tatsächlich geht, können wir es glaubich so lassen, auch wenn es ne menge Tipperei is mit viel Fehlerpotential bei der Eingabe auch noch mit gross/klein. Soo häufig wird das ja nicht benutzt. 

So richtig kapier ich das aber immer noch nicht: 

- Open3E_depictSystem.py liegt doch da, wo 'die Dateien die in Open 3e Modul sind'? 
- Open3E_depictSystem.py erzeugt die Dateien da, wo es sich befindet (also da, wo die Dateien die in Open 3e Modul sind)?!
- also liegen die datapoints_68x.py doch an der gleichen Stelle, wo auch z.B. die datapoints_Vcal.py liegt?!?

oder wo liegt mein Gedankenfehler??

ps. das _if('.py' in dev) ... else_ gab es schon immer!?

MyHomeMyData, 2024-11-17:
@philippoo66 @Philip-Wiege @SanMiggel Ich habe eine Version erstellt, die ohne -dev auskommt, siehe Branch `no_option_dev`. Es waren nur ein paar wenige Änderungen notwendig. Am besten ihr schaut euch das mal per Github-Compare (zu develop) an.

Alles was ich bisher ausprobiert habe, funktioniert: einfache Adressierung für 0x680, komplexe Adressierung ohne und mit -cnfg, depictSystem. Listener Mode habe ich noch nicht getestet.

Das Readme müsste man noch überarbeiten. Die komplexe Adressierung sollte ausführlicher erklärt werden und wir könnten eine Liste der bekannten Devices mit den zugehörigen Adressen ergänzen.
Die Hauptbotschaft sollte aber sein: Grundsätzlich einmal (und nach jedem SW-Update durch Viessmann) depictSystem ausführen, dass funktioniert's am besten.

Ich finde, das ist eine schlanke Lösung, die keine relevante Einschränkung verursacht und das Problem eines Datenabgleichs abschafft.

Oder habe ich was wichtiges übersehen?

philippoo66, 2024-11-17:
wie ist das wenn man die für die Vdens speziell codierte Datenpunkte mit eben dieser Codierung lesen will?

philippoo66, 2024-11-17:
ich sehe diese Möglichkeit nicht mehr. ich hatte ja die Overlay Geschichte nicht ohne Grund gemacht...

philippoo66, 2024-11-17:
> eine Liste der bekannten Devices mit den zugehörigen Adressen ergänzen

war es nicht so, dass da keine feste Zuordnung existiert, sondern das von der Anlagenzusammensetzung abhängig ist?

MyHomeMyData, 2024-11-17:
> wie ist das wenn man die für die Vdens oder Vair (z.B.) speziell codierte Datenpunkte mit eben dieser Codierung lesen will?

Das geht genauso wie bisher. Halt nur noch mit `-cnfg`. Aber es ist eh sinnvoll, ein depictSystem zu machen, wenn man spezifische Punkte lesen will, oder? Außerdem, wie viele sind das? Eine Hand voll?

> war es nicht so, dass da keine feste Zuordnung existiert, sondern das von der Anlagenzusammensetzung abhängig ist?

Stimmt. Für einige typische Konfigurationen ist es aber klar. Hauptgerät (vcal, vdens, ...) ist immer 0x680. VX3 ist - glaube ich - immer 0x6a1. Ansonsten gilt: Mach ein depictSystem und Du weißt Bescheid.

Philip-Wiege, 2024-11-17:
Ich versuche mal zusammenzufassen, welche Funktionen von Open3E auf die Overlay Files angewiesen sind oder diese erstellen:

**Depict System**
Scannt alle CAN IDs und alle DIDs die auf jeder Adresse funktionieren. Das Ergebnis wird in devices.json und in den Overlay Dateien für jede CAN ID gespeichert.

**-dev Argument**
Sorgt für die Benutzung einer speziellen Overlay Datei. Wichtig, falls ein DID anders als in der "Masterliste" codiert wird.

**-scanall Argument**
Gibt alle DIDs aus? Wie gibt man hier an, welche ECUs ausgelesen werden sollen? Werden hier die Overlay Dateien aus Depict System verwendet, damit man nur verfügbare DIDs versucht zu lesen?

**-cnfg Argument**
Gibt die Möglichkeit die devices.json einzulesen. Was wird damit dann intern gemacht?

**Komplexe Adressierung**
Man kann angeben, von welcher CAN-ID man einen DID lesen will. Manche DIDs gibt es zum Beispiel nur auf der VCMU und nicht auf der HPMU. Nutzt das Feature irgendwie Overlay Dateien im Hintergrund?

Ich glaube, wenn wir hier die Use-Cases alle mal aufgelistet haben, können wir an einem neuen, schlüssigen Konzept arbeiten.

Grüße
Philip

MyHomeMyData, 2024-11-17:
Guter Ansatz. Aus meiner Sicht stellt sich das so dar:

**Depict System**
Keine Änderung. Funktioniert genau gleich wie bisher. @philippoo66 **Oder habe ich das nicht richtig verstanden?**

**-dev Argument**
Wird ersetzt durch komplexe Adressierung. Device-spezifische Dids funktionieren nur noch mit `-cnfg`, also nachdem eine Depict System gelaufen ist.

**-scanall Argument**
Muss geändert werden. Funktioniert nur noch mit `-cnfg`, also nachdem Depict System gelaufen ist. Als Parameter wird die Adresse oder die Device-Bezeichnung angegeben, je nachdem, was in devices.json steht. Also z.B. `open3e -c can0 -cnfg devices.json -scanall 0x680`. Das ist im Moment im Branch `no_option_dev` nicht implementiert.

**-cnfg Argument**
Funktioniert genau gleich wie bisher.

**Komplexe Adressierung**
Muss immer benutzt werden, wenn auf einer anderen Adresse als 0x680 gearbeitet werden soll. Im Zusammenspiel mit `cnfg` funktioniert es genauso wie bisher.

Es kann gut sein, dass ich nicht alles korrekt verstanden habe und das Verhalten anders ist, als ober dargestellt. Dann bitte korrigieren und/oder ergänzen.

SanMiggel, 2024-11-17:
Hmm, bin nicht sicher, ob ich alle Kosequenzen verstehe. Aber die code Verschlankung von @MyHomeMyData sieht erstmal schön aus. Für 99% der user ändert sich doch dann vermutlich doch eh nix.

Ein case wundert mich noch: 
Z.B. in Open3EdatapointsVdens.py gibt es die Zeile:
`        401 : O3EInt16(5, "MixerOneCircuitPump", signed=True),`

So wie ich es verstanden habe, heißt dass, das dieser DP für dieses device anders kodiert ist. Wie würden wir denn so eine Information dann abbilden? In deinem Branch hast du das ja gelöscht.

philippoo66, 2024-11-17:
> Das geht genauso wie bisher. Halt nur noch mit -cnfg. Aber es ist eh sinnvoll, ein depictSystem zu machen, wenn man spezifische Punkte lesen will, oder?

depict schreibt immer RawCodec wenn die Länge anders ist. In der datapoints_Vxyz steht wolmöglich der richtige Codec drin (bei ein paar ist es so).

> Außerdem, wie viele sind das? Eine Hand voll?

genau wegen dieser Handvoll ist das System ursprünglich entstanden.

> Depict System
Keine Änderung. Funktioniert genau gleich wie bisher. @philippoo66 Oder habe ich das nicht richtig verstanden?

doch, ist richtig.

> -scanall Argument
Gibt alle DIDs aus? Wie gibt man hier an, welche ECUs ausgelesen werden sollen? 

bisher indem man `-dev _6yz` angibt (was aber so nicht mehr funktioniert). 

> Werden hier die Overlay Dateien aus Depict System verwendet, damit man nur verfügbare DIDs versucht zu lesen?

hier wird jede auf jedem Weg angegebene Overlay Datei benutzt. Deswegen hatte ich bomilkar ja das `-dev _6yz` nahegelegt, wenn er alle DIDs nur einer bestimmte ECU lesen will. 

> Komplexe Adressierung
Muss immer benutzt werden, wenn auf einer anderen Adresse als 0x680 gearbeitet werden soll. Im Zusammenspiel mit cnfg funktioniert es genauso wie bisher.

richtig. es funktioniert aber auch ohne -cnfg. Wenn die adressierte ECU nicht im dicEcus ist, wird sie angelegt mit der generischen datapoints Liste.

Achtung! die 680 ist die default ECU nur per default. sie kann aber auch geändert werden per `"-tx", "--ecuaddr"` (naja, konnte sie mal, irgendwer hat das aus der parser argslist rausgeworfen, aber der Code dazu existiert noch. Vielleicht war das sogar ich, als ich die komplexe Adressierung eingebaut hatte...)

alles in allem....

Ich verstehe nicht, warum wir auf Biegen und Brechen jetzt wieder einen Rückschritt machen wollen. Sooo wenige sind Vitodens und Vitoair und V... Nutzer ja nun auch nicht. Warum sollen wir ohne Not Möglichkeiten beschneiden, die ich eigentlich ziemlich perfekt ausgeklügelt habe? 

zugegeben - so viele 'ausformulierte' Codecs stehen in den datapoints_Vxyz auch nicht drin, aber zB bei den Stings wäre es ein Leichtes. Natürlich kann man die Codecs auch von Hand in die von depict erzeugten Dateien reinschreiben/reinkopieren. Aber dann sollten wir ne 'Blaupause' dafür machen, bevor die für immer weg sind.

MyHomeMyData, 2024-11-17:
> So wie ich es verstanden habe, heißt dass, das dieser DP für dieses device anders kodiert ist. Wie würden wir denn so eine Information dann abbilden? In deinem Branch hast du das ja gelöscht.

Ich habe es so verstanden, dass Depict System schaut, ob die Daten, die das Device liefert, zu den Werten aus der generischen Datenpunktliste passen. Also z.B. die Länge des Datenpunkts. Bzw. es gibt den Datenpunkt in der generischen Liste gar nicht. In solchen Fällen wird ein spezifische Datenpunkt in Open3Edatapoints_xyz.py angelegt.

Ich habe ja mit der Version aus dem neuen Branch Depict System laufen lassen (ohne die Dateien nach dem Muster Open3EdatapointsVdens.py) und da findet sich in der Datei Open3Edatapoints_680.py folgender Eintrag

```
        1553 : RawCodec(8, "ElectronicControlUnitOdxVersion"),
```

Da ich keine Vdens habe, weiß ich nicht, ob der von Dir gelistete Eintrag entstehen würden. Aber ich gehe davon aus.

@philippoo66 Woher kommt die Info `O3EInt16` für die Vdens? Kommt das aus Open3EdatapointsVdens.py? Das würde dann nicht funktionieren, sondern ein `RawCodec` erzeugen.

philippoo66, 2024-11-17:
> Woher kommt die Info O3EInt16 für die Vdens? Kommt das aus Open3EdatapointsVdens.py? Das würde dann nicht funktionieren, sondern ein RawCodec erzeugen.

ja, nur RawCodec. exakt das versuche ich ja die ganze Zeit zu erklären

MyHomeMyData, 2024-11-17:
> Ich verstehe nicht, warum wir auf Biegen und Brechen jetzt wieder einen Rückschritt machen wollen.

Meine Idee ist primär, das komplexe Gebilde mit Vxyz und _xyz zu vereinfachen. Wir bekommen halt vor allem zu `-dev` viele Fragen, eben weil es schwer zu durchschauen ist. Mit meinem Ansatz nötigen wir die Nutzer, Depict System zu machen. Ich finde das nicht verkehrt.

**Wäre es denn möglich, die Vxyz-Infos ins Depict System zu verlagern?** Gehört es da eigentlich nicht sogar hin? Dann könnten wir open3e vereinfachen und die Device-spezifischen Infos trotzdem behalten.

Ich möchte nichts übers Knie brechen. Die ursprüngliche Frage war ja, **wie synchronisieren wir die generische Liste und die Vxyz-Listen**. Ich habe mich darum jedenfalls bisher nicht gekümmert. Wir können es gerne so lassen wie es ist, aber wie lösen wir dann diese Frage?

philippoo66, 2024-11-17:
> Wäre es denn möglich, die Vxyz-Infos ins Depict System zu verlagern? 

du meinst, dass depict die datapoints_Vxyz benutzt und das gleich in die datapoints_6yz einbaut? das wäre eine elegante Lösung. braucht depict aber die Information, um was für ein device es sich handelt. Haben wir die schon identifiziert und enumeriert?

SanMiggel, 2024-11-17:
> Aber wenn es per `-dev Open3Edatapoints_680.py` tatsächlich geht, können wir es glaubich so lassen, auch wenn es ne menge Tipperei is mit viel Fehlerpotential bei der Eingabe auch noch mit gross/klein. Soo häufig wird das ja nicht benutzt.

Hab ich gestestet. Im aktuellen dev branch geht das so.

> So richtig kapier ich das aber immer noch nicht:
>
>     * Open3E_depictSystem.py liegt doch da, wo 'die Dateien die in Open 3e Modul sind'?
>
>     * Open3E_depictSystem.py erzeugt die Dateien da, wo es [...]

Um hier nochmal Licht ins Dunkel zu bringen:
Wenn ich `open3e_depictSystem` ausführe, werden die neuen .py datein in aktuellen Arbeitsverzeichnis erzeugt. Das hat nix damit zu tun wo das Skript liegt. Deshalb liegen normalerweise die open3e pip package files ganz wo anders, als die lokal erzeugten nach dem depict system.

MyHomeMyData, 2024-11-17:
> du meinst, dass depict die datapoints_Vxyz benutzt und das gleich in die datapoints_6yz einbaut?

Ja, genau das meine ich. Steckt die Info, die wir brauchen vielleicht im Datenpunkt 256?

MyHomeMyData, 2024-11-17:
> Wenn ich open3e_depictSystem ausführe, werden die neuen .py datein in aktuellen Arbeitsverzeichnis erzeugt. Das hat nix damit zu tun wo das Skript liegt. Deshalb liegen normalerweise die open3e pip package files ganz wo anders, als die lokal erzeugten nach dem depict system.

Ja, genau. Das muss so sein, weil dabei ein individuelles Abbild der Anlage des Anwenders entsteht. Das muss in dessen (selbst gewählten) Arbeitsverzeichnis stehen.

philippoo66, 2024-11-17:
wo wählt man denn das Arbeitsverzeichnis?

> Steckt die Info, die wir brauchen vielleicht im Datenpunkt 256?

möglich. in `O3EEnum(1, "DeviceProperty","Devices"), O3EEnum(1, "DeviceFunction","Devices")`?
```
    "Devices" : {
        0 : "NOTHING",
        1 : "BACKENDGATEWAY",
        2 : "HMUMASTER",
        3 : "HMUSLAVE",
        4 : "MCUMASTER",
        5 : "MCUSLAVE",
        6 : "HMI",
        7 : "BCU",
        8 : "GASAIRRATIOSYSTEM",
        9 : "ADIO",
        10 : "DIO",
        11 : "KNXGATEWAY",
        12 : "BACNETGATEWAY",
        13 : "MODBUSGATEWAY",
        14 : "MBUS",
        15 : "FRIWASTATION",
        16 : "FUELCELL",
        17 : "CSC",
        18 : "PHOTOVOLTAIC",
        19 : "PRODUCTIONGATEWAY",
        20 : "DIAGNOSTICGATEWAY",
        21 : "SDIO",
        22 : "MTWOIO",
        23 : "REMOTECONTROLLOCAL",
        24 : "CLIMASENSOR",
        25 : "TEMPERATURERADIATORVALVE",
        26 : "UNDERFLOORHEATINGVALVE",
        27 : "ENERGYMETER",
        28 : "EMCUMASTER",
        29 : "EMCUSLAVE",
        30 : "BMCU",
        31 : "HPMUMASTER",
        32 : "HPMUSLAVE",
        33 : "VCMU",
        34 : "EHCU",
        35 : "MZIO",
        36 : "PBE",
        37 : "HBMU",
        38 : "OSME",
        39 : "TWOSTEPPERMOTOR",
        40 : "INVERTER",
        41 : "SMARTROOMCONTROL",
        42 : "VCUMASTER",
        43 : "VCUSLAVE",
        44 : "CANOPENIOMODULE",
        45 : "FAN",
        46 : "ELECTRICALPREHEATER",
        47 : "ELECTRICALPOSTHEATER",
        48 : "GENERICHEATPUMP",
        49 : "HOMEENERGYMANAGEMENTSYSTEM",
        50 : "EEBUSHUB",
        51 : "HIO",
        52 : "BCUSLAVE",
        53 : "UNDERFLOORHEATINGBOX",
        54 : "ZIGBEERANGEEXTENDER",
        55 : "AIRQUALITYSENSOR",
        56 : "WATERSOFTENER",
        57 : "AIRPURIFIER",
        58 : "CONNECTIONADAPTERSTEPPERMOTOR",
        59 : "REFRIGERATIONFURNITURETYPE",
        60 : "SOLARLOGGATEWAY",
        61 : "SYSTEMCONTROLLEREMBEDDED",
        62 : "WALLBOX",
        63 : "APARTMENTTRANSFERSTATION",
        64 : "DHWTS",
        65 : "VENTILATIONUNIT",
    },
```

oder wohlmöglich in der VIN?

philippoo66, 2024-11-17:
wenn sie in den Devices steckt, bräuchten wir da eine Zuordung... (bei VIN auch)

MyHomeMyData, 2024-11-17:
> wo wählt man denn das Arbeitsverzeichnis?

Das Verzeichnis, aus dem man `open3e_depictSystem` startet.

MyHomeMyData, 2024-11-17:
> möglich. in O3EEnum(1, "DeviceProperty","Devices"), O3EEnum(1, "DeviceFunction","Devices")?

Können wir die Liste nicht direkt verwenden? Z.B.
*  31 : "HPMUMASTER" ==> Vitocal
* 28 : "EMCUMASTER" ==> VX3

philippoo66, 2024-11-17:
so einfach ist das nicht. Wenn z.B. die Vair Stand-alone ist, heisst sie anders, als wenn sie an einer Vcal hängt

philippoo66, 2024-11-17:
das war amals meine Intention, als ich drum gebeten hatte, dass mir möglicst alle Nuter ihre config.json mit einer kurzen Systemmbeschreibung schicken. dafür hatte ich extra die open3e at web eingerichtet. hat aber kaum jemand gemacht...

philippoo66, 2024-11-17:
dummmerweise hab ich in der json auch nur die DeviceProperty aufgegeben, nicht die DeviceFunction...

philippoo66, 2024-11-17:
dürfte für dich ja ein Leichtes sein, das bei dir mal grad nachzuschauen...

MyHomeMyData, 2024-11-18:
Für meine Geräte sieht das so aus:
Vitocal (0x680): "DeviceProperty": {"ID": 31,"Text": "HPMUMASTER"},"DeviceFunction": {"ID": 9,"Text": "ADIO"},
VX3 (0x6a1): "DeviceProperty": {"ID": 28,"Text": "EMCUMASTER"},"DeviceFunction": {"ID": 0,"Text": "NOTHING"},

Auch bei den anderen Geräten (HMI, EHCU, VCMU, BACKENDGATEWAYs) steht bei "DeviceFunction" immer {"ID": 0,"Text": "NOTHING"}.

Wie viele config.json hast Du denn?

Philip-Wiege, 2024-11-18:
Wo war denn der Aufruf, seine config.json zu teilen? Macht es Sinn, dafür einen Bereich anzulegen, wo hoffentlich einige Leute was hochladen können?

philippoo66, 2024-11-18:
```
Für 1. wäre es schön, wenn auf Dauer mal alle Leute entweder ihre _6yx.py's bei uns einschicken (z.B. an open3e/at/web.de) oder selber die entsprechenden Datenpunkte in der Open3Edatapoints.py im [open3e develop branch](https://github.com/open3e/open3e/tree/develop) ergänzen (per Fork -> einpflegen -> zum Mergen pushen).
```

in https://github.com/open3e/open3e/wiki/035-Open3Edatapoints-%E2%80%90-die-Zusammenh%C3%A4nge

philippoo66, 2024-11-18:
> Wie viele config.json hast Du denn?

da muss ich auch mal auf dem Rechner schauen, wo ich damals drauf gearbeitet habe. auf diesem hier ahbe ich eine Hand voll (muss auch genauer schauen, manche sind in zips und manche sind noch alt ohne 'prop'

philippoo66, 2024-11-18:
wenn wir --scanall mit einem Argument ausstatten, bitte darauf achten, dass auch weiterhin der Mechanismus funktioniert, dass wir - wie beim komplexen Adressieren - immer mit entweder der ECU Adresse oder den ECU Namen arbeiten können. Sollte man einfach copy&past übernehmen können, sollte aber auch gemacht werden.

Das umzusetzten ist dann ja völlig einfach (aufgrund meines 'genialen' Konzepts ;-)) ):
```
addr = getint(addr_of_dev(<arg beim -a arg>))
ecu = dicEcus[addr]
lst = ecu.readAll(args.raw)
```

philippoo66, 2024-11-18:
falls wir depictSystem anfassen, sollten wir es auf die damals von Pier-Eves extra dafür (read did without known length and codec) in udsoncan eingebaute (aber dann nie übernommene) Methode umbauen

https://github.com/pylessard/python-udsoncan/issues/188#issuecomment-1829617040

MyHomeMyData, 2024-11-18:
> Wo war denn der Aufruf, seine config.json zu teilen? Macht es Sinn, dafür einen Bereich anzulegen, wo hoffentlich einige Leute was hochladen können?

Finde ich eine gute Idee. Ein paar Gedanken dazu:
* Wir sollten vorher genau überlegen, welche Infos wir brauchen und wie die Nutzer diese am einfachsten kommunizieren können
* Per Mail wäre aus meiner Sicht besser, denn wir sollten Vertraulichkeit zusichern
* Sollen wir open3e_depictSystem um einen Hinweis ergänzen, dass die Ergebnisse bitte eingereicht werden?
* Wir könnten ein Gewinnspiel veranstalten und unter allen Einsendern z.B. einen Raspi Zero 2 verlosen. Würde ich spendieren.

philippoo66, 2024-11-18:
> Finde ich eine gute Idee. Ein paar Gedanken dazu:

> Wir sollten vorher genau überlegen, welche Infos wir brauchen und wie die Nutzer diese am einfachsten kommunizieren können
Per Mail wäre aus meiner Sicht besser, denn wir sollten Vertraulichkeit zusichern

per email haben wir dann gleich die Absenderadresse. ich kann das Account umstellen, dass die Daten da drin bleiben und euch den Zugang geben. Alle alten Mails schicke ich dann noch mal da hin, dass sie dann vorhanden sind.

> Sollen wir open3e_depictSystem um einen Hinweis ergänzen, dass die Ergebnisse bitte eingereicht werden?

keine schlechte Idee. das Wiki bis da unten hin liest ja kaum jemand... ;-)

> Wir könnten ein Gewinnspiel veranstalten und unter allen Einsendern 

und wann endet das Gewinnspiel und findet die Verlosung statt? ;-)

MyHomeMyData, 2024-11-18:
> und wann endet das Gewinnspiel und findet die Verlosung statt?

An Weihnachten ;-)

philippoo66, 2024-11-18:
aber 2025?!

philippoo66, 2024-11-18:
ich habe immer Verständnis dafür, wenn Leute ihre Datenscans nicht hochladen wollen. Wir haben open3e u.a. wegen Kritik an Viessmann's Datensammelwut entwickelt. Aber config.json und die datapoints_6xy.py enthalten ja 'keine Daten'. Von daher evtl ein zu erwägender Ansatz...

philippoo66, 2024-11-18:
*) Ich warne ja sogar davor, die Daten _öffentlich_ hochzuladen, weil da u.a. bsw. die Daten des Installateurs inkl. Kontaktdaten enthalten sind /sein können, und das ungenehmigte Veröffentlichen einen Verstoss gegen die DSGVO darstellt.

Es können möglicherweise ähnliche Daten enthalten sein, die wir noch garnicht identifiziert haben.

Wenn die Daten einmal auf github sind, können sie nicht (oder nur sehr schwer?) wieder gelöscht werden.

philippoo66, 2024-11-18:
ich hatte auch gerade einen Ordner depict_files angelegt und angefangen, ihn zu füllen, habe dann aber wegen einem unangenehmen Gefühl im Bauch bei den Daten aufgehört, die Dustin eh schon hochgeladen hat. Die Sache mit den Mails ist einfach die bessere Variante finde ich

Philip-Wiege, 2024-11-18:
Ich habe mal meine Datein per Mail geschickt. Ich seh in dort nix privates. Vielleicht sollten wir mal systematisch in einem Wiki Artikel die CAN IDs zu den verschiedenen Konfigurationen sammeln.

philippoo66, 2024-11-18:
ne, du hast nicht deine Daten geschickt, das ist nur die Konfiguration. Deine Daten werden geschrieben, wenn du bei depict die Option `-s | --simul` setzt ;-) die werden dann in virtdata_6xy.txt gespeichert. 

'virtdata' weil das die Files für die virtualE3 sind. damit kann man dann die Anlage zum Zeitpunkt des Snapshot simulieren und open3e damit testen als ob man live auf der Anage wäre.

Das hatte ich damals geschrieben, damit ich meine Programmiererei zumindest ansatzweise testen konnte, weil ich ja kein reales Zielsystem habe. Die kann sogar Daten 'dynamisieren' auf Basis von ziemlich schwer zu konfigurierenden Vorgaben. \sonnenbrillensmiley\

> Vielleicht sollten wir mal systematisch in einem Wiki Artikel die CAN IDs zu den verschiedenen Konfigurationen sammeln

Philip, Philip... Ich sag es nochmal gaaaanz deutlich: **Es gibt nicht DIE Datenpunktliste. Die ist abhängig von Gerätefamilie, Modell und Softwareversion. DESWEGEN habe ich das Overlay System und depictSystem entwickelt.**

philippoo66, 2024-11-18:
ich hab jetzt zumindest mal die devices.json's von diesem Rechner nach [depict_files](https://github.com/philippoo66/open3Edevelop/tree/main/depict_files) hochgeladen.

philippoo66, 2024-11-18:
@flecke-m  was ist denn CSC?

flecke-m, 2024-11-19:
> @flecke-m was ist denn CSC?

Ich musste erst noch einmal nachschauen. Da es 4 Einträge sind und ich mir die Datapoints dazu angeschaut habe, kann ich mit 99,9% Sicherheit sagen, dass es die Batteriemodule sind. Ich habe 10kwh Speicher: zwei Viessmann 5kwh Speichereinheiten a 2 Batteriemodule pro Speichereinheit.

philippoo66, 2024-11-19:
> und ich mir die Datapoints dazu angeschaut habe,

auf die Idee hätte ich auch kommen können... Sieht wirklich stark nach irgendwas mit Batterie aus ;-)

ich hab die DIDs mal in die general Liste übernommen.

philippoo66, 2024-11-19:
@flecke-m  hast du zu `2335 : RawCodec(8, "BatteryModuleCoulombCounters"),` schon ne Idee zu der Codierung? "Counters" klingt ja nach mindestens zwei. Skalierung?

[1 more reply]