open3e/open3e #154: open3e --scanall fails with NegativeResponseException

bomilkar, 2024-11-15:
After a firmware update on my Vitocal 250A it fails to generate a list of all DIDs:

```
$ open3e -c can0 -dev vcal --scanall
reading 0x680, 539 datapoints, please be patient...
Traceback (most recent call last):
  File "/home/pi/.local/bin/open3e", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 406, in main
    lst = ecu.readAll(args.raw)
          ^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/open3e/Open3Eclass.py", line 170, in readAll
    value,idstr = self.readByDid(int(did), raw=raw)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/open3e/Open3Eclass.py", line 154, in readByDid
    response = self.uds_client.read_data_by_identifier([did])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/udsoncan/client.py", line 167, in decorated
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/udsoncan/client.py", line 466, in read_data_by_identifier
    response = self.send_request(req)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/udsoncan/client.py", line 2258, in send_request
    raise NegativeResponseException(response)
udsoncan.exceptions.NegativeResponseException: ReadDataByIdentifier service execution returned a negative response ConditionsNotCorrect (0x22)

```

I ran open3e_depictSystem and all "normal" reads work like they did before. For instance:

```
$ open3e -c can0 -dev vcal -r 424,426 -v
vcal 424 MixerOneCircuitRoomTemperatureSetpoint {"Comfort": 23.0, "Standard": 21.0, "Reduced": 3.0, "Unknown2": "0000", "Unknown1": 0}
vcal 426 MixerTwoCircuitRoomTemperatureSetpoint {"Comfort": 23.0, "Standard": 21.0, "Reduced": 14.0, "Unknown2": "0000", "Unknown1": 0}
closing 0x680 - bye!
```

philippoo66, 2024-11-15:
If you interested in the main ecu only, you might use `-dev 680`. the datapoints_680 gets updated by depict_system.

bomilkar, 2024-11-15:
Thanks!
`open3e -c can0 -cnfg devices.json --scanall`
Does the job.
But both
```
open3e -c can0 -dev 680 --scanall
open3e -c can0 -dev 0x680 --scanall
```
crash with 
```
ModuleNotFoundError: No module named 'open3e.Open3Edatapoints680'
ModuleNotFoundError: No module named 'open3e.Open3Edatapoints0x680'
```
respectively.

philippoo66, 2024-11-15:
oh, sorry. so you need to do `open3e -c can0 -dev _680 --scanall` (inclluding the underscore)

philippoo66, 2024-11-15:
ps. more details regarding the -dev issue see here: https://github.com/open3e/open3e/issues/153

philippoo66, 2024-11-15:
fu.., now the .py is missing. I'm in a hurry right now, I will come back later. 

It is a problem related to the pip installatin. in earlier times it worked..

philippoo66, 2024-11-16:
sorry, I can't explain since I do not understand, but perhaps you like to follow the dickussion of it here: https://github.com/open3e/open3e/issues/153#issuecomment-2480809317

philippoo66, 2024-12-02:
hi @bomilkar Ronald!

just in terms of cleaning up... `-dev Open3Edatapoints_680.py` should wotk nowadays - does it for you? if so, I'd like to close the issue!?

bomilkar, 2024-12-02:
I think the workaround was to use `open3e  --scanall`

closed 2024-12-02 by bomilkar