open3e/open3e #160: different Codecs for different Data Lengths?!

philippoo66, 2024-12-02:
we might implement a way to use a codec depending on the data length of a datapoint!?

let me explain:

[here](https://github.com/open3e/open3e/discussions/138#discussioncomment-11430087) @Backes89 is working on the coding of 2068 (and similar) which is `2086 : RawCodec(68, "ZigBeeOneDeviceCurrentValues"),` with his equipment. 

In the datapoints.py it is `2086 : RawCodec(57, "ZigBeeOneDeviceCurrentValues"),` I think based on @DpunktS 's equipment.

The similar 'issue' appears with datapoints of different length with e.g. Vitocal and Vitodens.

SO I like to find and implement **a way to make it possible to define Codecs according to DID and DID length**.

Any idea?!

MyHomeMyData, 2024-12-02:
I thought, this is already implemented via device specific data point files, i.e. _xyz.py. What is the new requirement here?

philippoo66, 2024-12-02:
the problem is that depictSystem only puts in O3ERawCodec and would/does not know that there is/might be a codec for the length differring for the length in datapoints.py

philippoo66, 2024-12-02:
example: Backes will find the codec for 68 Bytes. He can (manually) put it in _his_ datapoints_6c3.py but since it does not fit for the 57 byte 2068, it will be 'lost'...

ok, now we could change the generic datapints.py to 68 bytes an the found codec, but what happens if Dustin will find the codec for his 57 bytes?!

Philip-Wiege, 2024-12-02:
Are both ZigbeeDeviceOne DIDs on the same ECU? Even Viessmann needs a way to switch between DID interpretations based on SW version of the ECU or something similar.

Usually there is an ODX file which describes the DID content. Each ECU has a matching ODX for its current SW version. There is one DID which tells the diagnostics client which ODX file to use.

@philippoo66 Since you seem to have the DIDs value snapshots from both systems: Are they the same ECU and what kind of ODX is defined in the DID 1553?

philippoo66, 2024-12-02:
Backes:
ECU 6c3:
2086 : RawCodec(68, "ZigBeeOneDeviceCurrentValues"),
no DID 1553
ECU 680.1553 210045010a095700

Dustin:
ECU 6c3:
2086 : None, 
so according to generic datapoints.py:
2086 : RawCodec(57, "ZigBeeOneDeviceCurrentValues"),
no DID 1553
ECU 680.1553 2100a901be080500

So where/how do we get these odx files?!?

philippoo66, 2024-12-02:
really interesting! I wasn't aware of that. 

- So who will search the 'sources' that we have / can get hold of for this?!
- Which are these sources? The Viguide apk we have. What else? I remember something regading a web app - was ist Viguide also? I think @DpunktS mentioned that this could get downloaded somehow?
- anything else?

Philip-Wiege, 2024-12-02:
> Backes:
>
> ECU 6c3:
>
> 2086 : RawCodec(68, "ZigBeeOneDeviceCurrentValues"),
>
> no DID 1553
>
> ECU 680.1553 210045010a095700
>
>
>
> Dustin:
>
> ECU 6c3:
>
> 2086 : None,
>
> so according to generic datapoints.py:
>
> 2086 [...]

Do you have the DID values of 1553 for ECU 0x6C3?

philippoo66, 2024-12-02:
as I metioned "no DID 1553" with the 6c3 data sets.

MyHomeMyData, 2024-12-02:
> the problem is that depictSystem only puts in O3ERawCodec and would/does not know that there is/might be a codec for the length differring for the length in datapoints.py

@philippoo66 In ioBroker adapter I put a data point with a length different from common list to a device specific list using RawCodec.

philippoo66, 2024-12-02:
ja das macht depict ja genauso. aber wenn wir den Codec für 68 Bytes kennen und den für (für den gleichen DID) dann ist aktuell weder depict noch der ioBroker Adapter in der Lage, den zur Datenlänge passenden Codec in die device specific datapoints Liste zu schreiben. und _das_ möchte ich ändern.

philippoo66, 2024-12-02:
das ginge bsw. mit einer Liste, wo die von der genischen datapoints.py abweichenden Codecs in Abhängigkeit von der Datenlänge für den betreffendennn DID stehen. Aber das will eben sauber durchdacht sein, damit es auch problemlos und zukunftsoffen funktioniert.

Aber vielleicht erübrigt sich das ja auch, wenn wir 'hinter' bzw an die odx files kommen...

MyHomeMyData, 2024-12-02:
> Aber vielleicht erübrigt sich das ja auch, wenn wir 'hinter' bzw an die odx files kommen...

Ja, ich glaube wir sind hier auf weitere Infos angewiesen. Es muss gelingen, ein Device aus den gelesenen Datenpunkten zu identifizieren, also z.B. "das ist eine Vitodens xyz. Dann brauchen wir noch die dafür passende Liste mit abweichenden Datenpunkten. Wäre super, wenn sich das aus den odx files extrahieren ließe.

philippoo66, 2024-12-02:
stell dir vor, Backes bekommt jetzt die 68 bytes entschlüsselt - was machen wir dann damit? Wir können das ja nicht einfach in die generische datapoints liste schreiben, weil da steht 57 bytes. Wenn wir das jetzt plötzlich zu 68 ändern, stürzt bei allen Leuten, wo in der _6c3 'None' steht, die Geschichte ab.

MyHomeMyData, 2024-12-02:
Stimmt, aber wir können den Raw-Code anlegen und im Readme eine Liste ergänzen, die bekannte gerätespezifische Codecs enthält. Dann kann jeder, der das braucht, das in seine _xyz.py Datei reinkopieren.

Als Zwischenlösung aus meiner Sicht praktikabel.

philippoo66, 2024-12-02:
als Zwischenlösung ja. aber ich hab ja immer gern eine Lösung mit Hand und Fuss ;-) Wenn wir eh eine Liste anlegen - warum soll depict die dann nicht verarbeiten statt der Anwender händich?

philippoo66, 2024-12-02:
eigentlich nicht so schwer: 

- depict stößt auf eine 'inkompatible Datenlänge
- es gibt eine Liste mit Datenlängen
- bei jedem Eintrag steht eine Liste mit DID, auf die der Codec, der dahinter steht, anzuwenden wäre.

bei einem Match nimmmt depict den Codec. fertig.

MyHomeMyData, 2024-12-02:
Nicht ganz. Denn es könnte für verschiedene Geräte unterschiedliche Codecs mit gleicher Länge geben. Und Du kennst den Gerätetyp nicht, nur die Adresse.

philippoo66, 2024-12-02:
hatten wir sowas schon?

philippoo66, 2024-12-02:
ok, wir brauchen die odx Files... ;--)

MyHomeMyData, 2024-12-02:
> ok, wir brauchen die odx Files... ;--)

So isses.

philippoo66, 2024-12-02:
ist es nicht aktuell sogar so, dass mache DIDs mit gleicher Länge auftauchen, aber der Codec trotzdem für unterschiedliche Geräte unterschiedlich ist? ich meine, da dunkel was zu erinnern mit irgendwelchem min max avg und act?!

MyHomeMyData, 2024-12-02:
Ja, auch das gibt es. Wir müssen das Gerät identifizieren können, sonst kommen wir nicht weiter. Oder der Anwender muss es händisch machen.

philippoo66, 2024-12-03:
> Es muss gelingen, ein Device aus den gelesenen Datenpunkten zu identifizieren, also z.B. "das ist eine Vitodens xyz". Dann brauchen wir noch die dafür passende Liste mit abweichenden Datenpunkten. Wäre super, wenn sich das aus den odx files extrahieren ließe.

So wie ich das verstanden habe, ist in den ODX Informationen die Codecs parktisch enthalten. Was für ein Gerät es ist, ist dann uninteressant - die ODX Infos sagen, war die Bytes des DID bedeuten? 

Ich habe versucht, die Viguide Web App 'runterzuaden', war aber wenig erfolgreich. Wenn man die Seite einfach als 'Webseite, komplett' speichert, läuft sie nicht mehr. Mit F12 bin ich auch nich weiter gekommen, aber damit kenne ich mich auch zu wenig aus. Vielleicht macht es Sinn, da mal weiter zu gucken.

Obwohl UDS immer wieder im Zusammengang mit ODX auftaucht, oder vielleicht besser andersrum, hab ich in udsoncan nix mit ODX gefunden, obwohl es ja wohl der Schüssel zur De/Codierung ist. Das verwundert nich ein wenig. @Philip-Wiege wie funktioniert das? 'Steckt man nicht das ODX Zeugs in das UDS', damit die Daten einen Sinn ergeben?

MyHomeMyData, 2024-12-03:
Es sind doch 2 Puzzleteile:
* Der Codec macht die Zuordnung von Bytes aus dem UDS-Datenstrom zur korrekten Information, z.B. 2 Bytes als signed Integer Sensorwert.
* Aus bestimmten Datenpunkten, z.B. 256_BusIdentification und 954_BusTopologyMatrix (und/oder ???) lässt sich ableiten, mit welchem Gerätetyp man spricht. Und dieses Puzzleteil fehlt und noch komplett.

Philip-Wiege, 2024-12-03:
Die UDS Daten zu interpretieren zum Beispiel mit einem ODX File ist eine Ebene über UDS in der Anwendung. UDS bekommt davon quasi nix mit.

Meiner Meinung nach ist eine ECU eindeutig über die Identfikation im DID 256 und der SW Version aus DID 580 beschrieben. Das wäre aber ein Kombinationsmonster, wenn wir füe jede ECU+SW Version eine Datapoints Datei hätten.

philippoo66, 2024-12-03:
sind unsere Codecs / die Codecs 'in' udsoncan nicht eigentlich auch 1 Ebene über UDS? USD sind doch nur Bytes, nicht aber die Interpretation der Bytes? Oder hab ich da grad was falsch aufm Schirm?

Philip-Wiege, 2024-12-03:
Genau, die Codecs sind auch schon eine Ebene über der reinen UDS Kommunikation.

philippoo66, 2024-12-09:
ich hab jetzt zumin mal ne Seite in's Wiki gemacht, wo wie 'divergete Codecs' sammeln können.

https://github.com/open3e/open3e/wiki/070-%E2%80%90-List-of-divergent-codecs

philippoo66, 2025-01-25:
Wenn die ODX Information in DID 1553 eine Zuordnung erlauben, würde ich gerne in datapoints.py ein weiteres dict anlegen, das 'Überschreibungen' für Codec Definitionen enthält, und eine 'Tabelle', in der dann steht, welche Überschreibungen bei welchem 1553_Wert anzuwenden sind.

Was meint ihr?! @Philip-Wiege @MyHomeMyData @ alle

Philip-Wiege, 2025-01-25:
In welchen Fällen haben wir denn schonmal verschiedene Codierung des selben DIDs gesehen?

War das dann immer die selbe CAN Adresse aber verschiedene Geräte? Die Adresse 0x680 kann ja z.B. ein HPMU oder ein HPMUMASTER sein.

philippoo66, 2025-01-25:
> In welchen Fällen haben wir denn schonmal verschiedene Codierung des selben DIDs gesehen?

in diversen. 

Die 680h kann alles möglich sein. In Prinzip jedes Gerät, solange es nicht an irgendwas angebunden ist. Aber bezüglich 680 interessiert doch nur die 1553 oder hab ich da was falsch verstanden?!