open3e/open3e #163: open3e.service doesn't start

ubik128, 2024-12-06:
Hallo!

Ich bin leider ein newbie bei dem Thema Raspberry Pi, aber durch eure Anleitung bin ich schon weit gekommen. Also durch das open3e Projekt kann ich meine Wärmepumpe auslesen und mittels MQTT auf dem loxberry bekomme ich es auch auf die loxone drauf.

Ein Problem habe ich aber noch und da vermute ich, dass ich bei den basics scheitere.
Wenn ich mein arg File in der virtuellen Umgebung starte per Hand, dann passt alles und ich bekomme Werte (solange das terminal läuft).
(.venv) root@loxberry:~/open3e# open3e @args

Damit mein Service lauffähig wurde musste ich den Benutzer und den Pfad anpassen auf.
User=loxberry
ExecStart=/bin/bash -c 'cd /opt/loxberry/open3e && /opt/loxberry/open3e/.venv/bin/open3e @args

Ich weiß, dass es zwei args Files sind (einmal in der virtuellen Umgebung) und einmal außerhalb. Ich habe auch beide gleich gemacht.
Ich bin mir ziemlich sicher, dass es etwas mit dem Pfad zu tun hat, dass er es nicht startet.

Danke für eure Hilfe
Christian

[image]

philippoo66, 2024-12-06:
moin Christan!

Wenn das oben beim ExecStart eine exakte Kopie der Zeile in dem Dienst-Skript ist, dann fehlt da am ende ein Hochkomma (einfaches Anführungszeichen). Das reciht für's nicht-Funktionieren-wie-gewollt ;-)

ubik128, 2024-12-06:
Guter Vorschlag aber leider war es nur ein Typo oben..... deswegen mache ich gerne screenshots.
also in dem Skript file sind beide ' drinnen.

Schade.... wäre auch zu einfach gewesen.

[image]

MyHomeMyData, 2024-12-06:
> Ich weiß, dass es zwei args Files sind (einmal in der virtuellen Umgebung) und einmal außerhalb.

Nach meinem Verständnis gibt es nur ein args-File, in Deinem Fall unter `/opt/loxberry/open3e`.

Eine mögliche Ursache könnten Zugriffsrechte sein, da die Files unter /opt liegen. Ist aber nur eine Vermutung. Die Installation sollte besser im User-Verzeichnis des Users loxberry liegen.

Wie ist denn das Verhalten? Startet der Service? Was liefert `sudo systemctl status open3e.service`?

ubik128, 2024-12-06:
in dem Verzeichnis /opt/loxberry/open3e kann ich aber leider das args File nicht starten, weil ich nicht in der virtuellen Umgebung bin...oder?

Anbei der Status vom Service.
Habe noch einmal reload gemacht und einen manuellen Start des Service.... siehe screenshot.
Dann habe ich kontrolliert, ob neue Werte im MQTT Broker angekommen sind --> leider nein :(

[image]

philippoo66, 2024-12-06:
heisst dein File jetzt Args oder args? Groß/Kleinschreibung ist wichtig!

es gibt noch einen anderen 'Ansatz' mit dem starten in der virtuellen Umgebung, vielleicht probierst du den mal:

```
[Unit]
Description=Optolink-Splitter Service
After=network-online.target

[Service]
Type=simple
ExecStartPre=/bin/sleep 10
User=pi
Group=pi
WorkingDirectory=/home/pi/optolink-splitter/
Environment="PATH=/home/pi/optolink-splitter/myvenv/bin"
ExecStart=/home/pi/optolink-splitter/myvenv/bin/python /home/pi/optolink-splitter/optolinkvs2_switch.py
Restart=always
RestartSec=3

[Install]
WantedBy=multi-user.target
```

musst du natürlich alles anpassen. mit args file dahinter musst du evtl das für das ExecStart in Anführungszeichen setzen, einfache oder doppelte weiss ich nich, alles ausprobieren (ohne/miteinfach/doppelt)...

ubik128, 2024-12-06:
Nach dem Start-Befehl kommt nichts. Siehe screenshot 1.
An dem args file kann es fast nicht liegen, weil das funktioniert wenn ich es mit (.venv) root@loxberry:~/open3e# open3e@args starte.

Ich glaube, es hat etwas mit den Verzeichnissen bei der Installation zu tun. Das mit dem ~ Verzeichnis habe ich nicht ganz verstanden :( (Newbie)

-->
manueller start vom args file mit:
(.venv) root@loxberry:~/open3e# open3e@args --> funktioniert

manueller start mit 
(.venv) root@loxberry:/opt/loxberry/open3e# open3e@args (analog wie im service file)--> funktioniert nicht... dort ist auch nicht das open3e file drinnen.

Danke Christian

[image]

[image]

[image]

philippoo66, 2024-12-06:
oj, jetzt solltest du vlt dein Passwort ändern ;-)

ubik128, 2024-12-06:
@philippoo66 : File heißt überall args ... Oben beim Abschreiben, dürfte es mir geändert haben.
--> ich vermute, dass ich irgendwo mit den Verzeichnissen ~/open3e und /opt/loxberry/open3e einen Hund drinnen habe.

Danke... vor lauter screenshoten, habe ich das passwort übersehen :)

philippoo66, 2024-12-06:
> Nach dem Start-Befehl kommt nichts. Siehe screenshot 1.

das ist ja normal bei einem Dienst. der sagt ja nur mit status <dienstname> was.

versuch auch mal `sudo journalctl -u <dienstname> --since "2024-12-06 00:00:00"` vlt sagt und das noch was

philippoo66, 2024-12-06:
aber dass er das devices json nicht findet ist ein klares Zeichen dass irgendwas mit dem Arbeitsverzeichnis nicht stimmt. versuch mal meinen Dienst-Skript Ansatz oben. Da ist das meiner Meinung nach klarer mit den Pfaden un der Umgebung.

philippoo66, 2024-12-06:
wo steht denn das devices.json? das sollte eigentlich in /home/loxberry/open3e stehen. 

was passiert (dann) bei `/bin/bash -c 'cd /home/loxberry/open3e && /home/loxberry/open3e/.venv/bin/open3e @args'`

ubik128, 2024-12-06:
--> soll heißen.... back to the roots und noch einmal installieren und gleich auf das richtige Arbeitsverzeichnis aufpassen?
Neuinstallation sollte kein Problem... jetzt bin ich eh schon fix..... soll ich das alte open3e im falschen Verzeichnis irgendwie löschen? Kann man es einfach löschen oder braucht man hier eine Routine?

unter home/loxberry/open3e gibt es nicht das devices.json file.

ich glaube, ich habe alles unter ~/open3e installiert, weil dort alles drinnen ist.
Für was steht ~ ???
ich habe im service file auch einmal den ~ Pfad probiert ohne Erfolg.

[image]

philippoo66, 2024-12-06:
> back to the roots und noch einmal installieren und gleich auf das richtige Arbeitsverzeichnis aufpassen?

das wäre gut! und dann genau nach Anleitung vorgehen.

> soll ich das alte open3e im falschen Verzeichnis irgendwie löschen?

ja, einfach löschen. und das Deinst-Skript auch ;-)

> Für was steht ~ ???

wo steht das?

philippoo66, 2024-12-06:
> und dann genau nach Anleitung vorgehen.

die Anleitung im Readme

philippoo66, 2024-12-06:
> wo steht das?

achso, das steht für dein home Verzeichnis, also /home/loxberry/

philippoo66, 2024-12-06:
MOMENT!

> unter home/loxberry/open3e gibt es nicht das devices.json file.

das ist wahrscheinlich der Knackpunkt. evtl reicht es, das config.json und die ganzen datapoints_6xy.py da hin zu verschieben

ubik128, 2024-12-06:
so....habe jetzt alles noch einmal neu installiert mit dem Verzeichnis /opt/loxberry/open3e .... das Verzeichnis ist jetzt auch überall drinnen und alle files sind dort.

Schlechte Nachricht.... der Dienst geht immer noch nicht :(

Im Anhang der Status + journal.
laut journal findet er das verzeichnis noch nicht :(

hat es etwas mit /.venv/bin/ zu tun ???

[image]

[image]

MyHomeMyData, 2024-12-06:
Warum arbeitest Du als root? Wenn Du das für den User loxberry einrichten willst, solltest Du unbedingt als **User loxberry arbeiten**.

Dann alles unter dem home-Verzeichnis von loxberry installieren. Vermutlich heißt das /home/loxberry. Da solltest Du ein Unterverzeichnis open3e/ anlegen, also /home/loxberry/open3e und der Anleitung im Wiki folgen.

ubik128, 2024-12-06:
mhm .... ok das probiere ich noch einmal aus.
Ich dachte mir, ich habe dann "nur" Rootrechte und der Benutzer bleibt gleich. Aber anscheinend ist das dann auch ein Rootuser :)

Werde ich einmal probieren.

Melde mich dann noch einmal

ubik128, 2024-12-06:
nein leider. neuer user nicht mehr im root und hat trotzdem nicht funktioniert.
kann man den ExecStart befehl auch ohne script starten bzw testen?
--> ich glaube dort liegt noch das problem.

ExecStart=/bin/bash -c 'cd /opt/loxberry/open3e && /opt/loxberry/open3e/.venv/bin/open3e @args'

Danke Christian

MyHomeMyData, 2024-12-06:
Hast Du es jetzt wieder unter /opt installiert??? Bitte, bitte, installiere open3e unter dem Home-Verzeichnis des Users loxberry.

ubik128, 2024-12-06:
Hallo!
Habe es endlich geschafft.
Habe dann alles in ein Home Verzeichnis gegeben. Einziges Thema was ich noch gehabt habe. Der User loxberry hat keine Rechte, jetzt musste ich wieder alles mit root installieren.
aber zum Starten funktioniert es ohne root Rechte .... ich bekomme zwar eine Warnung, dass er nicht aufs log File schreiben kann, aber das ist mir egal... hauptsache ich bekomme die Werte :) 
und auch nach einem Neustart hat alles funktioniert.

next step --> Werte schreiben.

lg Christian

closed 2024-12-06 by ubik128

MyHomeMyData, 2024-12-06:
Freut mich, dass es nun funktioniert. Viel Erfolg mit der Nutzung von open3e!