open3e/open3e #164: write did with MQTT messages

ubik128, 2024-12-06:
Hallo!

Gibt es schon ein Beispiel, wie man eine MQTT Nachricht verwenden kann um einen did Wert zu schreiben?
Ich habe leider nichts in die Richtung gefunden, aber funktionieren müsste es ja, weil der ioBroker ja auch nur MQQT verarbeitet oder?

Danke Christian

MyHomeMyData, 2024-12-06:
Ja, siehe Abschnitt [Listener Mode](https://github.com/open3e/open3e?tab=readme-ov-file#listener-mode) im Readme. Z.B.:
```
to write value of 21.5 to did 395 and value of 45.0 to did 396:
{"mode": "write", "data":[[395,21.5],[396,45.0]]}
```

Verwendest Du ioBroker?

ubik128, 2024-12-06:
habe ich mir überlegt, weil es eine "gute" Vorlage gibt für die Visualisierung.
Aber nachdem ich alles mit meiner loxone steuern möchte, mache ich alles mit dem loxberry und dem MQTT plugin.
Beim schreiben, werde ich mich wohl step by step vorarbeiten :(

MyHomeMyData, 2024-12-07:
Mit dem Docker-Plugin könntest Du ioBroker auf dem Loxberry installieren und die Kommunikation zu den Viessmann-Geräten mit dem [e3oncan-Adapter](https://github.com/MyHomeMyData/ioBroker.e3oncan) statt open3e machen.
* Der Adapter **ersetzt open3e komplett** und unterstützt weitere Geräte und Funktionen
* Die Konfiguration ist mit wenigen Mausklicks erledigt
* Alle Datenpunkte werden nach der Konfiguration des Adapters als States im Objektbaum angezeigt
* Das Schreiben von Datenpunkten geht ganz einfach durch Ändern des Werts des entsprechenden States. Das kann natürlich auch per MQTT oder per Skript erfolgen.

ubik128, 2024-12-07:
Mhm OK....
Ich habe mir das mit iobroker nicht angesehen, weil ich nicht noch ein Programm wollte und das Beispielprojekt hier ist "quasi" nur zur Visualisierung im iobroker.
Aber wenn du sagst e3oncan mit iobroker wäre zum empfehlen, dann würde ich es mir einmal ansehen.

Danke für deinen Input

MyHomeMyData, 2024-12-08:
@ubik128 Dann können wir den Issue schließen, oder?

Generell hab ich die Bitte, Fragen oder Diskussionsbeiträge in der Rubrik "Discussions" zu platzieren. Issues sollten für Problem-Meldungen und Vorschläge, z.B. für neue Features verwendet werden.

ubik128, 2024-12-08:
Ok, passt. Ich setze es auf close und wenn ich Hilfe brauche werde ich in die Diskussion schreiben.

Danke Christian

closed 2024-12-08 by ubik128