open3e/open3e #165: Temperatures scale 10?

philippoo66, 2024-12-09:
is this really true or should it be scale = 0.1?
```
        1627 : O3EInt16(2, "ExternalMixerOneCircuitFixedValueTargetTemperatureSetpoint", scale = 10, signed=True),
        1628 : O3EInt16(2, "ExternalMixerTwoCircuitFixedValueTargetTemperatureSetpoint", scale = 10, signed=True),
        1629 : O3EInt16(2, "ExternalMixerThreeCircuitFixedValueTargetTemperatureSetpoint", scale = 10, signed=True),
        1630 : O3EInt16(2, "ExternalMixerFourCircuitFixedValueTargetTemperatureSetpoint", scale = 10, signed=True),
        1643 : O3EInt16(2, "MixerOneCircuitCurrentTemperatureSetpoint", scale = 10, signed=True),
        1644 : O3EInt16(2, "MixerTwoCircuitCurrentTemperatureSetpoint", scale = 10, signed=True),
        1645 : O3EInt16(2, "MixerThreeCircuitCurrentTemperatureSetpoint", scale = 10, signed=True),
        1646 : O3EInt16(2, "MixerFourCircuitCurrentTemperatureSetpoint", scale = 10, signed=True),
        1659 : O3EInt16(3, "EndResultDomesticHotWaterTemperatureSetpoint", scale = 10, signed=True),
```

MyHomeMyData, 2024-12-09:
scale = 10 is often used for temperatures and it's the default value for O3EInt. Leads to a resolution of 0.1 °C. Why should it be different here?

philippoo66, 2024-12-09:
oh, dann hab ich was durcheinandergebracht...

closed 2024-12-09 by philippoo66