open3e/open3e #166: Vitocal 250-a can communication with ESPHome

Scarbous, 2024-12-14:
Hi,
this is not directly a question about the project, but it is about the Viessmann can communication.

I would like to use ESPHome with a CAN module to read the data from my Vitocal 250-a.

I have already created a config with which I can send all messages via MQTT, but only the following IDs arrive 0,1793,511,400,2021

Now I am unsure whether something additional needs to be activated or whether I need to actively query data via can.

I have not found anything about this in the installation instructions for open3e.

-----

Here is my current canbus ESPHome YAML.
It uses the canID as part of the topic.

```yaml
spi:
  id: McpSpi
  clk_pin: GPIO16
  mosi_pin: GPIO05
  miso_pin: GPIO04

canbus:
  - platform: mcp2515
    spi_id: McpSpi
    cs_pin: GPIO14
    can_id: 250
    bit_rate: 250KBPS
    use_extended_id: true
    on_frame:
    - can_id:      0x000  # listen to all messages
      can_id_mask: 0x000
      use_extended_id: false
      then:
      - mqtt.publish:
          topic: !lambda |-
           return "vitocal/can/" + std::to_string(can_id);
          payload: !lambda |-
            // Include ArduinoJson
            DynamicJsonDocument doc(256);  // Adjust size as needed

            // Add CAN frame ID
            doc["can_id"] = can_id;
            doc["message"] = ParseCanMessage::parse(x);
            std::string json;
            serializeJson(doc, json);
            return json;
```

MyHomeMyData, 2024-12-14:
I had same idea. Unfortunately, the performance of an ESPHome system is not sufficient to transmit the large number of CAN messages. I was also unsuccessful using a filter for certain CAN IDs.

I developed a solution using native ESP32 based on project Cannelloni. You may check it out [here](https://github.com/MyHomeMyData/ESP32cannelloni).

Scarbous, 2024-12-14:
@MyHomeMyData thank you for your answer.

Is it really the speed of the ESPHome implementation?

What is not understandable to me is why do I get so many messages on 0x0?

Here are the messages I receiv:
[image]

MyHomeMyData, 2024-12-14:
To my understanding there are two problems using ESPHome:
* main loop of ESPHome OS is executed every 16 ms. CAN-Message may arrive much faster. You may receive several CAN frames within this time frame.
* You need to implement a bi-directional communication to use open3e because it's based on protocol UDSonCAN. Therefore it won't even help to receive all CAN messages.

Scarbous, 2024-12-14:
thanks for the explanation

closed 2024-12-14 by Scarbous