open3e/open3e #173: Unplausible Werte bei DID2494 (eventuell Vorzeichenproblem)

Matt09xz, 2025-01-02:
Hallo zusammen,

ich logge die Daten meiner Vitocal 250-A und erhalte beim Abtauvorgang sehr große thermische Leistungen. Könnte es sein, dass DID2494 als 32 bit signed interpretiert werden muss? Bei Vertauschung Vorlauf / Rücklauf beim Abtauen würde das Sinn machen.

Auszug aus der Aufzeichnung mit Stellung 4/3-Wege Ventil, elektrischer Leistung und thermischer Leistung.
```
Uhrzeit          DID 2735       2486    2494
=============================
01.01.2025 02:43:01	1	4359	13529	
01.01.2025 02:43:33	1	4614	13888	
01.01.2025 02:44:05	1	4642	13550	
01.01.2025 02:44:37	1	3342	4294959637  (signed -7659)
01.01.2025 02:45:08	1	5000	4294954790  (signed - 12506)
01.01.2025 02:45:40	1	5518	4294948975  (signed -18321)
01.01.2025 02:46:12	1	1385	10598	
01.01.2025 02:46:43	0	831     6086	
01.01.2025 02:47:15	0	1353	6779	
01.01.2025 02:47:47	0	1226	5315	

```

Philip-Wiege, 2025-01-02:
Das klingt plausibel! Mir ist auch schon aufgefallen, dass bei Kältekreisumkehr Unsinn rauskommt.

Hast du den DID auch im Kühlbetrieb aufgezeichnet? Da hat man ja theoretisch auch negative Leistung.

Ich wühle mal in meinen aufgezeichneten Daten...

Philip-Wiege, 2025-01-02:
Ich hab gerade nochmal in meinen Daten geschaut. Beim Kühlen kommt tatsächlich auch Quatsch raus.

Dieser DID ist also die physikalische vorzeichenbehaftete Leistung der Außeneinheit.

Ich hatte das Problem "gelöst" indem ich in Home Assistant den Wert aus einem anderen DID berechnet habe. Irgendwo gabs noch nen DID für dich vorzeichenlose Leistung.

Willst du selber mal prüfen, ob signed die korrekte Leistung angezeigt wird?

Matt09xz, 2025-01-03:
Ich habe in Open3Edatapoints.py signed=True ergänzt und probiere es mal aus. 

```
2494 : O3EInt32(4, "CurrentThermalCapacityRefrigerantCircuit", scale = 1, signed=True),

```

Mein externer Wärmezähler zwischen Innen- und Außeneinheit zeigt in dem relevanten Zeitabschnitt übrigens die Leistung 0W an. Vermutlich kann der Zähler keine negativen Leistungen verarbeiten (Vorlauf < Rücklauftemperatur). 

Gruß
Matthias

Philip-Wiege, 2025-01-05:
Ich habs bei mir auch mal eingebaut. Sieht bisher alles plausibel aus. Mal den nächsten Abtauvorgang abwarten.

Vielleicht kann ichs heute noch in den Develop Branch einbauen.

Philip-Wiege, 2025-01-09:
Ich habe die Erkenntnis in den Develop Branch eingefügt. https://github.com/open3e/open3e/commit/929f42c80a344e4a786a321ad6fa819558f629e4

Habe die Werte getestet während des Abtaubetriebs.

Philip-Wiege, 2025-01-09:
Issue kann geschlossen werden, danke für eure Mitarbeit!

closed 2025-01-09 by Philip-Wiege