open3e/open3e #176: DepictSystem Skript wertet angegebenes CAN-Bus Argument nicht aus

Philip-Wiege, 2025-01-20:
Hi @philippoo66, ich glaube du bist ja der Meister des DepictSystem Skriptes oder?

In Zeile 67 wird entschieden, ob DoIP oder SocketCAN verwendet wird. Bei SocketCAN wird stumpf die variable "can" übergeben statt "args.can" daher kann man bei "-c" Argument glaube ich eingeben was man will und es wird immer "can0" versucht.

`                if(args.doip != None):
                    conn = DoIPClientUDSConnector(DoIPClient(args.doip, tx))
                else:
                    bus, conn = get_pycan_conn(can=can, ecurx=rx, ecutx=tx)`

philippoo66, 2025-01-21:
moin Philip @Philip-Wiege !

Sorry meine späte Antwort! 

Glaubst du das oder hast du das probiert? Die letzten Rückmeldungen waren meiner Meinung nach, dass es funktioniert, sogar mit AccepointPoint Mode / DoIP!?

Die eigentlich die Auswertung der Argumente passiert auf Module-Ebene, ab Zeile 312. Ursprünglich passierte ja das ganze 'main' auf Modulebene. https://github.com/open3e/open3e/blob/60d973a90c44e5cb8a4c18be9f779cb70677e6d9/Open3E_depictSystem.py 

Das hat @SanMiggel  geändert mit `[Wrap scripts in main() to allow for proper entry points.](https://github.com/open3e/open3e/commit/f54fdfd30faa0ba1415d1973006b8f245cf4798e)`

Wenn es aktuell nicht mehr funktionieren _sollte_ (aber das muss per Test verifiziert werden, ich meine mit `-c vcan0` hätte es bei mir auch geklappt), wäre es schön, wenn @SanMiggel  das noch mal anschaut!?

Grüsse!
Phil

Philip-Wiege, 2025-01-21:
Tach! Entschuldige bitte! Die Variable "can" wird unten tatsächlich nochmal überschrieben mit "args.can" Dann passt alles.

closed 2025-01-21 by Philip-Wiege