open3e/open3e #195: Open3e fails to start after update

0bscur3, 2025-02-07:
Hi all, 
today I upgraded open3e by do another `pip install git+https://github.com/open3e/open3e.git` and depictSystem. Afterwards I tried to restart the service but it fails with the following error:

```
Enter listener mode, waiting for commands on mqtt...
Traceback (most recent call last):
  File "/home/franz/open3e/.venv/bin/open3e", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/franz/open3e/.venv/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 457, in main
    listen(args.read, args.timestep)
  File "/home/franz/open3e/.venv/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 289, in listen
    cmnd_loop()
    ^^^^^^^^^^^
  File "/home/franz/open3e/.venv/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 203, in cmnd_loop
    readbydid(addr, didsub[0], json=(cd['mode']=='read-json'), raw=(cd['mode']=='read-raw'), sub=didsub[1])
  File "/home/franz/open3e/.venv/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 296, in readbydid
    showread(addr, did, value, idstr, json, msglvl)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/franz/open3e/.venv/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 323, in showread
    publishStr = mqttformatstring.format(
                 ^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: Unknown format code 'd' for object of type 'str'
```

My args are:
```
--can
can0
--read
268,269,271,274,284,318,320,321,322,324,325,355,381,389,391,396,491,538,548,1043,1659,1769,1771,1772,1775,1776,2333,2334,2346,2351,2369,2487,2488,2496,2735,2806,3016
--timestep
60
--mqtt
192.168.1.254:1883:open3e/vitocal252
--mqttformatstring
{device}_{ecuAddr:03X}_{didNumber:04d}_{didName}
--mqttuser
xxx:xxx
--listen
open3e/cmnd
--config
devices.json
```

I removed this formatter from the mqttformat string and it starts again: "04d"

Is this not recommended to use anymore(I still see it in some examples)or is it a bug?

neandr, 2025-02-07:
You have in your args
```
--mqttformatstring
{device}_{ecuAddr:03X}_{didNumber:04d}_{didName}
```
What is the ```{device}``` for? I never saw it in ```the mqttformatstring```!?

0bscur3, 2025-02-07:
I got this from the wiki: https://github.com/open3e/open3e/wiki/030-Installation-und-Inbetriebnahme-von-open3E Check the args there. It inserts the device name so "0x680" for standard devices.json

neandr, 2025-02-07:
OK!  Because I don't use the args, I was surprised to have that leading ```{device}```.
It seems that the error has a different cause.

philippoo66, 2025-02-07:
the `{device}` returns e.g. 'Vitocal' if you named your  0x680 ECU 'Vitocal' in devices.json

[image]

for the 0x684 ECU `{device}` will return '0x684' what is (almost) equal to `{ecuAddr:03X}` except the leading '0x'.

I could imagine thet the "04d" spcifier causes problems since didNumber now can also be 'BusIdentification' instead of 256 and for that handled asa string which could not get formatted 04d....

thank you for reporting @0bscur3 , we will fix that!

philippoo66, 2025-02-07:
@0bscur3 I fixed that in [branch develop](https://github.com/open3e/open3e/tree/develop). 
        modified:   src/open3e/Open3Eclass.py
        modified:   src/open3e/Open3Eclient.py

If it doesn't mean no problem to you it would be great if you could test it (with your mfstr back to `{device}_{ecuAddr:03X}_{didNumber:04d}_{didName}`)

0bscur3, 2025-02-10:
Thanks @philippoo66, I will see if I have some time today to test it. Thanks for fix.

philippoo66, 2025-03-17:
@0bscur3  can we close this issue?

philippoo66, 2025-07-24:
https://github.com/open3e/open3e/discussions/1#discussioncomment-13873014