open3e/open3e #221: Writing in JSON data format fails

MaWa1302, 2025-03-26:
When trying to learn writing with open3e, I followed the README.

+ Simple reading as described
`open3e -j -r 538`
```
{'Mode': 0, 'State': 0}
```

+ Verbose reading to get visual name
`open3e -j -v -r 538` 
```
0x680 538 ExternalDomesticHotWaterTargetOperationMode {"Mode": 0, "State": 0}
closing 0x680 - bye!
closing 0x684 - bye!
closing 0x68c - bye!
closing 0x6a1 - bye!
closing 0x6c3 - bye!
closing 0x6c5 - bye!
closing 0x6cf - bye!
```
Why are double quotes used here?

+ Testing example from README (just copied)
`open3e -w 538='{"Mode": 1, "State": 0}'` 
```
write: 1664.538 = {"Mode": 1
2025-03-26 21:06:42 [ERROR] UdsClient: [TypeError] : string indices must be integers, not 'str'
Traceback (most recent call last):
  File "/home/pi/open3e/_venv/bin/open3e", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/pi/open3e/_venv/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 523, in main
    val = writeArg[1]   # must be decoded form
          ~~~~~~~~^^^
IndexError: list index out of range
```

+ Exchanging single and double-quotes
`open3e -w 538="{'Mode': 1, 'State': 0}"` 
```
write: 1664.538 = {'Mode': 1
2025-03-26 21:09:19 [ERROR] UdsClient: [TypeError] : string indices must be integers, not 'str'
Traceback (most recent call last):
  File "/home/pi/open3e/_venv/bin/open3e", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/pi/open3e/_venv/lib/python3.11/site-packages/open3e/Open3Eclient.py", line 523, in main
    val = writeArg[1]   # must be decoded form
          ~~~~~~~~^^^
IndexError: list index out of range
```
Is this description sufficient? Am I doing something wrong?

MyHomeMyData, 2025-03-26:
Unfortunately, in readme the option `-j` is missing. Pls. try `open3e -j -w 538="{'Mode': 1, 'State': 0}"`

I'll fix the readme.

The double-quotes are needed to make it a valid json object.

MaWa1302, 2025-03-27:
Thanks for the quick reply. Works fine now.

closed 2025-03-27 by MyHomeMyData