open3e/open3e #226: @Boisi-73 Ich empfehle, 2 Unterverzeichnisse zu machen, jeweils depictSystem laufen zu lassen. Dann 2 open3e Instanzen zu starten wie von @neandr beschrieben. Es ist nur eine Installation von open3e dafür nötig.

Boisi-73, 2025-04-03:
@Boisi-73 Ich empfehle, 2 Unterverzeichnisse zu machen, jeweils depictSystem laufen zu lassen. Dann 2 open3e Instanzen zu starten wie von @neandr beschrieben. Es ist nur eine Installation von open3e dafür nötig.

Warum verwendest Du nicht den ioBroker.e3oncan Adapter? 2 Instanzen erstellen, eine für WP und eine für Gas verwenden. Sollte problemlos funktionieren. 
Wenn Du Daten per MQTT benötigst, kann das ja in ioBroker konfiguriert werden.

_Originally posted by @MyHomeMyData in https://github.com/open3e/open3e/discussions/223#discussioncomment-12705987_

philippoo66, 2025-04-03:
ich glaub hier is was schief gelaufen? können wir diesen issue schliessen oder löschen?

MyHomeMyData, 2025-04-04:
@Boisi-73 Was ist der Zweck dieses Issues?

Ich kann gerne [hier](https://github.com/open3e/open3e/wiki/030-Installation-und-Inbetriebnahme-von-open3E) ein paar Zeilen ergänzen, um dieses Thema abzudecken. Den Issue würde ich dann schließen.

Boisi-73, 2025-04-05:
> [@Boisi-73](https://github.com/Boisi-73) Was ist der Zweck dieses Issues?
> 
> Ich kann gerne [hier](https://github.com/open3e/open3e/wiki/030-Installation-und-Inbetriebnahme-von-open3E) ein paar Zeilen ergänzen, um dieses Thema abzudecken. Den Issue würde ich dann schließen.

..das war ein Fehler von mir, sorry.
Der issue kann geschlossen werden.

MyHomeMyData, 2025-04-05:
Das Verfahren ist jetzt im [Wiki](https://github.com/open3e/open3e/wiki/030-Installation-und-Inbetriebnahme-von-open3E#mehrere-unabh%C3%A4ngige-ger%C3%A4te-mit-open3e-betreiben) beschrieben.

closed 2025-04-05 by MyHomeMyData