open3e/open3e #232: Videotutorial + Beitrag für VX3 in Homeassistant einbinden

cryd-s, 2025-04-24:
Hi zusammen,

ich hab ein Videotutorial und einen Beitrag zur Installation und Einbindung in Homeassistant gemacht.
Bei Interesse würde ich mich freuen wenn ihr das in euren Anleitungen mit verlinkt.

Ich selbst hab trotz eurer Doku mit der VX3 ganz schön zu kämpfen gehabt bis es lief. Wenn man weiß wie gehts aber super einfach und mega schnell. 

Danke auf jedenfall nochmal für eure Super Arbeit!!!

Beitrag und Videorelease sind am 27.04 um 9Uhr

https://youtu.be/u_fkwtIARug
https://crydteam.de/2025/04/27/viessmann-vx3-in-homeassistant/

Solltet ihr noch irgendwas haben schauts gern kurz am Discord vorbei bin da eig fast immer online.
https://discord.gg/2vEVdejeBZ

philippoo66, 2025-04-25:
hey @cryd-s ! 

klasse Sache - vielen Dank! 

aktuell ist allerdings das yt video noch privat geschaltet und bei dem site link bekomme ich 404. könntest du das noch richten?! 

und hast du eine bestimmte Anleitung im Blick (wo wir das verlinken sollen)? wir haben inzwischen so viele...

Grüsse!
Phil

cryd-s, 2025-04-25:
@philippoo66 

Hi Phil,
die beiden Links sind ab Sonntag 09:00 erreichbar. (da is meine normale Releasezeit für die Videos)
Wenn du vorher nen Blick drauf werfen magst sag mir kurz Bescheid dann können wir gemeinsam mal drüber schauen wenn du magst.

philippoo66, 2025-04-25:
eilt nicht. ich hab von HomeAssistant eh keinen Schimmer, bin da also eigentlich nicht der richtige ;-)

cryd-s, 2025-04-25:
OK dann pack das gern einfach in die Wiki mit rein is Grundsätzlich die VX3 Konfiguration + MQTT + Smarthomesystem (im Video und der Anleitung Homeassistant)

cryd-s, 2025-04-27:
so @philippoo66 ist jetzt soweit alles online

neandr, 2025-04-28:
@cryd-s Im Video wird angekündigt die HomeAssistant Lösung zu erweitern, um die VX3-Batterie auch durch Strom aus dem Netz zu laden. Wäre super wenn das ginge.
Gibt es dafür schon einen Zeitplan?

neandr, 2025-04-28:
OK, mein Wunsch, die VX3-Batterie aus dem Netz zu laden, würde bei flexiblen Strom-Tarifen Sinn machen. Das Laden in Zeiten niedriger Preise würde die Amortisation des VX3/PV + WP sicherlich deutlich verbessern. 
In diesem Sinne habe ich die Aussage des Videos wohl nicht richtig gehört, sorry.
Aber schön wäre es schon ... vielleicht ein weiteres Projekt?

cryd-s, 2025-04-30:
muss ich mir nach dem Urlaub mal anschauen

cryd-s, 2025-04-30:
> OK, mein Wunsch, die VX3-Batterie aus dem Netz zu laden, würde bei flexiblen Strom-Tarifen Sinn machen. Das Laden in Zeiten niedriger Preise würde die Amortisation des VX3/PV + WP sicherlich [...]

aber hast du die Installation wie von mir erklärt durchgeführt? hat alles geklappt?

philippoo66, 2025-04-30:
moin @cryd-s !

so, jetz hab ich endlich das Video auch mal geschaut - is ja echt vooolll top!! Richtig Klasse Video, richtig professionell und so. War bestimmt auch einiges an Arbeit und Zeit?! Vielen, vielen Dank nochmal!!

Ich werd das an mehreren Stellen verlinken, auch die nicht-Home Assistant User können ja die ersten 2/3 bestens gebrauchen.

Eine Sache ist mir aufgefallen. In deinem args.txt

- ist `--scanall` drin. Das wird ignoriert, wenn `--listen` gesetzt wird. Kann der Klarheit wegen also auch weggelassen werden.
- folgt auf `--listen` ein `1`. Hinter `listen` steht ja der MQTT Topic, auf den open3e lauscht und auf den die commads geschickt werden, wenn man denn was schicken will. Da wir da draus einen String machen, geht das offensichtlich, ist aber ziemlich 'ungewöhnlich'. Der default ist hier ja `open3e/cmnd` was schon deutlich sprechender ist.

Natürlich macht es keinen Sinn, wegen dieser Feinheiten das Video zu ändern, und wenn man nur das args.txt ändert, passt es nicht mehr zum Video. Aber vielleicht macht es Sinn, in der Nähe des args.txt eine kleine entsprechende Randnotiz zuzufügen?

beste Grüsse!
Phil

neandr, 2025-04-30:
@cryd-s 
> aber hast du die Installation wie von mir erklärt durchgeführt? hat alles geklappt?    

Die beiden Projekte 'open3e' und 'E3onCAN' sind für mich die Grundlage für eine Lösung RPI/OneBase Systeme (WP/PV-Batterie/Wallbox) mit Node-RED. Daran arbeite ich -- um zu einer guten Implementierung zu kommen, leider mit zu wenig Zeit. 
Insofern interessieren mich andere Bemühungen wie HomeAssistant oder ioBroker, bzw viel mehr mit welchen Anlagen Parametern  kann man was machen. Jede Detailinfo /-Erkenntnis ist hilfreich. Und der Ansatz für HA sieht schon mal sehr gut aus.

_PS   @philippoo66 Sollte der Beitrag nicht besser als Diskussion plaziert werden? Lässt sich das verschieben?_

cryd-s, 2025-04-30:
> moin [@cryd-s](https://github.com/cryd-s) !
> 
> so, jetz hab ich endlich das Video auch mal geschaut - das is ja echt vooolll top!! Richtig Klasse Video, richtig professionell und so. War bestimmt auch einiges an Arbeit und Zeit?! Vielen, vielen Dank nochmal!!
> 
> Ich werd das an mehreren Stellen verlinken, auch die nicht-Home Assistant User können ja die ersten 2/3 bestens gebrauchen.
> 
> Eine Sache ist mir aufgefallen. In deinem args.txt
> 
> * ist `--scanall` drin. Das wird ignoriert, wenn `--listen` gesetzt wird. Kann der Klarheit wegen also auch weggelassen werden.
> * folgt auf `--listen` ein `1`. Hinter `listen` steht ja der MQTT Topic, auf den open3e lauscht und auf den die commads geschickt werden, wenn man denn was schicken will. Da wir da draus einen String machen, geht das offensichtlich, ist aber ziemlich 'ungewöhnlich'. Der default ist hier ja `open3e/cmnd` was schon deutlich sprechender ist.
> 
> Natürlich macht es keinen Sinn, wegen dieser Feinheiten das Video zu ändern - es funktioniert ja offenschtlich. Und wenn man nur das args.txt ändert, passt es nicht mehr zum Video. Aber vielleicht macht es Sinn, in der Nähe des args.txt oder in einer Fussnotiz eine kleine entsprechende Randnotiz zuzufügen?
> 
> beste Grüsse! Phil

ich hab mal das --scanall raus genommen.
Das sollte kein Thema sein weil genau mit dem Video vergleicht es eh niemand. Danke für den Hinweis. Bin nicht so extrem tief in der Materie aber der Weg hat für mich super funktioniert und das wollte ich eben gern so verständlich und nachvollziehbar rüberbringen damit auch Leute die nicht soviel technisches Verständnis haben das Projekt auch umsetzen können.

Wenn es irgendwie noch gute Features oder so gibt die ich nicht kenne oder so gern her damit ich mach gerne solche Tutorial Videos.

MyHomeMyData, 2025-05-01:
@cryd-s Auch von meiner Seite herzlichen Dank für Deine Initiative und das super Video! Das ist für neue Anwender von open3e sehr hilfreich.
Ich habe noch zwei Anmerkungen
* den Wert -t 1 sehe ich kritisch. Damit werden die gelisteten Datenpunkte jede Sekunde abgerufen. Jede einzelne Kommunikation braucht eine gewisse Zeit und manche Datenpunkte sind ziemlich lang, bis zu 181 Byte Nutzdaten für einen Datenpunkt!
In einem Intervall von einer Sekunde können schätzungsweise ca. 10-20 Datenpunkte abgefragt werden, je nach Datenpunkten, Anlagenkonfiguration, verwendetem Raspi usw. Macht man die Liste zu lang, wird es scheinbar immer noch korrekt funktionieren. Aber: open3e pusht alle Aufträge in eine Warteschlange. Reicht nun die Sekunde nicht, um diese Warteschlange abzuarbeiten, wird sie immer länger. D.h. die Werte, die man per MQTT erhält werden immer älter. Das hat dann mit Echtzeit nichts mehr zu tun.
Mein Empfehlung: Entweder nur wenige Werte abfragen (max. 10) oder auf 5 oder 10 Sekunden Intervall gehen.
* Als Entwickler des [ioBroker-Adapters e3oncan](https://github.com/MyHomeMyData/ioBroker.e3oncan/blob/main/README.md) würde es mich freuen, wenn Du einen Hinweis unterbringen könntest, dass es für ioBroker einen Adapter gibt, der die SW-Seite zum Kinderspiel macht und zudem weitere Features bietet, z.B. das Auslesen von Energiezählern. open3e muss dabei gar nicht installiert werden.

Viele Grüße
Jürgen

philippoo66, 2025-05-01:
@MyHomeMyData  es werden ur ein paar DIDs abgefragt:
```
--read
0x68f.1664,0x680.1836,0x680.1837,0x680.1838,0x680.1603,0x680.1690,0x680.1831,0x680.1832,0x680.1833
```
ich hatte nich auch zuerst an em scanall aufgehängt, aber es sind ja nur die 9 Stück.

@cryd-s 'unschöner' als das scann all, das eh ignoriert wird, finde ich das 
```
--listen
1
```
weil ein Topic mit Namen "1" ist nicht wirklich toll...

MyHomeMyData, 2025-05-01:
@philippoo66 Na ja, im Beispiel im Video werden nur diese Datenpunkte abgefragt. Das ist ok. **Aber die Anwender werden doch die Liste länger und länger machen**. Vor allem, wenn sie mehrere Geräte haben. Und dann wird das seltsame Effekte zeitigen, weil es scheinbar funktioniert. Aber irgendwann sind die per MQTT erhaltenen Werte den Echtzeitdaten ein paar Stunden hinterher. Das wird keiner nachvollziehen können.

Bzgl. `--listen 1` bin ich vollkommen bei Dir. Aber das hattest Du ja bereits angemerkt.

cryd-s, 2025-05-01:
> [@philippoo66](https://github.com/philippoo66) Na ja, im Beispiel im Video werden nur diese Datenpunkte abgefragt. Das ist ok. **Aber die Anwender werden doch die Liste länger und länger machen**. [...]

Mit dem --listen 1 läuft mit den wenigen Daten sehr gut eigentlich. Ich mach aber noch ne Notiz dazu das bei Problemen hier das listen erhöht werden sollte.

philippoo66, 2025-05-03:
im Readme in develop is der Hinweis auf das Video und die Seite schon drin. Im Wiki fehlt es noch. Coming soon ;-)

philippoo66, 2025-05-03:
> Wenn es irgendwie noch gute Features oder so gibt die ich nicht kenne oder so gern her damit ich mach gerne solche Tutorial Videos.

open3e hat inzwischen sooo viele Features, die alle irgendwer brauchen kann (deswegen haben wir sie reingebaut). aber wenn man die alle vorstellen will, muss man schon ziemlich tief drin stecken.

Was vielleicht noch interessant wäre, ist das [E2onCANcollect](https://github.com/MyHomeMyData/E3onCAN) - das sammelt die vom System versendeten Daten ein, ohne dass es irgendwas requestet. Nebenbei erscheinen da auch die Daten vom Energiezähler, den man mit open3e nicht packen kann, weil er zugekauft ist und kein UDS spricht.

Und bestimmt interessant ist der ioBroker Adapter E3onCAN, der aber glaubich so einfach ist, dass das Video ziemlich kurz würde (wobei die Möglichkeiten auzuzeigen sich wahrscheinlich doch wieder lohnt).

Allem gemeinsam ist der Hardware Part und das 'in-Betrieb-Setzen' des CAN Busses - und das hast du ja schon supi beschrieben.

philippoo66, 2025-07-07:
moved to Discussions