open3e/open3e #234: Sub-DID mit Index 0 lässt sich per JSON über MQTT nicht setzen

Philip-Wiege, 2025-05-07:
Hallo zusammen,

ich möchte gerne den Modus des Heizkreis 1 per MQTT aus Home Assistant steuern.
Dazu möchte ich nur den Sub-DID mit Index 0 verändern.
Leider verschluckt die JSON Notierung das ".0" und somit wird statt "1415.0" nur "1415" übermittelt, was nicht funktioniert.
Wenn ich die 1415.0 als String formatiere also "1415.0" schreibe versteht open3E das glaube ich nicht.
Bei anderen Sub Indizes klappt das natürlich, da diese nicht weggebügelt werden.

Stehe ich da auf dem Schlauch oder haben wir da wirklich eine Schwäche?

Viele Grüße
Philip

hier der JSON Datensatz der nach dem Verschlucken an Open3E übermittelt wird:

{
  "mode": "write",
  "data": [
    [
      1415,
      0
    ]
  ]
}

Philip-Wiege, 2025-05-07:
@MyHomeMyData Hast du den JSON Interpretationscode geschrieben? Ist es ein großer Umbau, wenn die DID Nummer als String übermittelt werden KANN?

Wird z.B. 1415.1 (Float in JSON) von Open3E als Float entgegengenommen und dann in 1415 und 1 interpretiert?

Oder wird 1415.1 (Float in JSON) sowieso nach String gecastet und dann am Punkt zerlegt?

Ich glaube Option 2 ist es nicht oder, da Open3E mein Kommando mit "1415.0" nicht akzeptiert.

Eventuell könnte man die Logik so umbauen, dass Inputs die JSON Floats oder Strings beide verarbeitet werden können.

MyHomeMyData, 2025-05-07:
@Philip-Wiege Aus meiner Sicht arbeitet open3e hier vollkommen korrekt. Der Inhalt von 1415.0 ist aber keine Zahl, sondern ein JSON-Objekt. Das sieht man, wenn man den Datenpunkt abfragt:
```
open3e -v -r 0x680.1415.0
0x680 1415 Mode {"ID": 1, "Text": "Heating"}
```
Per MQTT kann man diese Daten schreiben, wenn man "1415.0" oder "1415.Mode" schreibt und das vollständige JSON-Objekt übergibt. Für die möglichen Werte von 0 bis 5 sieht das also so aus:
```
{"mode":"write", "data":[["1415.0",{"ID": 0, "Text": "Off"}]]}
{"mode":"write", "data":[["1415.0",{"ID": 1, "Text": "Heating"}]]}
{"mode":"write", "data":[["1415.0",{"ID": 2, "Text": "Parallel Operation: Heating HotWater"}]]}
{"mode":"write", "data":[["1415.0",{"ID": 3, "Text": "Parallel Operation: Heating Cooling"}]]}
{"mode":"write", "data":[["1415.0",{"ID": 4, "Text": "TestMode"}]]}
{"mode":"write", "data":[["1415.0",{"ID": 5, "Text": "Cooling"}]]}
```
Genauso funktioniert z.B. `{"mode":"write", "data":[["1415.Mode",{"ID": 0, "Text": "Off"}]]}`
Man muss das leider genau so angeben, incl. Text-Feld. Schreiben von tieferen Strukturen, z.B. "1415.Mode.ID" ist - glaube ich - nicht implementiert.
Alternativ könnte natürlich "write-raw" verwendet werden.

Philip-Wiege, 2025-05-07:
Ahh ok, es liegt also daran, dass der Sub-DID 1415.0 vom Type "Enum" ist. Bei den Sub-DIDs die numerisch sind, kann man per JSON über MQTT einfacher schreiben.

Kann ich denn den 1415.0 einfach mit write-raw isoliert, numerisch verstellen, ohne dass HomeAssistant die Übersetzung der Werte kennen muss und ohne dass der Rest von 1415 mitgesendet werden muss?

MyHomeMyData, 2025-05-07:
> Kann ich denn den 1415.0 einfach mit write-raw isoliert, numerisch verstellen, ohne dass HomeAssistant die Übersetzung der Werte kennen muss und ohne dass der Rest von 1415 mitgesendet werden muss?

Ja, das geht mit (hier für Wert 2): `{"mode":"write-raw", "data":[["1415.0","02"]]}` 1415.1 bleibt dann unverändert.

MyHomeMyData, 2025-05-07:
@Philip-Wiege Hab noch eine andere Idee. Wenn Du die Datei `Open3Edatapoints_680.py` änderst und `1415 : None,` ersetzt durch `1415 :  O3EComplexType(2, "MixerOneCircuitOperationState", [O3EByteVal(1, "Mode"), O3EByteVal(1, "State")]),` dann funktioniert der Befehl `{"mode":"write", "data":[["1415.Mode",2]]}`einwandfrei.

Wenn Du wieder depictSystem startest, könnte das allerdings überschrieben werden.
@philippoo66 Ist das so? Oder bleiben Änderungen durch den Anwender erhalten?

philippoo66, 2025-05-07:
das ist eine gute Idee! Leider schreibt depict tatsächlich jedes mal die Datei komplett neu. Aber man macht ja depict nicht jeden Tag...

> {"mode":"write", "data":[["1415.Mode",{"ID": 0, "Text": "Off"}]]}
Man muss das leider genau so angeben, 

das ist es, weswegen ich von json alles andere als überzeugt bin. Wenn ich den Mode setzenn will, dann will ich ihn entweder auf 0 oder auf Off setzen. 

So denkt man, das ist intuitiv. 

Wenn ich erst in der datapoints.py nachgucken muss, welcher Enum das ist, dann mir den passenden Text zu dem Wert aus der enums.py raussuchen muss (oder andersrum), dann wissen, dass das eine ID heisst und das andre Text, und dann das erst in einen json checker geben muss, damit es formal richtig ist, ist das schlicht und ergreifend nicht praktikabel. 

Beim Lesen ist es ja schön und gut, wenn man Wert und Text bekommt (wobei das auch einfach durch einen Doppelpunkt getrennnt sein könnte), aber bein Schreiben ist das einfach Mist (gelinde gesagt)...

MyHomeMyData, 2025-05-07:
@philippoo66 Du musst nicht in die datapoints.py gucken. Einfach den Datenpunkt als json abfragen und Du weißt, was Du schreiben musst.
Speziell für enum-dekodierte Daten ist das wenig praktikabel. Aus meiner Sicht wäre die beste Abhilfe, auch tiefere Sub-Dids zu implementieren. Dann könnte man hier 1415.Mode.ID als Zahl schreiben.

Für ioBroker.e3oncan habe ich das implementiert. Das hat mich einiges an Hirnschmalz gekostet.

philippoo66, 2025-05-07:
ps. nochwas: wenn ich im ioBroker einen Wert in influx bringen will, um ihn hinterher mit grafana darzustellen, kann ich das Objekt anklicken, das Häkchen bei influx setzen und fertig. 

Wenn es allerdings ein json Objekt ist, muss ich erst ein Alias anlegen, in dem Alias die Formel zum Rauspicken des Wertes reinkopieren (vorher suchen...) und kann dann den influx Haken beim Alias setzen. Ich bin da jedes mal generft, wenn ich den ganzen Umweg gehen muss...

Wenn ich den Wert nicht mehr brauche, ist es beim json Objekt auch wieder entsprechend mehr Aufwand, den ganzen Rattenschwanz rauszuwerfen.

Philip-Wiege, 2025-05-07:
@MyHomeMyData Danke für den Tipp mit dem RAW Schreiben! Ich konvertiere den Int Wert in HomeAssistant einfach in einen Hex-String und schreibe mit write-raw isoliert den 1415.0.

closed 2025-05-07 by Philip-Wiege

philippoo66, 2025-05-07:
@MyHomeMyData 
> Du musst nicht in die datapoints.py gucken. Einfach den Datenpunkt als json abfragen und Du weißt, was Du schreiben musst.

Das ist ein Trugschluss! Wenn ich den Datenpunkt abfrage, bekomme ich das Päärchen, wie es ist, ich will aber einen ANDEREN Wert setzen,

MyHomeMyData, 2025-05-07:
@philippoo66 Ja, das hat mich auch genervt. Grafana habe ich schnell wieder weggelegt. Ist mir alles zu umständlich.

Hat mich so geärgert, dass ich [ioBroker.flexcharts](https://github.com/MyHomeMyData/ioBroker.flexcharts#readme) entwickelt habe. Damit kann man so ziemlich alles machen, was [Apache eCharts](https://echarts.apache.org/en/index.html) kann und das alles komplett innerhalb (!) von ioBroker. Läuft alles lokal. Man muss bereit sein, Javascript zu nutzen, aber damit kann man auch json-Daten wunderbar in Charts verwandeln.

philippoo66, 2025-05-07:
> auch tiefere Sub-Dids zu implementieren. Dann könnte man hier 1415.Mode.ID als Zahl schreiben.

eigentlich muss nur der Encoder geändert werden. Wenn man ihn mit einer Zahl füttert, nimmt er die, wenn man ihn mit Text füttert, versucht er den zuzuordnen, und wenn es denn unbedingt ein json Objekt ist, dann nimmt er das eben als solches,

philippoo66, 2025-05-07:
> ViGuide, Bedienfeld an den Anlage oder einfach warten, bis die Anlage von sich aus in den gewünschten Zustand geht.

das ist jetzt nicht dein Ernst, oder?

MyHomeMyData, 2025-05-07:
> eigentlich muss nur der Encoder geändert werden. Wenn man ihn mit einer Zahl füttert, nimmt er die, wenn man ihn mit Text füttert, versucht er den zuzuordnen, und wenn es denn unbedingt ein json Objekt ist, dann nimmt er das eben als solches,

Ich halte nichts von solchen Strukturbrüchen. Entweder wir machen es richtig, oder lassen es so. Es gibt ja Workarounds. Können wir aber gerne nächste Woche diskutieren.