open3e/open3e #248: write DID:  kein Codec nicht erkannt?!

philippoo66, 2025-07-07:
ich hab grad gesehen, dass bei 

```
    def writeByDid(self, did, val, raw:bool, useService77=False, sub=None, readecu=None):
        #print( did, val, raw, useService77, sub)

        try:
            idid = self.get_did_as_int(did)

            if(sub is None):
                return self._writeByDid(idid, val, raw, useService77)

            if(idid not in self.dataIdentifiers):
                raise ValueError(f"No Codec specified for DID {idid}")

            ...            
```

die Reihenfolge wohl nicht stimmt?!

Müsste  `if(idid not in self.dataIdentifiers):` nicht vor dem `if(sub is None):` und dem Aufruf von dem eigentlichen writeByDid kommen?!

ich weiss, ich hab das selber geschrieben, aber wenn ich versuche, dem nachzugehen, bin ich schnell bei `self.uds_client.write_data_by_identifier` und bei der Überlagerung (die ist von Jürgen) bin ich mir nicht mehr sicher.

@MyHomeMyData könntest du da bitte mal kurz anschauen!?

MyHomeMyData, 2025-07-08:
@philippoo66 Hallo Phil, ja, sehe ich auch so. Das sollten wir vertauschen.

Bei readByDid() ist der Ablauf zwar auch so, aber da erfolgt in _readByDid() nochmal eine Prüfung:
```
    def _readByDid(self, did:int, raw:bool):
        if(did in self.dataIdentifiers): 
```
Die gibt es bei _writeByDid() nicht.

Für Service77 sehe ich da keinen Unterschied. Da wird die Datenpunktdefinition genauso gebraucht wie beim normalen Service.

philippoo66, 2025-07-08:
da war noch was mit DID 3335, aber der Mensch meldet sich nicht... Ansonsten denke ich, meine kleine Reihenfolgentauschaktion ist ziemlich gefahrlos

philippoo66, 2025-07-09:
irgendwann sollten wir auch mal das überflüssige `import open3e.Open3Ecodecs` (oder das `from open3e.Open3Ecodecs import *`) auch mal aus den datapoints.py rausnehmen (also auch in depict). ich vergess es immer wieder...