open3e/open3e #255: open3e not connecting to EMQX MQTT broker

nielssch, 2025-08-01:
I am using the EMQX MQTT broker instead of Mosquitto. when trying to start open3e, I keep getting this response:

Enter listener mode, waiting for commands on mqtt...
mqtt broker disconnected. reason_code = Unspecified error
mqtt broker disconnected. reason_code = Unspecified error
mqtt broker disconnected. reason_code = Unspecified error

Other clients are using the MMQX broker with no problems using the same user credentials. I also created a dedicated user for open3e - no change. 

To validate my setup, I installed a Mosquitto docker. Using that broker, open3e is starting with no errors and sending data. 

I hope one of you has ideas how to fix this since I am happily using EMQX since a long time and would hate to change just for open3e.

philippoo66, 2025-08-02:
obviously there is a log
`docker logs <emqx-container-name>`

please check also this:
https://chatgpt.com/share/688de376-f850-8005-b6e9-2743462b6a60

nielssch, 2025-08-02:
Thanks again @MyHomeMyData  and @philippoo66 , your comments helped me progressing. Running open3e with the command below fired no error:

`open3e -c can0 -dev vcal -m 10.1.5.3:1883:open3e -muser mqtt:mqtt -l open3e/cmnd`

This means I will need to find the mistake in my argument (arg) file that I'm using when launching open3e.

closed 2025-08-02 by nielssch