open3e/open3e #259: DID 3016 showing wrong data for Vitocal 250-A

nielssch, 2025-08-08:
First off, many thanks to all open3e contributors for the great work!!!

After getting open3e working, my Vitocal 250-A is sending 167 topics to my MQTT broker. They all look good except 3016

[image]

How can I fix this DID to show the correct value?

MyHomeMyData, 2025-08-08:
These are typical values when a specific sensor is not present. Is a heating buffer temperature sensor present in your system?

nielssch, 2025-08-08:
Good question... I have no clue. Is there a way to find out?

philippoo66, 2025-08-09:
I think the `Unknown = 4` points to absence. Isnt't that usually 0 if present and no fault?

MyHomeMyData, 2025-08-09:
Bei mir sieht das so aus:
```
{
  "Actual": 22.7,
  "Minimum": 20.2,
  "Maximum": 46.4,
  "Average": 22.6,
  "Unknown": 0
}
```
Passt also zu 0 => ok. Wenn das der Sensor-Status ist, warum steht dann im DID "unkown"?

philippoo66, 2025-08-09:
haben wir denn eine enum zu sensor_status?

ich hab das hier gefunden

```
        switch (a.a[n1dVar.ordinal()]) {
            case 1:
                return zkb.screen_service_layer_buffer_mid_temperature_sensor_status_no_error;
            case 2:
                return zkb.screen_service_layer_buffer_mid_temperature_sensor_status_interruption;
            case 3:
                return zkb.screen_service_layer_buffer_mid_temperature_sensor_status_short_circuit;
            case 4:
                return zkb.screen_service_layer_buffer_mid_temperature_sensor_status_electrical_fault;
            case 5:
                return zkb.screen_service_layer_buffer_mid_temperature_sensor_status_not_available;
            case 6:
                return zkb.screen_service_layer_buffer_mid_temperature_sensor_status_invalidates;
            default:
                throw new NoWhenBranchMatchedException();
        }

        switch (a.a[n1dVar.ordinal()]) {
            case 1:
                return zkb.screen_service_layer_dhw_flow_sensor_status_no_error;
            case 2:
                return zkb.screen_service_layer_dhw_flow_sensor_status_interruption;
            case 3:
                return zkb.screen_service_layer_dhw_flow_sensor_status_short_circuit;
            case 4:
                return zkb.screen_service_layer_dhw_flow_sensor_status_electrical_fault;
            case 5:
                return zkb.screen_service_layer_dhw_flow_sensor_status_not_available;
            case 6:
                return zkb.screen_service_layer_dhw_flow_sensor_status_invalidates;
            default:
                throw new NoWhenBranchMatchedException();
        }
```

das harmoniert aber so nicht mit unserer Vermutung. höchstens wenn wir 1abziehen...

philippoo66, 2025-08-09:
bei den Optolink Vcals ist es so, dass mit den Sensorwerten auch der Status mitgeliefert wird. und da ist es ja 'halbwegs dokumentiert'...

philippoo66, 2025-08-09:
sollen wir die enum von oben mal mit dem -1 shift als "experimental" in die Enums schreiben un ein paar Sensoren zuordnen? oder sollen wir warten, bis @DpunktS vielleicht was sagt?

DpunktS, 2025-08-09:
Ja der letzte byte sollte der Errorstatus des Sonsors sein, aber welcher Wert für welchen Status steht kann ich auch nicht sagen.

derretro, 2025-10-05:
> First off, many thanks to all open3e contributors for the great work!!!
> 
> After getting open3e working, my Vitocal 250-A is sending 167 topics to my MQTT broker. They all look good except 3016
> <img alt="Image" width="331" height="103" src="https://private-user-images.githubusercontent.com/15388338/476186826-f6ad1f2a-027f-42cb-9555-7cd00eda8a21.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTk3MDE0NzEsIm5iZiI6MTc1OTcwMTE3MSwicGF0aCI6Ii8xNTM4ODMzOC80NzYxODY4MjYtZjZhZDFmMmEtMDI3Zi00MmNiLTk1NTUtN2NkMDBlZGE4YTIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEwMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMDA1VDIxNTI1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdiODEwOGVlNDExNzQxOTM0OTFkNDUxZGZmYjQzYTM2ODA1YjZhMTVhY2Q4NDk5ZTU0OGM2NmE4YmQ2YTljZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.IhPQe-jz4reePC6u58ra7Y1roJbjqPYr9vMdjx6kh6o">
> 
> How can I fix this DID to show the correct value?

Das liegt daran, dass du vmtl. einen Puffer hast der Heizen **und Kühlen** kann. Deshalb braucht man nicht die 3016 sondern die 3018. Hatte den Fehler bei mir auch.